platformio/platform-espressif32 #1033: Issue in execution of ESP32S3 after flashing through PlatformIO and windows 11

RitaAJ, 2023-02-17:
We are using ESP32-S3-WROOM-1 module based PCB in our project with VS code and platformio framework. We noticed some strange things while flashing code in the module.

We also have ESP32s3devkitc1. When we use the machine with WINDOWS10, flashing of code through VS code to both ESP32-S3-WROOM-1 module based PCB and ESP32S3devkitc1 goes fine, everything is smooth and we get desired output on serial monitor and execution of program e.g., the behavior of port pins all is as expected.

But the same thing when we do with Windows 11 then we only get the expected results on the ESP32S3-devkit-C1 module and not on the ESP32-S3-WROOM-1 module-based PCB. If we flash the program with no change in hardware and software setup on ESP32-S3-WROOM-1 module-based PCB then we get the "Core dumb flash config is corrupted" msg on the serial monitor. And also, did not get the expected result on module port pins.
Here is this mail attaching a screenshot of the message on the serial monitor after flashing through Windows 11. Please guide us for the same.We are using platformio for programming and flashing. We are not using ESP IDF framework for our projects in VS code
So in platform how to check and change menuconfig settings. We were unable to find those settings.
When we used to do programs with ESP IDF at that time also we were facing the same issue with windows11 setup. With IDF we were only able to do the program through the UART port on ESP32S3devkitC1.
We have taken the trials on 4 different laptops, and the result is the same for windows 11. This is what we have observed. Please guide us further in detail.
In Windows 10 VS code version is 1.75.1 with platformio and IDF framework version V4.4.1
In Windows 11 VS code version is 1.75.1 with platformio and IDF framework version V5.0

valeros, 2023-02-20:
Hi @RitaAJ 

> So in platform how to check and change menuconfig settings. We were unable to find those settings.

Just run `pio run -t menuconfig` in the IDE terminal.

> In Windows 10 VS code version is 1.75.1 with platformio and IDF framework version V4.4.1
In Windows 11 VS code version is 1.75.1 with platformio and IDF framework version V5.0

So, basically you are testing two different versions of framework? Does Windows 10 VS with IDF v5.0 work?

RitaAJ, 2023-02-22:
Thanks Valeros for your reply . 
On windows 11 latest version is recommended  when we install esp32S3 in platform io that's why we are having two different versions on two different operating systems. 
After running command you have given I am getting the below error.PS C:\Users\admin> pio run -t menuconfig
pio : The term 'pio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was 
included, verify that the path is correct and try again.
At line:1 char:1
+ pio run -t menuconfig
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

PS C:\Users\admin>

valeros, 2023-02-22:
> On windows 11 latest version is recommended when we install esp32S3 in platform io that's why we are having two different versions on two different operating systems.

Where did you see that recommendation ? Please try Windows 10 VS with IDF v5.0.

> After running command you have given I am getting the below error.PS C:\Users\admin> pio run -t menuconfig

You need to run that command directly in the IDE terminal, see https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli

closed 2023-04-02 by stale[bot]