platformio/platform-espressif32 #1034: ESP32-S3-DevKitC-1 Build Issue

mtt39, 2023-02-17:
I am having issues being able to build for this board using platformio, but have no issues using the Arduino IDE. Attached is the terminal output with the error at the bottom. I have come across some older issues that produce a similar issue, but was unable to fid any solutions. Any help is much appreciated

Processing esp32-s3-devkitc-1 (platform: espressif32@2.0.0; board: esp32-s3-devkitc-1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------Platform Manager: Installing espressif32 @ 2.0.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@2.0.0 has been installed!
Error: Unknown board ID 'esp32-s3-devkitc-1'

 *  The terminal process "C:\Users\mtoplack\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task: C:\Users\mtoplack\.platformio\penv\Scripts\platformio.exe run --target upload 

Processing esp32-s3-devkitc-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.0.1) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32-s3-devkitc-1\src\main.cpp.o
Building .pio\build\esp32-s3-devkitc-1\bootloader.bin
Generating partitions .pio\build\esp32-s3-devkitc-1\partitions.bin
esptool.py v4.4
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\USB.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\USBCDC.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\USBMSC.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\WString.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\base64.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-adc.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-bt.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-cpu.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-dac.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-gpio.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-i2c-slave.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-i2c.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-ledc.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-matrix.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-misc.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-psram.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-rgb-led.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-tinyusb.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\main.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32-s3-devkitc-1\libFrameworkArduino.a
Linking .pio\build\esp32-s3-devkitc-1\firmware.elf
c:/users/mtoplack/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot find  
Inc\Documents\PlatformIO\Projects\Darko Embeded Test\.pio\build\esp32-s3-devkitc-1\firmware.map: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-s3-devkitc-1\firmware.elf] Error 1
================================================================= [FAILED] Took 21.92 seconds =================================================================

Jason2866, 2023-02-17:
Delete the folder `c:/users/mtoplack/.platformio/` Platformio will rebuild.
Probably you have old stuff there which generates this issue.
Use `platform = espressif32 @ 6.0.1`

mtt39, 2023-02-17:
I appreciate the reply. Unfortunately, after following these suggestions, I still have the same issue with the exact same output

Jason2866, 2023-02-17:
Please post your platformio.ini

mtt39, 2023-02-17:
As requested

`[env:esp32-s3-devkitc-1]
platform = espressif32 @ 6.0.1
board = esp32-s3-devkitc-1
framework = arduino
`

Jason2866, 2023-02-17:
Okay. This looks strange `Inc\Documents\PlatformIO\Projects\Darko Embeded Test.pio\build\esp32-s3-devkitc-1\firmware.map: `
Looks like whitespaces in path

mtt39, 2023-02-17:
Ah, thank you there were actually two white space issues, first with the project name and also since this computer is set up with OneDrive. I have corrected both of these issues and now my project builds successfully.

closed 2023-02-17 by mtt39