platformio/platform-espressif32 #1035: Platform cant push code to use serial.read()

Treyglass726, 2023-02-17:
I am having trouble getting serial.read() to work on platform io.

`
#include<Arduino.h>
int led = LED_BUILTIN;

void setup() {
  // Some boards work best if we also make a serial connection
  Serial.begin(115200);

  // set LED to be an output pin
  pinMode(led, OUTPUT);
}

void loop() {
  // Say hi!
  Serial.println("Hello!");
  
  digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(500);                // wait for a half second
  digitalWrite(led, LOW);    // turn the LED off by making the voltage LOW
  delay(500);                // wait for a half second

}`

this code works when uploaded with the Arduino ide will work and you can use both the Arduino serial monitor and the platform ide monitor to see the "hello!". 

when uploaded with platformio the light still flashes but neither serial monitor can read it.

valeros, 2023-02-20:
Hi @Treyglass726 ! Please provide more info. What board are you using? What version of the development platform is installed?

Treyglass726, 2023-02-20:
> 

Using the ESP32-S3-DevkitC-1

using vs code 1.75.1 with platform io 6.1.6

valeros, 2023-02-20:
> using vs code 1.75.1 with platform io 6.1.6

I meant the version of the installed `espressif32` development platform.

closed 2023-04-02 by stale[bot]