platformio/platform-espressif32 #1036: Manually modify sdkconfig

marchingband, 2023-02-19:
I understand that the standard way to modify sdkconfig is via menuconfig.
However I would like to copy/paste my sdkconfig from another project, and it is extremely difficult to do so manually via menuconfig. Things are all named a little different in menuconfig compared to the file, and often many menus deep.

Is there any way to just modify the sdkconfig file directly, and have pio respect the changes, rather than overwrite them?

thank you!

Jason2866, 2023-02-19:
Copy the settings you want in `sdkconfig.defaults`

marchingband, 2023-02-19:
oh amazing. thank you!
Is there a good place I could make a PR to add this to the docs?

marchingband, 2023-02-20:
@Jason2866 your solution does not seem to work. Are there additional steps needed to have pio respect the sdkconfig.defaults file?

Jason2866, 2023-02-20:
Nothing else needed. See the examples https://github.com/platformio/platform-espressif32/tree/develop/examples

valeros, 2023-02-20:
@marchingband PlatformIO uses the IDF build system behind the scenes, so the IDF documentation is relevant here as well  https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#using-sdkconfig-defaults
To apply the settings from your default configuration just clean your project, remove the autogenerated `sdkconfig.YOURENVNAME` file in the root folder of your project and run again.

closed 2023-02-20 by valeros

marchingband, 2023-02-20:
Thank you @valeros the extra step of running clean and deleting the current file is what I needed. The docs you linked mention 'idf.py save-defconfig' is this command available via pio? It would be very helpful for me.

valeros, 2023-02-20:
> is this command available via pio? It would be very helpful for me

Unfortunately, it's not available.

tanchakra, 2023-06-14:
hi,

how to set GPIO's foe any of the interface from sdkconfig or do i need to set from anywhere?
Kind Regards,
Tanuja

Nick-Miras, 2024-05-14:
I have read ESP-IDF documentation about `sdkconfig.defaults`. I have deleted the automatically generated sdkconfig.`env_name` file and executed `pio run -t clean -e env_name` before executing `pio run -e env_name`. 

The compiler still throws an error regarding the configurations I've set in `sdkconfig.defaults`.
```
src/main.c: In function 'someFunction':
src/main.c:line:column: error: 'CONFIG_SOME_CONFIG_HERE' undeclared (first use in this function);
```

After the run fails. The deleted autogenerated `sdkconfig.env_name` file is regenerated again.