platformio/platform-espressif32 #1037: How to use a different Python for esptool.py?

probonopd, 2023-02-19:
Since I am running PlatformIO on FreeBSD and this core (and others) are not available for FreeBSD, I am using a venv with Linux Python. This works exceptionally well for the complication, but uploading fails because the Python inside the venv does not have access to the serial port.

So I need to run esptool.py with another Python (outside of the venv).

How can I achieve this?

Jason2866, 2023-02-19:
Have you tried to install Platformio this way https://docs.platformio.org/en/latest/core/installation/integration.html
It should use the Python version from the system.
With VSC it is possible to specify the Python version which is used for Platformio.
Platformio can be run without a pyenv.

probonopd, 2023-02-19:
Thing is, I want platformio inside the venv for everything except for esptool.py.

Jason2866, 2023-02-19:
You like it complicated ;-)  Why not using the system installed Python Version for Platformio?

probonopd, 2023-02-19:
> You like it complicated ;-) Why not using the system installed Python Version for Platformio?

My system is FreeBSD. If I use FreeBSD Python, it wants to download the Espressif cores for FreeBSD, which fails because they don't exist. So as I wrote, I am using a Linux version of Python in the venv.

But for uploading I must not use the Python from the venv, but the one from outside the venv. And for this I am looking for a way to get this accomplished.

Jason2866, 2023-02-19:
> My system is FreeBSD. If I use FreeBSD Python, it wants to download the Espressif cores for FreeBSD, which fails because they don't exist.

This part is the more interesting one (for me at least). What do you mean with cores? 
There is just IDF and Arduino core. The are same for all OS. What does not exists, or fail to install?

Jason2866, 2023-02-19:
@probonopd Here is an example how you can override the default uploader 
https://github.com/arendst/Tasmota/blob/development/pio-tools/http-uploader.py

probonopd, 2023-02-19:
> What do you mean with cores?
There is just IDF and Arduino core. The are same for all OS. What does not exists, or fail to install?

Sorry, I meant the toolchains for the cores in the various Espessif processors.

Jason2866, 2023-02-19:
Okay which processor type/architecture?

probonopd, 2023-02-19:
```
% pio run
Processing esp32 (platform: espressif32@6.0.1; board: esp32dev; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5' requirements for your system 'freebsd_amd64'
```

probonopd, 2023-02-19:
So the culprit seems to be that `$PYTHONEXE` is populated, hence disregarding the shebang that is set in `esptool.py`. However I did not find a way to change it.

Jason2866, 2023-02-19:
> So the culprit seems to be that `$PYTHONEXE` is populated somewhere, hence disregarding the shebang that is set in `esptool.py`. However I did not find a way to change it.

Yes, finding a way to override `$PYTHONEXE` would be the easiest way to solve?
Did not mentioned, since i have no clue how to do.

Jason2866, 2023-02-19:
> E.g.,
> 
> ```
> % pio run
> Processing esp32 (platform: espressif32@6.0.1; board: esp32dev; framework: arduino)
> -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
> Error: Could not find the package with 'espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5' requirements for your system 'freebsd_amd64'
> ```
> 
> Possibly also for the others, like the riscv5 based ones.

Is there any special setup needed, or would it work "just" to accept the corresponding arch Linux toolchains?

valeros, 2023-02-20:
Hi @probonopd ! I'm wondering how it works with Python outside your virtual environment. Is that Python installed globally with root privileges? Have you updated `udev.rules` for your board? Anyway, although it's not recommended and may lead to some issues, it should be possible to override the default Python interpreter via the `PYTHONEXEPATH` environment variable.

probonopd, 2023-02-20:
Thanks @valeros, 

```
env PYTHONEXEPATH=/usr/local/bin/python3 pio run -e esp32cam -t upload
```

will allow me to use the system-provided Python, but it will do so not only for `esptool.py` but also for the compilation step. However, this will fail on my FreeBSD system because the system-provided Python has no access to a suitable compiler.

What I need is to use a different Python just for `esptool.py` while leaving the Python for everything else unchanged.

On Unix, there is normally an elegant mechanism for this, called "shebang". It is the first line in a Python script that tells the system which Python executable to use. If it is set to `#!/usr/bin/env python3`, then whatever `python3` is on the `$PATH` should be used (in my case, that would be the one in the virtualenv). If it is set to `#!/usr/local/bin/python3`, then that `python3`(in my case, the system-provided one) should be used. This is how it should work on Unix-based systems.

Unforatunately, PlatformIO seems to be completely ignoring it.

probonopd, 2023-02-20:
@Jason2866 

> Is there any special setup needed, or would it work "just" to accept the corresponding arch Linux toolchains?

If the user has properly set up the Linuxulator, then running Linux binaries on FreeBSD will mostly "just work" like they were FreeBSD binaries. (It's not an emulation, and it's not slow - think of it as the "Linux subsystem for FreeBSD"; just some additional syscalls to support Linux ELFs.)

valeros, 2023-02-21:
@probonopd 

> However, this will fail on my FreeBSD system because the system-provided Python has no access to a suitable compiler (sh: xtensa-esp32-elf-g++: not found)

TBH, there is something off with your setup, but if you know what you're doing, then you can override the python interpreter only for the upload command via an extra script:
```ini
[env:esp32dev]
platform = espressif32
framework = arduino
board = esp32dev
extra_scripts =
    replace_python.py
```
Where `replace_python.py` is the following:
```python
Import("env")

env.Replace(
    UPLOADCMD=env.get("UPLOADCMD").replace("$PYTHONEXE", "/path/to/your/python")
)
```

probonopd, 2023-03-25:
Thanks @valeros. Thanks to your help I figured out that instead of making this change in every one of my projects, I can edit `.platformio/platforms/espressif32/builder/main.py` and append the code at the end of that file. However, the original value of the `$UPLOADER` variable is `"$UPLOADER" $UPLOADERFLAGS $ESP32_APP_OFFSET $SOURCE`, so trying to replace `$PYTHONEXE` there does not work.

In fact, the whole replacement seems to be without effect, since if I use

```
Import("env")

print("Original UPLOADCMD:")
print(env.get("UPLOADCMD"))

env.Replace(
    UPLOADCMD=env.get("UPLOADCMD").replace(
        '"$UPLOADER"', "echo 'Hello world!'")
)

print("New UPLOADCMD:")
print(env.get("UPLOADCMD"))
```

It outputs

```
Building in release mode
"$UPLOADER" $UPLOADERFLAGS $ESP32_APP_OFFSET $SOURCE
Original UPLOADCMD:
"$UPLOADER" $UPLOADERFLAGS $ESP32_APP_OFFSET $SOURCE
New UPLOADCMD:
echo 'Hello world!' $UPLOADERFLAGS $ESP32_APP_OFFSET $SOURCE
Retrieving maximum program size .pio/build/esp32/firmware.elf
Checking size .pio/build/esp32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.6% (used 67364 bytes from 327680 bytes)
Flash: [=======   ]  68.5% (used 1346213 bytes from 1966080 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...

Warning! Please install `99-platformio-udev.rules`. 
More details: https://docs.platformio.org/en/latest/core/installation/udev-rules.html

Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
=========================================================================== [FAILED] Took 7.78 seconds ===========================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:07.776
```
So my code is clearly running and is changing the variable, but the actual uploadtool still is unchanged.

Jason2866, 2023-03-25:
It is doing what to expect 
```
Original UPLOADCMD:
"$UPLOADER" $UPLOADERFLAGS $ESP32_APP_OFFSET $SOURCE
New UPLOADCMD:
echo 'Hello world!' $UPLOADERFLAGS $ESP32_APP_OFFSET $SOURCE
```
`$Uploader` is replaced with `Hello World!`

probonopd, 2023-03-25:
What is printing `Looking for upload port...` and the subsequent lines then?

Jason2866, 2023-03-25:
That's the output of the Autodetect port function

probonopd, 2023-03-25:
Uh. Since my virtualenv in which I am running pio doesn't have access to serial ports, I also need to get this to run outside of the venv.

probonopd, 2023-04-26:
[image]

> This issue has been automatically marked as stale because it has not had recent activity. Please provide more details or it will be closed if no further activity occurs. Thank you for your contributions.

An issue is solved when it is solved.

Not when some bot sees that it "had no recent activity".

What is the purpose of this annoying bot? Annoy people who take the time to write feature requests and bug reports?

probonopd, 2023-06-18:
> This issue has been automatically marked as stale because it has not had recent activity. Please provide more details or it will be closed if no further activity occurs. Thank you for your contributions.

This issue persists.

probonopd, 2023-06-18:
Please reopen!

probonopd, 2023-08-10:
@stale is counterproductive to getting things done.

probonopd, 2025-04-25:
https://github.com/Stale is still counterproductive to getting things done.

probonopd, 2025-04-25:
So now this is closed without being resolved?