platformio/platform-espressif32 #1042: ESP32-S2 compile error: tusb.h: No such file or directory

nplan, 2023-02-22:
I'm having trouble compiling a project that requires USB CDC support. Using `espidf` framework with `arduino` as a component.

I modified the `espidf-arduino-blink` example to showcase the issue.

Here is my `platformio.ini` file:

```ini
[env]
platform = espressif32@6.0.0
framework = arduino, espidf
build_flags =
    -D CONFIG_BLINK_GPIO=2
    -DCONFIG_TINYUSB_ENABLED   ; added by me
    -DARDUINO_USB_CDC_ON_BOOT  ; added by me
monitor_speed = 115200

[env:esp32s2]
board = esp32-s2-kaluga-1
```

And `sdkconfig.defaults`:
```

CONFIG_AUTOSTART_ARDUINO=y
# CONFIG_WS2812_LED_ENABLE is not set
CONFIG_FREERTOS_HZ=1000
CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y

# below added by me
CONFIG_TINYUSB=y
CONFIG_TINYUSB_CDC_ENABLED=y
CONFIG_TINYUSB_CDC_RX_BUFSIZE=64
CONFIG_TINYUSB_CDC_TX_BUFSIZE=64
```

The error:

```

In file included from /Users/nejc/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-tinyusb.c:36:
/Users/nejc/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-tinyusb.h:24:10: fatal error: tusb.h: No such file or directory
```

Jason2866, 2023-02-22:
The IDF tinyusb component is missing in this setup. It is not part of IDF.
You have to add this component. See https://github.com/espressif/esp32-arduino-lib-builder/blob/master/tools/update-components.sh#L12
and https://github.com/espressif/esp32-arduino-lib-builder/tree/master/components

This is not an Platformio issue.

valeros, 2023-02-23:
Hi @nplan ! @Jason2866 correctly commented that you need to add the [esp_tinyusb](https://github.com/espressif/idf-extra-components/tree/master/usb/esp_tinyusb) component to you project. More info on adding standalone IDF components to PlatformIO project can be found [here](https://docs.platformio.org/en/latest/frameworks/espidf.html#esp-idf-components).

closed 2023-02-23 by valeros

nplan, 2023-02-23:
Sorry, I'm new to esp-idf.

I see the tinyusb library is already located on my system here:
`/Users/nejc/.platformio/packages/framework-espidf@3.40403.0/components/tinyusb`

Why is it not picked up? Can I point to this one somehow?

Jason2866, 2023-03-01:
@nplan  [this](https://github.com/Jason2866/Arduino_IDF_TinyUSB) setup should work. Not tested with a real tinyusb example

nplan, 2023-03-04:
@Jason2866 it works. Thanks a lot!