platformio/platform-espressif32 #1045: Project Initialization fails during Dependency Check (ESP32 Dev Module)

denmuellner, 2023-02-24:


valeros, 2023-02-27:
Hi @denmuellner ! Strange, but I cannot reproduce the problem. For some reason `pip` installed binaries for `ARM64` while Python modules are expecting shared libraries for `x86_64`. What's your host machine? Are you on a Mac with the M1 or M2 processor?

denmuellner, 2023-02-28:
Yeah I'm on a M1 based machine. I actually found an interesting workaround: I switched on the option to start VS Code using `Rosetta`. PlatformIO was re-installed upon startup and I was able to build the blank project successfully. However, I got some warnings about compatibility of the `CMake` extension and wasn't able to update `PlatformIO` to the latest version. I then switched off the `Rosetta` option again, restarted VS Code and after everything was initialized, building the project worked fine. Probably switching on Rosetta triggered a different script which installed the missing `ARM64` libs.

aapav01, 2023-03-05:
I also have the same issue. It started happening when I updated from 5.x to 6.0.1. I tried doing a fresh installation and the problem still exists. The issue is not limited to esp32dev but also esp32cam.
I'm using Apple M1 pro chip with vscode.

zapta, 2023-03-06:
I had similar issue when installing vscode/platformio/espidf on a fresh Apple M1 computer.  More details here https://community.platformio.org/t/error-running-platformio-espidf-on-apple-m1-mac/32403

aapav01, 2023-03-06:
> I had similar issue when installing vscode/platformio/espidf on a fresh Apple M1 computer. More details here https://community.platformio.org/t/error-running-platformio-espidf-on-apple-m1-mac/32403

I did figure out that... it's working (copy, paste) but the issue is the 'charset_normalizer' package which doesn't support arm. and I guess platformio uses this package to decode messages for debugging and the method you mentioned in the link that package is not even being installed by the platfromio core as we are copying pasting 'chardet' package which I think is different as 'charset_normalizer' has more features. But I'm still confused about how this package worked before the update as I tried forcing a lower version espressif32@5.30. I think it is more over the issue with the 'charset_normalizer' than platfromio core actually.

Jason2866, 2023-03-07:
What happens when a alternative python version is installed with `brew`?
Never had this issue (use the brew python version) with my Mac M1 Air

tomkuzma, 2023-04-20:
this is still an ongoing issue for me.  output from trying to import an example IDF project:

Could not import project
PIO Core Call Error: "Resolving esp32dev dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...

Error: Processing esp32dev (board: esp32dev; platform: espressif32; framework: espidf)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.0.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50000.0 (5.0.0) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 11.2.0+2022r1
Reading CMake configuration...
-- Found Git: /opt/local/bin/git (found version \"2.39.1\") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/tomkuzma/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/tomkuzma/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/tomkuzma/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Users/tomkuzma/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /Users/tomkuzma/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also \"/Users/tomkuzma/Documents/PlatformIO/Projects/230419-224225-espidf-hello-world/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log\".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /Users/tomkuzma/.platformio/packages/framework-espidf/tools/cmake/build.cmake:519 (message):
 Traceback (most recent call last):

 File \"/Users/tomkuzma/.platformio/penv/.espidf-5.0.0/lib/python3.10/site-packages/requests/compat.py\", line 11, in <module>
 import chardet

 ModuleNotFoundError: No module named 'chardet'

 

 During handling of the above exception, another exception occurred:

 

 Traceback (most recent call last):

 File \"/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/runpy.py\", line 196, in _run_module_as_main
 return _run_code(code, main_globals, None,
 File \"/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/runpy.py\", line 86, in _run_code
 exec(code, run_globals)
 File \"/Users/tomkuzma/.platformio/penv/.espidf-5.0.0/lib/python3.10/site-packages/idf_component_manager/prepare_components/__main__.py\", line 4, in <module>
 from .prepare import main
 File \"/Users/tomkuzma/.platformio/penv/.espidf-5.0.0/lib/python3.10/site-packages/idf_component_manager/prepare_components/prepare.py\", line 18, in <module>
 from ..core import ComponentManager
 File \"/Users/tomkuzma/.platformio/penv/.espidf-5.0.0/lib/python3.10/site-packages/idf_component_manager/core.py\", line 17, in <module>
[... 11 lines cut ...]
 File \"/Users/tomkuzma/.platformio/penv/.espidf-5.0.0/lib/python3.10/site-packages/charset_normalizer/cd.py\", line 9, in <module>
 from .md import is_suspiciously_successive_range

 ImportError:
 dlopen(/Users/tomkuzma/.platformio/penv/.espidf-5.0.0/lib/python3.10/site-packages/charset_normalizer/md.cpython-310-darwin.so,
 0x0002): tried:
 '/Users/tomkuzma/.platformio/penv/.espidf-5.0.0/lib/python3.10/site-packages/charset_normalizer/md.cpython-310-darwin.so'
 (mach-o file, but is an incompatible architecture (have 'arm64', need
 'x86_64')),
 '/System/Volumes/Preboot/Cryptexes/OS/Users/tomkuzma/.platformio/penv/.espidf-5.0.0/lib/python3.10/site-packages/charset_normalizer/md.cpython-310-darwin.so'
 (no such file),
 '/Users/tomkuzma/.platformio/penv/.espidf-5.0.0/lib/python3.10/site-packages/charset_normalizer/md.cpython-310-darwin.so'
 (mach-o file, but is an incompatible architecture (have 'arm64', need
 'x86_64'))

Call Stack (most recent call first):
 /Users/tomkuzma/.platformio/packages/framework-espidf/tools/cmake/project.cmake:440 (idf_build_process)
 CMakeLists.txt:6 (project)

========================== [FAILED] Took 1.52 seconds ==========================

Environment Status Duration
------------- -------- ------------
esp32dev FAILED 00:00:01.519
==================== 1 failed, 0 succeeded in 00:00:01.519 ===================="

tomkuzma, 2023-04-20:
just following up that the fix described here: https://community.platformio.org/t/error-running-platformio-espidf-on-apple-m1-mac/32403/3

where the chardet lib needs to be copied from 

.platformio/penv/.espidf-5.0.0/lib/python3.10/site-packages/pip/_vendor/chardet 

to: 

.platformio/penv/.espidf-5.0.0/lib/python3.10/site-packages

worked for me.

rashedtalukder, 2023-05-21:
Not stale. Still an annoying issue.

valeros, 2023-05-22:
Could anyone experiencing this issue run the following command in the IDE terminal?
```bash
/Users/USER/.platformio/penv/.espidf-5.0.0/bin/pip debug -v
```

rashedtalukder, 2023-05-22:
The script to download/install the espidf never executed. So that command for any platform version results in a file not found.

rashedtalukder, 2023-05-22:
That path is incorrect for me since there is no `.espidf-5.x.x` folder. The correct path for me is:
`~/.platformio/penv/bin/pip debug -v`

```
pip version: pip 23.1.2 from /Users/user/.platformio/penv/lib/python3.9/site-packages/pip (python 3.9)
sys.version: 3.9.6 (default, Mar 10 2023, 20:16:38) 
[Clang 14.0.3 (clang-1403.0.22.14.1)]
sys.executable: /Users/user/.platformio/penv/bin/python
sys.getdefaultencoding: utf-8
sys.getfilesystemencoding: utf-8
locale.getpreferredencoding: UTF-8
sys.platform: darwin
sys.implementation:
  name: cpython
'cert' config value: global
REQUESTS_CA_BUNDLE: None
CURL_CA_BUNDLE: None
pip._vendor.certifi.where(): /Users/user/.platformio/penv/lib/python3.9/site-packages/pip/_vendor/certifi/cacert.pem
pip._vendor.DEBUNDLED: False
vendored library versions:
  CacheControl==0.12.11
  colorama==0.4.6
  distlib==0.3.6
  distro==1.8.0
  msgpack==1.0.5
  packaging==21.3
  platformdirs==3.2.0
  pyparsing==3.0.9
  pyproject-hooks==1.0.0
  requests==2.28.2
  certifi==2022.12.07
  chardet==5.1.0
  idna==3.4
  urllib3==1.26.15
  rich==13.3.3 (Unable to locate actual module version, using vendor.txt specified version)
  pygments==2.14.0
  typing_extensions==4.5.0 (Unable to locate actual module version, using vendor.txt specified version)
  resolvelib==1.0.1
  setuptools==67.7.2 (Unable to locate actual module version, using vendor.txt specified version)
  six==1.16.0
  tenacity==8.2.2 (Unable to locate actual module version, using vendor.txt specified version)
  tomli==2.0.1
  webencodings==0.5.1 (Unable to locate actual module version, using vendor.txt specified version)
Compatible tags: 411
  cp39-cp39-macosx_13_0_arm64
  cp39-cp39-macosx_13_0_universal2
  cp39-cp39-macosx_12_0_arm64
  cp39-cp39-macosx_12_0_universal2
  cp39-cp39-macosx_11_0_arm64
  cp39-cp39-macosx_11_0_universal2
  cp39-cp39-macosx_10_16_universal2
  cp39-cp39-macosx_10_15_universal2
  cp39-cp39-macosx_10_14_universal2
  cp39-cp39-macosx_10_13_universal2
  cp39-cp39-macosx_10_12_universal2
  cp39-cp39-macosx_10_11_universal2
  cp39-cp39-macosx_10_10_universal2
  cp39-cp39-macosx_10_9_universal2
  cp39-cp39-macosx_10_8_universal2
  cp39-cp39-macosx_10_7_universal2
  cp39-cp39-macosx_10_6_universal2
  cp39-cp39-macosx_10_5_universal2
  cp39-cp39-macosx_10_4_universal2
  cp39-abi3-macosx_13_0_arm64
  cp39-abi3-macosx_13_0_universal2
  cp39-abi3-macosx_12_0_arm64
  cp39-abi3-macosx_12_0_universal2
  cp39-abi3-macosx_11_0_arm64
  cp39-abi3-macosx_11_0_universal2
  cp39-abi3-macosx_10_16_universal2
[... 361 lines cut ...]
  py30-none-macosx_10_15_universal2
  py30-none-macosx_10_14_universal2
  py30-none-macosx_10_13_universal2
  py30-none-macosx_10_12_universal2
  py30-none-macosx_10_11_universal2
  py30-none-macosx_10_10_universal2
  py30-none-macosx_10_9_universal2
  py30-none-macosx_10_8_universal2
  py30-none-macosx_10_7_universal2
  py30-none-macosx_10_6_universal2
  py30-none-macosx_10_5_universal2
  py30-none-macosx_10_4_universal2
  cp39-none-any
  py39-none-any
  py3-none-any
  py38-none-any
  py37-none-any
  py36-none-any
  py35-none-any
  py34-none-any
  py33-none-any
  py32-none-any
  py31-none-any
  py30-none-any
```

valeros, 2023-05-23:
@rashedtalukder That's strange, a new virtual env should be installed when you compile an IDF project for the first time. Could you please run any IDF project and share logs here?

gomiles, 2023-05-27:
I have the same error, anything I can help with?

valeros, 2023-05-29:
Hi @gomiles, what folders do you see in `~/.platformio/penv`. Also please run `~/.platformio/penv/bin/pip debug -v`

gomiles, 2023-05-29:
@valeros 
```
➜  penv pwd
/Users/mdelwig/.platformio/penv
➜  penv ll
total 24
drwxr-xr-x@ 24 mdelwig  staff   768B May 27 21:16 bin
drwxr-xr-x@  2 mdelwig  staff    64B May 27 21:15 include
drwxr-xr-x@  3 mdelwig  staff    96B May 27 21:15 lib
-rw-r--r--@  1 mdelwig  staff    16B May 27 21:15 pip.conf
-rw-r--r--@  1 mdelwig  staff   104B May 27 21:15 pyvenv.cfg
-rw-r--r--@  1 mdelwig  staff   261B May 27 21:34 state.json
```

```
➜  penv ~/.platformio/penv/bin/pip debug -v
WARNING: This command is only meant for debugging. Do not use this with automation for parsing and getting these details, since the output and options of this command may change without notice.
pip version: pip 23.1.2 from /Users/mdelwig/.platformio/penv/lib/python3.9/site-packages/pip (python 3.9)
sys.version: 3.9.6 (default, Mar 10 2023, 20:16:38) 
[Clang 14.0.3 (clang-1403.0.22.14.1)]
sys.executable: /Users/mdelwig/.platformio/penv/bin/python
sys.getdefaultencoding: utf-8
sys.getfilesystemencoding: utf-8
locale.getpreferredencoding: UTF-8
sys.platform: darwin
sys.implementation:
  name: cpython
'cert' config value: global
REQUESTS_CA_BUNDLE: None
CURL_CA_BUNDLE: None
pip._vendor.certifi.where(): /Users/mdelwig/.platformio/penv/lib/python3.9/site-packages/pip/_vendor/certifi/cacert.pem
pip._vendor.DEBUNDLED: False
vendored library versions:
  CacheControl==0.12.11
  colorama==0.4.6
  distlib==0.3.6
  distro==1.8.0
  msgpack==1.0.5
  packaging==21.3
  platformdirs==3.2.0
  pyparsing==3.0.9
  pyproject-hooks==1.0.0
  requests==2.28.2
  certifi==2022.12.07
  chardet==5.1.0
  idna==3.4
  urllib3==1.26.15
  rich==13.3.3 (Unable to locate actual module version, using vendor.txt specified version)
  pygments==2.14.0
  typing_extensions==4.5.0 (Unable to locate actual module version, using vendor.txt specified version)
  resolvelib==1.0.1
  setuptools==67.7.2 (Unable to locate actual module version, using vendor.txt specified version)
  six==1.16.0
  tenacity==8.2.2 (Unable to locate actual module version, using vendor.txt specified version)
  tomli==2.0.1
  webencodings==0.5.1 (Unable to locate actual module version, using vendor.txt specified version)
Compatible tags: 411
  cp39-cp39-macosx_13_0_arm64
  cp39-cp39-macosx_13_0_universal2
  cp39-cp39-macosx_12_0_arm64
  cp39-cp39-macosx_12_0_universal2
  cp39-cp39-macosx_11_0_arm64
  cp39-cp39-macosx_11_0_universal2
  cp39-cp39-macosx_10_16_universal2
  cp39-cp39-macosx_10_15_universal2
  cp39-cp39-macosx_10_14_universal2
  cp39-cp39-macosx_10_13_universal2
  cp39-cp39-macosx_10_12_universal2
  cp39-cp39-macosx_10_11_universal2
  cp39-cp39-macosx_10_10_universal2
[... 374 lines cut ...]
  py30-none-macosx_10_15_universal2
  py30-none-macosx_10_14_universal2
  py30-none-macosx_10_13_universal2
  py30-none-macosx_10_12_universal2
  py30-none-macosx_10_11_universal2
  py30-none-macosx_10_10_universal2
  py30-none-macosx_10_9_universal2
  py30-none-macosx_10_8_universal2
  py30-none-macosx_10_7_universal2
  py30-none-macosx_10_6_universal2
  py30-none-macosx_10_5_universal2
  py30-none-macosx_10_4_universal2
  cp39-none-any
  py39-none-any
  py3-none-any
  py38-none-any
  py37-none-any
  py36-none-any
  py35-none-any
  py34-none-any
  py33-none-any
  py32-none-any
  py31-none-any
  py30-none-any
```

valeros, 2023-05-29:
@gomiles thanks, could you please also try to run your project in verbose mode (`pio run -v`) and attach here the entire log with the error message?

valeros, 2023-05-29:
@gomiles There are several IDF examples in the platform folder, for example `~/.platformio/platforms/espressif32/examples/espidf-hello-world`

TravisJoe, 2023-06-18:
Based on some of the comments I ran this and it seemed to at least get me past this issue. 

`~/.platformio/penv/bin/pip install chardet`

Frankie-420, 2023-07-05:
I am pretty sure this is an issue is from platform IO using a seperate python install and failing to install the chardet library. Hence if you `pip install chardet` direct from terminal it will only install it to your default python, not the platform IO version.

I fixed the issue by installing chardet file and copying it across into this directory `~/.platformio/penv/lib/python3.10/site-packages`

rashedtalukder, 2023-08-10:
Not stale, still an issue. Chardet is not being installed in the virtual environment.

Jason2866, 2023-08-10:
Probably fixed when updating platformio core to development version.
There is a fix for "chardet" https://github.com/platformio/platformio-core/commit/9170eee6e465b0baba9977e0fe964689f0e667bb

rashedtalukder, 2023-08-10:
That would be the fix. Hopefully this gets merged in to the next patch release so we can close this out.

Thanks for pointing that out @Jason2866

hermanius, 2023-11-13:
It looks like I have the same problem with a clean install of Platform 6.1.11. It only installs esp-idf 4.4.5, while the release notes for 6.1.11 say it has been upgraded to esp-idf 5.1.1
`
--------------------------  ----------------------------------------------
PlatformIO Core             6.1.11
Python                      3.9.6-final.0
System Type                 darwin_arm64
Platform                    macOS-14.0
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /Users/hjansen/.platformio
PlatformIO Core Executable  /Users/hjansen/.platformio/penv/bin/platformio
Python Executable           /Users/hjansen/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       2
Tools & Toolchains          12
--------------------------  ----------------------------------------------`

And when building it shows:
` - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5`

hermanius, 2023-11-13:
My issue is solved (I don't even know wether it relates to the original issue)
In a combined Arduino/espidf framework project esp-idf 4.4.5 is used.
When using an only espidf project (no Arduino) esp-idf 5.5.1 is used.

zapta, 2023-11-13:
" Arduino/espidf framework project"

How do you set this kind of project? Is it just that the Arduino framework
already includes and exposes the espidf APIs?

hermanius, 2023-11-13:
" Arduino/espidf framework project"

In platform.ini:

...
platform = espressif32
framework = 
    espidf
    arduino
...

As I mentioned, apparently you don't get the latest esp-idf version.
On the upside you can mix and match Espressif code and Arduino code.

rashedtalukder, 2023-11-13:
@hermanius  did you post in the wrong issue? Your problem is not related to this and wouldn’t solve the chardet dependency issue

hermanius, 2023-11-13:
@rashedtalukder Yes, I'm afraid so. It looked like the same problem at first, but is not. Sorry about that.

closed 2025-04-25 by stale[bot]