platformio/platform-espressif32 #1049: espidf-http-request example getaddrinfo error

torntrousers, 2023-03-04:
I'm trying the https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-http-request example, but the domain name fails to resolve. Any hints on what the problem is? 

The code is unchanged, here's the console log:
```
 *  Executing task: C:\Users\tornt\.platformio\penv\Scripts\platformio.exe device monitor 

--- Terminal on COM8 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
␛[0;32mI (13) boot: ESP-IDF 5.0.1 2nd stage bootloader␛[0m
␛[0;32mI (13) boot: compile time 11:33:47␛[0m
␛[0;32mI (14) boot: chip revision: v3.0␛[0m
␛[0;32mI (15) boot.esp32: SPI Speed      : 40MHz␛[0m
␛[0;32mI (20) boot.esp32: SPI Mode       : DIO␛[0m
␛[0;32mI (25) boot.esp32: SPI Flash Size : 4MB␛[0m
␛[0;32mI (29) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (35) boot: Partition Table:␛[0m
␛[0;32mI (38) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (45) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (53) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (60) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (68) boot: End of partition table␛[0m
␛[0;32mI (72) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1eae8h (125672) map␛[0m
␛[0;32mI (126) esp_image: segment 1: paddr=0002eb10 vaddr=3ffb0000 size=01508h (  5384) load␛[0m
␛[0;32mI (128) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=7d72ch (513836) map␛[0m
␛[0;32mI (317) esp_image: segment 3: paddr=000ad754 vaddr=3ffb1508 size=01e5ch (  7772) load␛[0m
␛[0;32mI (321) esp_image: segment 4: paddr=000af5b8 vaddr=40080000 size=14c20h ( 85024) load␛[0m
␛[0;32mI (368) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (368) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (380) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (380) cpu_start: Starting app cpu, entry point is 0x40081240␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (396) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (397) cpu_start: cpu freq: 160000000 Hz␛[0m
␛[0;32mI (397) cpu_start: Application information:␛[0m
␛[0;32mI (401) cpu_start: Project name:     http-request␛[0m
␛[0;32mI (407) cpu_start: App version:      v6.0.1-11-g48c4226-dirty␛[0m
␛[0;32mI (413) cpu_start: Compile time:     Mar  4 2023 11:33:21␛[0m
␛[0;32mI (419) cpu_start: ELF file SHA256:  c95408b22e6b86a7...␛[0m
␛[0;32mI (425) cpu_start: ESP-IDF:          5.0.1␛[0m
␛[0;32mI (430) cpu_start: Min chip rev:     v0.0␛[0m
␛[0;32mI (435) cpu_start: Max chip rev:     v3.99 ␛[0m
␛[0;32mI (439) cpu_start: Chip rev:         v3.0␛[0m
␛[0;32mI (444) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (452) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM␛[0m
␛[0;32mI (457) heap_init: At 3FFB7488 len 00028B78 (162 KiB): DRAM␛[0m
␛[0;32mI (464) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (470) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (477) heap_init: At 40094C20 len 0000B3E0 (44 KiB): IRAM␛[0m
␛[0;32mI (484) spi_flash: detected chip: generic␛[0m
␛[0;32mI (487) spi_flash: flash io: dio␛[0m
␛[0;33mW (491) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.␛[0m
␛[0;32mI (506) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (554) example_connect: Start example_connect.␛[0m
I (564) wifi:wifi driver task: 3ffbefd8, prio:23, stack:6656, core=0
␛[0;32mI (564) system_api: Base MAC address is not set␛[0m
␛[0;32mI (564) system_api: read default base MAC address from EFUSE␛[0m
I (574) wifi:wifi firmware version: 17afb16
I (574) wifi:wifi certification version: v7.0
I (574) wifi:config NVS flash: enabled
I (574) wifi:config nano formating: disabled
I (584) wifi:Init data frame dynamic rx buffer num: 32
I (584) wifi:Init management frame dynamic rx buffer num: 32
I (594) wifi:Init management short buffer num: 32
I (594) wifi:Init dynamic tx buffer num: 32
I (604) wifi:Init static rx buffer size: 1600
I (604) wifi:Init static rx buffer num: 10
I (604) wifi:Init dynamic rx buffer num: 32
␛[0;32mI (614) wifi_init: rx ba win: 6␛[0m
␛[0;32mI (614) wifi_init: tcpip mbox: 32␛[0m
␛[0;32mI (624) wifi_init: udp mbox: 6␛[0m
␛[0;32mI (624) wifi_init: tcp mbox: 6␛[0m
␛[0;32mI (624) wifi_init: tcp tx win: 5744␛[0m
␛[0;32mI (634) wifi_init: tcp rx win: 5744␛[0m
␛[0;32mI (634) wifi_init: tcp mss: 1440␛[0m
␛[0;32mI (644) wifi_init: WiFi IRAM OP enabled␛[0m
␛[0;32mI (644) wifi_init: WiFi RX IRAM OP enabled␛[0m
␛[0;32mI (654) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07␛[0m
I (764) wifi:mode : sta (24:0a:c4:f8:7e:ec)
I (764) wifi:enable tsf
␛[0;32mI (764) example_connect: Connecting to BT-P2CKZW...␛[0m
␛[0;32mI (764) example_connect: Waiting for IP(s)␛[0m
I (3174) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (3864) wifi:state: init -> auth (b0)
I (3864) wifi:state: auth -> assoc (0)
I (3884) wifi:state: assoc -> run (10)
I (3924) wifi:connected with BT-P2CKZW, aid = 19, channel 11, BW20, bssid = ac:84:c9:b7:8f:82
I (3924) wifi:security: WPA2-PSK, phy: bgn, rssi: -65
I (3924) wifi:pm start, type: 1

I (4014) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (5434) wifi:<ba-add>idx:0 (ifx:0, ac:84:c9:b7:8f:82), tid:0, ssn:2, winSize:64
␛[0;32mI (5554) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:260a:c4ff:fef8:7eec, type: ESP_IP6_ADDR_IS_LINK_LOCAL␛[0m
␛[0;32mI (6424) esp_netif_handlers: example_netif_sta ip: 192.168.1.81, mask: 255.255.255.0, gw: 192.168.1.254␛[0m
␛[0;32mI (6424) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.1.81␛[0m    
␛[0;32mI (6434) example_common: Connected to example_netif_sta␛[0m
␛[0;32mI (6434) example_common: - IPv4 address: 192.168.1.81,␛[0m
␛[0;32mI (6444) example_common: - IPv6 address: fe80:0000:0000:0000:260a:c4ff:fef8:7eec, type: ESP_IP6_ADDR_IS_LINK_LOCAL␛[0m
␛[0;31mE (6484) example: DNS lookup failed err=202 res=0x0␛[0m
␛[0;31mE (7484) example: DNS lookup failed err=202 res=0x0␛[0m
␛[0;31mE (8484) example: DNS lookup failed err=202 res=0x0␛[0m
```

valeros, 2023-03-06:
Hi @torntrousers ! I'm not sure the issue has something to do with PlatformIO. I tried the `espidf-http-request` example on a few boards and they seem to work as expected. What's you hardware? Have you tried other WiFi networks?

torntrousers, 2023-03-06:
I was using an M5Stack Core2, but i've lots of other modules so i can go and try one of those. And I suppose i can try with a phone hotspot as an alternative Wifi. Will go try those ...

torntrousers, 2023-03-06:
I've now also tried with the Wifi from my Android phone hotspot and also with an ESP32-C3. All get the same "DNS lookup failed err=202 res=0x0"

valeros, 2023-03-07:
Could you please try a similar code but using the Arduino framework?

torntrousers, 2023-03-07:
It turns out change [this line](https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-http-request/src/http_request_example_main.c#L27) to use httpbin.org instead of example.com gets it working. Not sure I understand why though. But i guess i don't need to know.

closed 2023-03-07 by torntrousers