platformio/platform-espressif32 #1052: rainmaker simpleBLE provisioning not working on 6.0+

MingyaoLiu, 2023-03-10:
The rainmaker example for ESP32C3 (with Bluetooth provisioning) stopped working on 6.0+ version.

Last working version is 5.30, I suspect something to do with the arduino-esp32 2.0.7 version .

Tested in platformio and arduinoIDE.

Jason2866, 2023-03-10:
When it does not work in Arduino IDE too it is not a Platformio issue.

valeros, 2023-03-13:
Hi @MingyaoLiu ! It sounds like a breaking change in the Arduino core, please forward to https://github.com/espressif/arduino-esp32. 

You can also downgrade your platform version and continue to use the Arduino core v2.0.6:
```ini
[env:esp32dev]
platform = espressif32 @ ~6.0.0
framework = arduino
board = esp32dev
```

closed 2023-03-13 by valeros