platformio/platform-espressif32 #1053: Memory usage with Platformio and VS code

RitaAJ, 2023-03-15:
Hello, I am using ESP32S3 in VScode with Platform IO. In my code, I just included libraries required for WiFi and Bluetooth in my Project.My program code is also very small. When I build a project individually with only Bluetooth and only WiFI enabled, at that time compiler don't shows memory usage issue while building and everything goes fine. But when I select both bluetooth and WIFI functionality in my code at that time it shows the below error message while building the project.
RAM:   [==        ]  17.6% (used 57768 bytes from 327680 bytes)
Flash: [===Error: The program size (1313277 bytes) is greater than maximum allowed (1310720 bytes)
==*** [checkprogsize] Explicit exit, status 1
=====]  100.2% (used 1313277 bytes from 1310720 bytes)

What is the solution. ?

Jason2866, 2023-03-15:
You have to use a different partition scheme. You run out of space. Your firmware is too big to fit in the current used one

valeros, 2023-03-15:
Hi @RitaAJ ! Please use a partition table with bigger slot for your main application, for example `huge_app.csv`:

```ini
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
board_build.partitions = huge_app.csv
```

For the future, this repository is not the right place for such generic questions, please forward to https://community.platformio.org/

closed 2023-03-15 by valeros