platformio/platform-espressif32 #1057: openocd via mcu-link and cmsis-dap fails with error "3221225477"

jschoch, 2023-03-25:
I can connect and start a debug session from the openocd-esp32 github repo's most recent release.

The included version fails silently

output from verbose upload
```
Total                                                                               5767840<lambda>(["upload"], [".pio\build\esp\firmware.bin"])
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpaCURRENT: upload_protocol = cmsis-dap
openocd -d2 -s C:/Users/jesse/.platformio/packages/tool-openocd-esp32/share/openocd/scripts -f interface/cmsis-dap.cfg -f board/esp-wroom-32.cfg -c "adapter speed 1500" -c "program_esp {{.pio\build\esp\firmware.bin}} 0x10000 verify" -c "program_esp {{C:/Users/jesse/Documents/PlatformIO/Projects/steptest/stepper_testing_teensytest/.pio/build/esp/bootloader.bin}} 
0x1000 verify" -c "program_esp {{C:/Users/jesse/Documents/PlatformIO/Projects/steptest/stepper_testing_teensytest/.pio/build/esp/partitions.bin}} 0x8000 verify" -c "program_esp {{C:/Users/jesse/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin}} 0xe000 verify" -c "reset run; shutdown"
*** [upload] Error 3221225477
=============================== [FAILED] Took 2.96 seconds ===============================

 *  The terminal process "C:\Users\jesse\.platformio\penv\Scripts\platformio.exe 'run', '--verbose', '--target', 'upload', '--environment', 'esp'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.
```

output from manual run
```
PS C:\Users\jesse\.platformio\packages\tool-openocd-esp32> .\bin\openocd.exe -f .\share\openocd\scripts\interface\cmsis-dap.cfg --debug  -f .\share\openocd\scripts\target\esp32.cfg
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 3 3 options.c:63 configuration_output_handler(): debug_level: 3
User : 4 9 options.c:63 configuration_output_handler():
Debug: 5 10 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 6 13 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 7 16 options.c:246 add_default_dirs(): exepath=C:/Users/jesse/.platformio/packages/tool-openocd-esp32/bin
Debug: 8 20 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 9 22 configuration.c:44 add_script_search_dir(): adding C:/Users/jesse/AppData/Roaming/OpenOCD
Debug: 10 24 configuration.c:44 add_script_search_dir(): adding C:\Users\jesse/.config/openocd
Debug: 11 27 configuration.c:44 add_script_search_dir(): adding C:\Users\jesse/.openocd
Debug: 12 29 configuration.c:44 add_script_search_dir(): adding C:/Users/jesse/.platformio/packages/tool-openocd-esp32/bin/../share/openocd/site
Debug: 13 33 configuration.c:44 add_script_search_dir(): adding C:/Users/jesse/.platformio/packages/tool-openocd-esp32/bin/../share/openocd/scripts
Debug: 14 38 command.c:166 script_debug(): command - ocd_find .\share\openocd\scripts\interface\cmsis-dap.cfg
Debug: 15 41 configuration.c:99 find_file(): found .\share\openocd\scripts\interface\cmsis-dap.cfg
Debug: 16 44 command.c:166 script_debug(): command - adapter driver cmsis-dap
Debug: 17 46 command.c:166 script_debug(): command - cmsis_dap_backend auto
Debug: 18 48 command.c:166 script_debug(): command - transport select jtag
User : 19 50 options.c:63 configuration_output_handler(): jtagUser : 20 52 options.c:63 configuration_output_handler():
Debug: 21 54 command.c:166 script_debug(): command - ocd_find .\share\openocd\scripts\target\esp32.cfg
Debug: 22 56 configuration.c:99 find_file(): found .\share\openocd\scripts\target\esp32.cfg
Debug: 23 60 command.c:166 script_debug(): command - transport select jtag
Warn : 24 62 transport.c:287 jim_transport_select(): Transport "jtag" was already selected
Debug: 25 64 command.c:166 script_debug(): command - ocd_find target/esp_common.cfg
Debug: 26 66 configuration.c:99 find_file(): found C:/Users/jesse/.platformio/packages/tool-openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg
Debug: 27 73 command.c:166 script_debug(): command - add_help_text program_esp write an image to flash, address is only required for binary images. verify, reset, exit, compress, restore_clock are optional
Debug: 28 78 command.c:166 script_debug(): command - add_usage_text program_esp <filename> [address] [verify] [reset] [exit] [compress] [no_clock_boost] [restore_clock]
Debug: 29 82 command.c:166 script_debug(): command - add_help_text program_esp_bins write all the images at address specified in flasher_args.json generated while building idf project
Debug: 30 88 command.c:166 script_debug(): command - add_usage_text program_esp_bins <build_dir> flasher_args.json [verify] [reset] [exit] [compress] [no_clock_boost] [restore_clock]
Debug: 31 93 command.c:166 script_debug(): command - add_help_text esp_get_mac Print MAC address of the chip. Use a `format` argument to return formatted MAC value
Debug: 32 97 command.c:166 script_debug(): command - add_usage_text esp_get_mac [format]
Debug: 33 100 command.c:166 script_debug(): command - jtag newtap esp32 cpu0 -irlen 5 -expected-id 0x120034e5
Debug: 34 103 tcl.c:569 jim_newtap_cmd(): Creating New Tap, Chip: esp32, Tap: cpu0, Dotted: esp32.cpu0, 4 params
Debug: 35 106 tcl.c:593 jim_newtap_cmd(): Processing option: -irlen
Debug: 36 108 tcl.c:593 jim_newtap_cmd(): Processing option: -expected-id
Debug: 37 110 core.c:1472 jtag_tap_init(): Created Tap: esp32.cpu0 @ abs position 0, irlen 5, capture: 0x1 mask: 0x3
Debug: 38 113 command.c:166 script_debug(): command - jtag newtap esp32 cpu1 -irlen 5 -expected-id 0x120034e5
Debug: 39 117 tcl.c:569 jim_newtap_cmd(): Creating New Tap, Chip: esp32, Tap: cpu1, Dotted: esp32.cpu1, 4 params
Debug: 40 120 tcl.c:593 jim_newtap_cmd(): Processing option: -irlen
Debug: 41 121 tcl.c:593 jim_newtap_cmd(): Processing option: -expected-id
Debug: 42 123 core.c:1472 jtag_tap_init(): Created Tap: esp32.cpu1 @ abs position 1, irlen 5, capture: 0x1 mask: 0x3
Debug: 43 126 command.c:166 script_debug(): command - target create esp32.cpu0 esp32 -endian little -chain-position esp32.cpu0 -coreid 0 -rtos FreeRTOS
Debug: 44 130 target.c:2239 target_free_all_working_areas_restore(): freeing all working areas
Debug: 45 135 target.c:2239 target_free_all_working_areas_restore(): freeing all working areas
Debug: 46 137 FreeRTOS.c:1387 freertos_create(): freertos_create
Debug: 47 140 command.c:300 register_command(): command 'esp' is already registered
Debug: 48 142 command.c:300 register_command(): command 'esp32' is already registered
Debug: 49 145 command.c:300 register_command(): command 'esp32' is already registered
[... 296 lines cut ...]
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x1fc9:0x0143, serial=ITQNHXDMMTMH5
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: UART via USB COM port supported
Info : CMSIS-DAP: FW Version = 2.1.1
Info : CMSIS-DAP: Serial# = ITQNHXDMMTMH5
Info : CMSIS-DAP: Interface Initialised (JTAG)
Info : SWCLK/TCK = 0 SWDIO/TMS = 1 TDI = 0 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : cmsis-dap JTAG TLR_RESET
Info : cmsis-dap JTAG TLR_RESET
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32.cpu0] Target halted, PC=0x400F047A, debug_reason=00000001
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] Target halted, PC=0x400F047A, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
```

jschoch, 2023-03-27:
I was able to flash by using the latest openocd-esp32 release with the following command.  note the config if also changed, not sure if it is correct because i literally have no fucking clue what i'm doing and the deprication of esp-wroom...cfg is concerning.

```
.\bin\openocd.exe -d2 -s ./share/openocd/scripts -f interface/cmsis-dap2.cfg -f target/esp32.cfg -c "adapter speed 500" -c "program_esp {{C:/Users/jesse/Documents/PlatformIO/Projects/test_usb_bridge/.pio\build\death\firmware.bin}} 0x10000 verify" -c "program_esp {{C:/Users/jesse/Documents/PlatformIO/Projects/test_usb_bridge/.pio/build/death/bootloader.bin}} 0x1000 verify" -c "program_esp {{C:/Users/jesse/Documents/PlatformIO/Projects/test_usb_bridge/.pio/build/death/partitions.bin}} 0x8000 verify" -c "program_esp {{C:/Users/jesse/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin}} 0xe000 verify" -c "reset run; shutdown"
```

with the old scripts path it complained of these issues

```
PS C:\Users\jesse\Downloads\openocd-esp32-win32-0.12.0-esp32-20230313\openocd-esp32> .\bin\openocd.exe -d2 -s C:/Users/jesse/.platformio/packages/tool-openocd-esp32/share/openocd/scripts -f interface/cmsis-dap2.cfg -f target/esp32.cfg -c "adapter speed 500" -c "program_esp {{C:/Users/jesse/Documents/PlatformIO/Projects/test_usb_bridge/.pio\build\death\firmware.bin}} 0x10000 verify" -c "program_esp {{C:/Users/jesse/Documents/PlatformIO/Projects/test_usb_bridge/.pio/build/death/bootloader.bin}} 0x1000 verify" -c "program_esp {{C:/Users/jesse/Documents/PlatformIO/Projects/test_usb_bridge/.pio/build/death/partitions.bin}} 0x8000 verify" -c "program_esp {{C:/Users/jesse/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin}} 0xe000 verify" -c "reset run; shutdown"
Open On-Chip Debugger v0.12.0-esp32-20230313 (2023-03-13-09:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

jtag
Warn : Transport "jtag" was already selected
adapter speed: 500 kHz

Info : Using CMSIS-DAPv2 interface with VID:PID=0x1fc9:0x0143, serial=ITQNHXDMMTMH5
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: UART via USB COM port supported
Info : CMSIS-DAP: FW Version = 2.1.1
Info : CMSIS-DAP: Serial# = ITQNHXDMMTMH5
Info : CMSIS-DAP: Interface Initialised (JTAG)
Info : SWCLK/TCK = 0 SWDIO/TMS = 1 TDI = 0 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : cmsis-dap JTAG TLR_RESET
Info : cmsis-dap JTAG TLR_RESET
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Error: XTensa core not configured; is xtensa-core-openocd.cfg missing?
Warn : target esp32.cpu0 examination failed
Error: XTensa core not configured; is xtensa-core-openocd.cfg missing?
Warn : target esp32.cpu1 examination failed
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : cmsis-dap JTAG TLR_RESET
Info : cmsis-dap JTAG TLR_RESET
Info : cmsis-dap JTAG TLR_RESET
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Error: XTensa core not configured; is xtensa-core-openocd.cfg missing?
Error: XTensa core not configured; is xtensa-core-openocd.cfg missing?
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Error: Target not examined yet
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Error: Target not examined yet
Error: XTensa core not configured; is xtensa-core-openocd.cfg missing?
embedded:startup.tcl:1510: Error: ** Unable to reset target **
in procedure 'program_esp'
in procedure 'program_error' called at file "C:/Users/jesse/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 104
at file "embedded:startup.tcl", line 1510
```

the output from PIO is pretty useless

```
Executing task: C:\Users\jesse\.platformio\penv\Scripts\platformio.exe run --verbose --target upload --environment death 

Processing death (platform: espressif32; board: esp32dev; framework: arduino; debug_tool: cmsis-dap; upload_protocol: cmsis-dap; debug_speed: 500)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
[... 236 lines cut ...]
framework = arduino
debug_tool = cmsis-dap
upload_protocol = cmsis-dap
debug_speed = 500

```

main.cpp
```
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

int foo = 0;
int bar = 0;
void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("foo");
  delay(1000);
  foo++;
}
```

valeros, 2023-04-11:
Resolved in https://github.com/platformio/platform-espressif32/commit/2a74cfaca2f6a026c06f18946504a48b4895e694

closed 2023-04-11 by valeros

valeros, 2023-04-29:
I had to revert back to OpenOCD v0.11 as we're experiencing a strange issue with the v0.12 version described here https://github.com/espressif/openocd-esp32/issues/285

Note that the v0.12 packages are still available, so the latest version can be set directly in the `platformio.ini` file:
```ini
[env:esp32dev]
platform = espressif32
...
platform_packages = 
  tool-openocd-esp32 @ ~2.1200.0
```