platformio/platform-espressif32 #1059: After the transition of the arduino => esp-idf + arduino, there is a problem with SPIFFS, LIttleFS

butaikis, 2023-03-27:
Hello. After switching to `framework = esp-id, arduino ` libraries SPIFFS, LittleFS stopped working correctly. 
Concerns EspAsyncWebServer
Does not open files over http
`WebServer.serveStatic("/", FS_WebUI,"/").setDefaultFile("index.html");`

I described my problem in [detail here](https://github.com/me-no-dev/ESPAsyncWebServer/issues/1296)

Perhaps there is a solution to the problem ?

Jason2866, 2023-03-28:
For the not working LittleFS there is an explanation. The needed IDF source code is not there when using native IDF. You have to add this https://github.com/joltwallet/esp_littlefs as a Component.

butaikis, 2023-03-29:
Thank you @Jason2866 , yes I tried.

1. Created a folder in the root of the project /components and copied the contents of the [repository there](https://github.com/joltwallet/esp_littlefs).

2. In menuconfig Component config -> LittleFS it was displayed, everything is ok.

3. When building the project, he swears that he does not find the file
`C:/.platformio/packages/framework-arduinoespressif32/libraries/LittleFS/src/LittleFS.cpp:23:10: fatal error: esp_littlefs.h: No such file or directory`

I have specifically specified the full path to `components\esp_littlefs-master\include\esp_littles.h`

The project was assembled, but the result is the same. Maybe I'm doing something wrong?

butaikis, 2023-03-29:
@Jason2866  In your example, `serveStatic()` also doesn't work

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d5163  PS      : 0x00060330  A0      : 0x800d5282  A1      : 0x3ffc7b70  
A2      : 0x3ffb7794  A3      : 0x3ffc7fe8  A4      : 0x4015c580  A5      : 0x00000000  
A6      : 0x3ffb7794  A7      : 0x3ffc81f4  A8      : 0x800d51d6  A9      : 0x3ffc7b50  
A10     : 0x3ffc7fe8  A11     : 0x156bb5fc  A12     : 0x3ffc3bd0  A13     : 0x3ffc7f6e  
A14     : 0x0000000d  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x156bb60b  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe   

Backtrace: 0x400d5160:0x3ffc7b70 0x400d527f:0x3ffc7bd0 0x400d5385:0x3ffc7c10 0x400d5579:0x3ffc7c70 0x4014c635:0x3ffc7c90 0x4014c681:0x3ffc7cc0 0x4014c6c5:0x3ffc7ce0 0x4014c7e5:0x3ffc7d00 0x4014c8c2:0x3ffc7d20 0x4008be51:0x3ffc7d50
```

closed 2023-03-30 by butaikis

Jason2866, 2023-04-01:
@butaikis Updated with a real LittleFS test example https://github.com/Jason2866/Arduino_IDF_LittleFS
Verified working for ESP32, ESP32-S2, ESP32-S3 and ESP32-C3

butaikis, 2023-04-01:
@Jason2866 When working with ESPAsyncWebServer + serveStatic(), 
you need to change the sdk config. I left it [here](https://github.com/me-no-dev/ESPAsyncWebServer/issues/1296#issuecomment-1489801497).