platformio/platform-espressif32 #1063: Vim (neovim): clangd and ccls gives errors with esp32s3box

Tloe, 2023-03-29:


closed 2023-05-08 by stale[bot]

Tloe, 2023-05-29:
This is still an issue so should not be closed.

Axel-ex, 2024-03-15:
Having the same problem here, could you actually fix your problem? Thanks!

TristanWebber, 2024-06-27:
I'm also having this issue, specifically for the <b>esp32s3</b> and the arduino framework.

I have clangd working well for <b>esp32</b> and <b>esp32c3</b>, and I'm achieving that by setting the `--query-driver` to point to the relevant toolchain in clangd.setup, and adjusting some other flags in a .clangd file in the project. Everything also works well for the esp32s3 in an espidf project. This same approach is not working for the esp32s3 in an Arduino project.

It looks like either the compile_commands.json commands are not correct, or clangd is not able to parse those commands.

Has anyone found the root of the problem,  and a solution for this?

Tloe, 2024-06-27:
This python script fixes it for me. :) As you wrote, the compile_commands ned the full path

```
import shutil
import os

with open('compile_commands.json', 'r') as file :
  filedata = file.read()

filedata = filedata.replace('xtensa-esp32s3-elf-g++', os.path.expanduser('~') + '/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-g++')
filedata = filedata.replace('xtensa-esp32s3-elf-gcc', os.path.expanduser('~') + '/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc')

with open('compile_commands.json', 'w') as file:
  file.write(filedata)
```

TristanWebber, 2024-06-27:
Thanks for the tip @Tloe but that one hasn't solved it in my case for the s3 :(. To be fair, the LSP is mostly working really well, but it's the errors I'm getting for Arduino functions that I can't seem to get rid of.

Do you mind sharing what you have in your .clangd and clangd.setup to make this one work?

[image]
[image]

Tloe, 2024-06-27:
I don't have a clangd.setup

clangd:
```
CompileFlags:
  Add: [-mlong-calls, -ferror-limit=0, -isysroot=/home/t/.platformio/packages/toolchain-xtensa-esp32s3]
  Remove: [-fno-tree-switch-conversion, -mtext-section-literals, -mlongcalls, -fstrict-volatile-bitfields]
```

Tloe, 2024-06-27:
oh sorry.. been a while since I set this up.. thought clangd.setup was a file :p

```

require 'lspconfig'.clangd.setup {
  capabilities = require('cmp_nvim_lsp').default_capabilities(vim.lsp.protocol.make_client_capabilities()),
  on_attach = on_attach,
  cmd = {
--    "/usr/local/bin/clangd",
    "/home/t/dev/llvm-project-xtensa/build/bin/clangd",
    "--background-index",
    "--query-driver=/home/t/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-g*",
  },
  filetypes = { "c", "cpp", "objc", "objcpp", "cuda", "proto" },
}
```

Tloe, 2024-06-27:
/home/t/dev/llvm-project-xtensa/ is a build of the patched version that I linked to in the first post..

TristanWebber, 2024-06-27:
Perfect, thanks for that - that will be the missing piece of the puzzle. Otherwise, everything else is equivalent. I appreciate the help!