platformio/platform-espressif32 #1064: How to add upload option "--no-stub"

Forairaaaaa, 2023-03-30:
Hi,

Since esptool version going up to v4.5, my dev board meets `no response` during uploading occasionally. 
This issue can be fix by adding option `--no-stub` to the esptool upload command, slower but more stable.
I'm using pio in vscode on win11, and I wonder if I can config my `platformio.ini` to apply this option? Saving me from copying the upload command and add it myself.

Thanks!

valeros, 2023-03-30:
Hi @Forairaaaaa, just use the [upload_flags](https://docs.platformio.org/en/latest/projectconf/sections/env/options/upload/upload_flags.html) option.

Forairaaaaa, 2023-03-30:
> Hi @Forairaaaaa, just use the [upload_flags](https://docs.platformio.org/en/latest/projectconf/sections/env/options/upload/upload_flags.html) option.

Thanks a lot