platformio/platform-espressif32 #1068: firmware.map: Invalid argument

litian85734, 2023-04-05:
HI, Sir:
     I meet a  build issue that happen "firmware.map: Invalid argument" when I set " platform = espressif32 "  in plagformio.ini file.   as below
     plagformio.ini  file:
                                    [env:esp wrover kit]
                                    platform = espressif32    

     build information:
    =================================================================================
    c:/users/tommy/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32- 
      elf/bin/ld.exe: cannot open map file F:/TestPlatformIOV1.6/.pio/build/esp wrover kit/firmware.map: Invalid argument
      collect2.exe: error: ld returned 1 exit status
      *** [.pio\build\esp wrover kit\firmware.elf] Error 1
    ====================================================================================
      [image]

     But if set "espressif32 @ ~3.5.0" the build is ok,
                     [env:esp wrover kit]
                      platform = espressif32 @ ~3.5.0

      May  I know how to fix this issue when I use latset espressif32 vserion?

Thanks

valeros, 2023-04-05:
There are non-ascii characters in the project path on the screenshot. Try to move the project to a folder with only English letters in path.

jasonwang01, 2023-05-04:
I meet the same issue in stm32 platformio

c:/users/jason/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open map file 'D:\Personal\Desktop\public\github\bootloader\src\.pio\build\flash_2400/firmware.map': Invalid argument
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\flash_2400\firmware.elf] Error 1

closed 2023-06-18 by stale[bot]

QFFS, 2023-06-28:
> I meet the same issue in stm32 platformio
>
> c:/users/jason/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open [...]

I also meet this problem in stm32 platformio, Did you solve it?