platformio/platform-espressif32 #1071: Board "LilyGo T-Display-S3" fails to initialize

brha8286, 2023-04-07:
When creating a new project with board [LilyGo T-Display-S3](https://docs.platformio.org/en/latest/boards/espressif32/lilygo-t-display-s3.html?utm_source=platformio&utm_medium=piohome#id1), `Error: Unknown board ID 'lilygo-t-display-s3'` appears.  Attempted on multiple machines, VSCode up-to-date.

valeros, 2023-04-10:
Hi @brha8286, are you sure you're using the latest version of the platform v6.1.0? Please run `pio pkg update -g -p espressif32` in the IDE terminal.

cyberman54, 2023-04-30:
Wrong board name. The correct one is:
lilygo_t_display_s3

valeros, 2023-05-01:
Closing as no further information was given, feel free to reopen if you still need help.

closed 2023-05-01 by valeros

xbmcnut, 2023-11-14:
> Hi @brha8286, are you sure you're using the latest version of the platform v6.1.0? Please run `pio pkg update -g -p espressif32` in the IDE terminal.

Thanks so much. This helped me build [this](https://github.com/rzeldent/esp32-flightradar24-ttgo/tree/feature/lilygo-t-display-s3) project which was throwing up the same error.

pixelwave, 2023-11-17:
> Hi @brha8286, are you sure you're using the latest version of the platform v6.1.0? Please run `pio pkg update -g -p espressif32` in the IDE terminal.

That solved the issue!

fightforlife, 2024-07-08:
I still get this error after trying everything above.
Vscode on Windows with platformio: Core 6.1.15·Home 3.4.4

```
[env:lilygo-t-display-s3]
platform = espressif32
board = lilygo-t-display-s3
```

````
Processing lilygo-t-display-s3 (platform: espressif32; board: lilygo-t-display-s3; framework: arduino)
---------------------------------------------------------------------------------------------------------------
UnknownBoard: Unknown board ID 'lilygo-t-display-s3'
````