platformio/platform-espressif32 #1072: LOLIN C3 PICO

JohnCallingham, 2023-04-08:
Please add support for the LOLIN C3 PICO board.

chaoscrack, 2023-11-14:
Do you know when the PICO board would be on the list?
What should I take in the meanwhile?

bakercxj, 2024-10-13:
Please add support for Lolin C3 Pico.    

My Lolin C3 Mini works fine, but I like some of the features on the C3 pico.

Thanks

fjulian79, 2025-01-26:
I wanted to share my findings regarding the compatibility of the Wemos C3 Mini and C3 Pico with PlatformIO, as I was recently exploring whether the C3 Pico is officially supported.

After comparing the schematics of the two boards:

[C3 Mini v2.1.0](https://www.wemos.cc/en/latest/_static/files/sch_c3_mini_v2.1.0.pdf)
[C3 Pico v1.0.0](https://www.wemos.cc/en/latest/_static/files/sch_c3_pico_v1.0.0.pdf)

I noticed that they are identical in terms of buttons and the WS2812B LED. The main difference is that the Pico includes an additional battery charging circuit, with IO3 used for reading the battery voltage.

Given this similarity, it seems reasonable to use the C3 Mini support for both boards. However, adding the Pico as an explicit build target in PlatformIO would make things more straightforward and user-friendly.

I’d be happy to assist with testing if needed. ;-)

Br J.

mitra42, 2025-06-21:
By `project` above do you mean your projects own directory ? Or create a directory called project
i.e. if my project is `foo` 
Do you mean` foo/project/boards/lolin_c3_pico/pins_arduino.h.txt`
OR 
`foo/boards/lolin_c3_pico/pins_arduino.h.txt`

mp-se, 2025-06-23:
> By `project` above do you mean your projects own directory ? Or create a directory called project
> i.e. if my project is `foo` 
> Do you mean` foo/project/boards/lolin_c3_pico/pins_arduino.h.txt`
> OR 
> `foo/boards/lolin_c3_pico/pins_arduino.h.txt`

See the platformio documentation on custom boards for a guide

mitra42, 2025-06-23:
Part of the reason I'm asking about location is because your instructions aren't working for me. 
https://docs.platformio.org/en/latest/platforms/creating_board.html
and discussion at https://community.platformio.org/t/how-to-define-custom-board-within-project/39424/6
Give the location `<project>/boards/lolin_c3_pico.json`
and its picking up that board correctly.

However its complaining about 
```
/Users/mitra/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory
```
That file does a `#include "pins_arduino.h"` which isnt picking up your `pins_arduiono.h` 

None of the custom boards docs I could find talk about custom pins_arduino.h

mp-se, 2025-06-23:
Probably the setting where ir should pick up the custom board settings

board_build.variants_dir = boards