platformio/platform-espressif32 #1075: Unable to flash Adafruit QT Py ESP32-S3

tedcook94, 2023-04-13:
I have a project using both `esp32dev` modules and `adafruit_qtpy_esp32s3_nopsram` modules, but I'm having trouble uploading to the QT Py's.

Here is my `platformio.ini`:
```
[env]
platform = espressif32
monitor_speed = 115200
framework = arduino
build_flags = -DDEFAULT
lib_deps =
	bblanchon/ArduinoJson@^6.20.1
	adafruit/Adafruit SSD1306@^2.5.7
	rlogiacco/CircularBuffer@^1.3.3
	adafruit/Adafruit ADS1X15@^2.4.0
	adafruit/Adafruit ADXL345@^1.3.2
	https://github.com/tzapu/WiFiManager.git@^2.0.14-beta

[env:esp32dev]
board = esp32dev
build_flags = -DESP32DEV

[env:adafruit_qtpy_esp32s3_nopsram]
board = adafruit_qtpy_esp32s3_nopsram
build_flags = -DQTPY
```

All compiles succeed, and I can upload to the `esp32dev` and it works fine. When I try to upload to the QT Py, the upload will succeed, but the QT Py is unresponsive even after a reset. If I flash the same project using the Arduino IDE, the QT Py is responsive on both I2C devices and the serial monitor.

I've tried adding `-DARDUINO_USB_MODE = 1` to the `build_flags`, but that seemed to have no effect. Same for setting `board_build.flash_mode = qio`.

tedcook94, 2023-04-13:
I'll also add I'm seeing the same behavior with two different boards, so I'm fairly certain this isn't a hardware issue (especially since they both work with the Arduino IDE).

tedcook94, 2023-04-13:
Here are the upload settings that work for me in Arduino IDE:

[image]

valeros, 2023-04-13:
Hi @tedcook94 ! Sadly, I don't see any obvious reasons why it shouldn't work and I don't have this board at hand to reproduce the issue.

- What version of the `espressif32` dev-platform are you using?
- A simple blink doesn't work as well, right?

tedcook94, 2023-04-13:
Here are all of my dependency versions:
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/adafruit_qtpy_esp32s3_nopsram.html
PLATFORM: Espressif 32 (2023.2.0) > Adafruit QT Py ESP32-S3 No PSRAM
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.7 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.21.2
|-- Adafruit SSD1306 @ 2.5.7
|   |-- Adafruit GFX Library @ 1.11.5
|   |   |-- Adafruit BusIO @ 1.14.1
|   |   |   |-- Wire @ 2.0.0
|   |   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- CircularBuffer @ 1.3.3
|-- Adafruit ADS1X15 @ 2.4.0
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- Adafruit ADXL345 @ 1.3.2
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- Adafruit Unified Sensor @ 1.1.9
|-- WiFiManager @ 2.0.15-rc.1+sha.4611c75
|   |-- DNSServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WebServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- Adafruit GFX Library @ 1.11.5
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
```

And yes, I've tried the simple blink, WifiScan and LittleFS example sketches. All work fine in Arduino IDE, but not in PIO.

valeros, 2023-04-13:
> PLATFORM: Espressif 32 (2023.2.0)

Are you using a custom development platform? The latest version in this repository is [6.1.0](https://github.com/platformio/platform-espressif32/releases).

tedcook94, 2023-04-13:
@Jason2866's suggestion seems to have fixed it!

I did try to specify [this specific version](https://github.com/tasmota/platform-espressif32/releases/tag/2023.02.00) of the platform based off a suggestion in a forum post I found while trying to troubleshoot. I thought reverting back to `espressif` without any version specified would go back to the default package, but I guess not. Maybe a caching issue?