platformio/platform-espressif32 #1079: Please add XIAO ESP32-S3 Board

mneuhaus, 2023-04-19:
Please add the ESP32-S3 Board :) 

Here are the specifications: https://wiki.seeedstudio.com/xiao_esp32s3_getting_started

closed 2023-04-21 by valeros

mneuhaus, 2023-04-21:
thanks @valeros !

AlbertoLopSie, 2023-12-21:
Hi!

Debugging is not working.  According to documentation, the S3 supports 1-click debugging through its internal JTAG USB interface however, I cannot make i work.

``` yaml
// platformio.ini

[env:seeed_xiao_esp32s3]
platform = espressif32
board = seeed_xiao_esp32s3
framework = arduino
build_type = debug 
debug_tool = esp-builtin
; debug_tool = cmsis-dap              // Even when doc says this the default it doesn't work
; upload_protocol = esp-builtin       // Doesn't work

debug_init_break = tbreak setup

debug_port = /dev/cu.usbmodem21301     // on my Mac
monitor_port = /dev/cu.usbmodem21301
upload_port = /dev/cu.usbmodem21301
```
Only causes debugger to start but freeze with the following at the debug console:

```sh
undefinedOpen On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (34:85:18:91:38:88)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Reading symbols from /Users/Alberto/dev/DMI/HelloS3/.pio/build/seeed_xiao_esp32s3/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
```
and debugging session never starts.

It looks like the JTAG interface connects ok because the CPU is recognized, but after that nothing happens.

Uploading through the `esp-builtin` protocol doesn't work either.

Any hint?

commit de49a767 (2023-04-21): Update Arduino core to v2.0.8

Resolves #1079