platformio/platform-espressif32 #1080: `framework=espidf, arduino` isn't setting flags early enough for the Libraray Dependency Finder

j4m3s, 2023-04-20:
In trying to write a concise title for the issue I have made an assumption based on my investigation.

TLDR:
The Library Dependency Finder creates a different dependency tree (for code in a subdir in the `lib` folder) when it's run with `framework = espidf, arduino` vs `framework = arduino`, where an include is conditional upon the ARDUINO build flag (using `lib_ldf_mode = chain+` and even when `lib_compat_mode = off`).

So:

Given a folder structure of:
```
|_ lib
  |_A
    |_ src
      |_libAHeader.h
|_ src
  |_ main.cpp
```

And the `libAHeader.h` file containing a conditional include as follows:
```cpp
#ifdef ARDUINO
#    include <FastLED.h>
#endif

const colour = CRGB{255,255,255};
```

The dependency tree for A _includes_ `FastLED`, so including the `libAHeader.h` in `src/main.cpp` **succeeds,** when compiling with `framework = arduino`.  
The dependency tree for A _does not include_ `FastLED` when compiling with `framework = espidf, arduino` so the compile **fails**

I suspect this is because the `ARDUINO` flag is not set before the dependency finder runs for `framework = espidf, arduino`, but it is set beforehand for `framework = arduino`?  It's certainly set by the time the actual compilation happens.  I expect the same applies to `ESP32` and `ARDUINO_ARCH_ESP32`?

j4m3s, 2023-04-20:
As a workaround I've created a `library.json` file in `lib/A` that hard-codes the dependencies, which seems to be working.  It's not ideal though as this isn't a proper library it's just the way the code is organised to allow tests to be separated between native and MCU.

valeros, 2023-05-03:
Hi @j4m3s thanks for reporting, I believe I've found the reason why it doesn't work. Could you please revert back your workaround with `library.json` and compile your project using the latest platform revision from the development branch  (a Git client is requried)? For example:
```ini
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino, espidf
board = esp32dev
lib_ldf_mode = chain+
```

valeros, 2023-05-03:
@j4m3s Mind sharing a minimal project to reproduce the issue?

j4m3s, 2023-05-03:
The project this is hitting is far from minimal but I'll create a small one, NP :)

j4m3s, 2023-05-03:
Interestingly I've yet to reproduce the issue in [the simple project](https://github.com/j4m3s/pio-issue-1080), but I don't understand why.

Running `pio run -v` when `lib_ldf_mode=chain` results in a dependency graph that shows A depends on FastLED:
```
|-- A (License: Unknown, Path: pio-issue-1080/lib/A)
|   |-- FastLED @ 3.5.0 (License: MIT, Path: pio-issue-1080/.pio/libdeps/esp32dev/FastLED)
|-- B (License: Unknown, Path: pio-issue-1080/lib/B)
|   |-- A (License: Unknown, Path: pio-issue-1080/lib/A)
|   |   |-- FastLED @ 3.5.0 (License: MIT, Path: pio-issue-1080/.pio/libdeps/esp32dev/FastLED)
|-- FastLED @ 3.5.0 (License: MIT, Path: pio-issue-1080/.pio/libdeps/esp32dev/FastLED)
```

When `lib_ldf_mode=chain+`, LDF fails to find the dependency from A on FastLED which suggests that the ARDUINO flag that's used in an `#ifdef` wrapping the include isn't set:
```
|-- A (License: Unknown, Path: pio-issue-1080/lib/A)
|-- B (License: Unknown, Path: pio-issue-1080/lib/B)
|   |-- A (License: Unknown, Path: pio-issue-1080/lib/A)
|-- FastLED @ 3.5.0 (License: MIT, Path: pio-issue-1080/.pio/libdeps/esp32dev/FastLED)
```
_but yet library `A` compiles successfully._   Given that the LDF doesn't know that A depends on FastLED, surely the compile should fail?  It does on my bigger project :)

Based on [my understanding of how the LDF works](https://docs.platformio.org/en/latest/librarymanager/ldf.html#ldf-mode) I would have expected it to compile successfully when `lib_ldf_mode=chain` but fail when it's `chain+`.

j4m3s, 2023-05-08:
Nice work @valeros, you've cracked it! Works great now in both the minimal example and the big project, thank you!  I have to admit that's a big relief - I had no idea where to look next to understand why the minimal project worked but the big complex one didn't :)

commit da133384 (2023-05-03): Process internal Arduino libraries in mixed Arduino/IDF projects

Resolves #1080
closed 2023-05-08 by j4m3s