platformio/platform-espressif32 #1081: PlatformIO tries to download non-existent toolchain for ESP32

oliverschmidt, 2023-04-21:
The FujiNet project (https://github.com/FujiNetWIFI/fujinet-platformio) requires ESP32 PlatformIO version 3.4.0 (https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/platformio-sample.ini#L13).

Since yesterday the necessary PlatformIO environment can't be successfully initialized anymore. It always fails on trying to download `espressif/toolchain-riscv32-esp @ 8.4.0+2021r1`.

My understanding is that for ESP32 PlatformIO version 3.4.0 there's just no RISC-V support (and never will be). My guess is that the PlatformIO initialization now __always__ tries to download the RISC-V toolchain beside the Xtensa toolchain. But that doesn't makes sense for old ESP32 PlatformIO versions.

ivankravets, 2023-04-21:
Could you change the line in `platformio.ini` from `esp32_platform_version = 3.4.0` to `esp32_platform_version = 3.5.0`? Does it work?

oliverschmidt, 2023-04-21:
Thanks for taking care of my issue :-)

3.4.0: Fails with `Error: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r1' requirements for your system 'linux_x86_64'`

3.5.0: Fails with `Error: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2' requirements for your system 'linux_x86_64'`

ivankravets, 2023-04-22:
Please add the following line into the `[env]` section in the `platformio.ini`:
```ini
[env]
platform_packages =
    toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
```

If it works, we will provide the PR to the https://github.com/FujiNetWIFI/fujinet-platformio repository.

mozzwald, 2023-04-22:
> Please add the following line into the `[env]` section in the `platformio.ini`

I can confirm this fixes the issue with fujinet-platformio on Windows 10.

oliverschmidt, 2023-04-22:
@mozzwald: Thanks for following this issue - and providing the feedback :-)

ivankravets, 2023-04-22:
@oliverschmidt, could you add these lines to the config while you are locked to the v3.4.0 of dev-platform?

```ini
[env]
platform_packages =
    toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
```

oliverschmidt, 2023-04-22:
I did so and it works :-) It says:
```
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-riscv32-esp@8.4.0+2021r2-patch5 has been installed!
```

oliverschmidt, 2023-04-24:
> how is difficult to use stable ESP-IDF 4.0? The dev-platform version is 5.3.0.

It's for sure not easy as it doesn't even build as-is:

```
PLATFORM: Espressif 32 (5.3.0) > #FujiNet v1 8MB
...
PACKAGES:
 - framework-espidf @ 3.40403.0 (4.4.3)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-ninja @ 1.7.1
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
...
/.../.platformio/packages/framework-espidf/components/console/esp_console_repl.c:17:10: fatal error: esp_vfs_usb_serial_jtag.h: No such file or directory
```

ivankravets, 2023-04-24:
Please merge https://github.com/FujiNetWIFI/fujinet-platformio/pull/594 as a workaround while you stay on ESP-IDF 3.0.

ivankravets, 2023-04-24:
Resolved in https://github.com/FujiNetWIFI/fujinet-platformio/pull/594

Please update your `platformio.ini` in accordance with https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/platformio-sample.ini

closed 2023-04-24 by ivankravets

cirobruno, 2024-03-25:
I've got the same issue, but none of these tips have solved it!
Any other idea?

Thank you all.
Regards

```
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r1
UnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r1' requirements for your system 'darwin_x86_64'
```

ivankravets, 2024-03-25:
@cirobruno, please ensure that you use the latest `espressif32` development platform and run `pio pkg update` in the project folder.

cirobruno, 2024-03-25:
Thank you @ivankravets . I'll take care of that. I guess my platform is quite outdated.

Nevertheless, during the dawn, Boris Jäger (sivar2311) has forwarded to me recent post with a solution given by Maximilian Gerhardt (maxgerhardt) which has solved this specific issue. It looks like the abscence of those packages forces the platform to replace them with the latest ones. (links below)

Thank you very much and, again, congratulations for your job.

Best regards,

https://community.platformio.org/t/could-not-find-the-package-with-espressif-toolchain-riscv32-esp-8-4-0-2021r1/39301

https://community.platformio.org/t/esp32-s3-devkitc-1u-n8-nd-project/39074

KP4good, 2025-01-10:
> Please add the following line into the `[env]` section in the `platformio.ini`:
> 
> ```ini
> [env]
> platform_packages =
>     toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
> ```
> 
> If it works, we will provide the PR to the https://github.com/FujiNetWIFI/fujinet-platformio repository.

Works! Thank you!