platformio/platform-espressif32 #1083: Backtrace not shown on crash - ESP32-S3 over UART

joshcorn, 2023-04-24:
I have a custom board using an ESP32-S3-WROOM-1 module (N8R2) and I'm connecting to it using a USB<>UART converter instead of the onboard USB. I'm able to upload firmware and view the serial monitor but I don't see any backtrace whenever the device crashes. I've read that there may be some flags that can be set using ESP-IDF (maybe disable CONFIG_ESP_DEBUG_OCDAWARE) [source](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/fatal-errors.html) but I know that isn't a flag in Arduino. Has anybody else experienced this or have a way of getting the backtrace when using UART? 

I just receive this when the device crashes: 
ESP-ROM:esp32s3-20210327  
Build:Mar 27 2021  
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)  
Saved PC:0x420dd4e2 
SPIWP:0xee  
mode:DIO, clock div:1  
load:0x3fce3808,len:0x43c  
load:0x403c9700,len:0xbec  
load:0x403cc700,len:0x2a3c

Here is the board file I have been using if anyone sees any flags that should be added/modified:
{
  "build": {
    "arduino": {
      "ldscript": "esp32s3_out.ld",
      "memory_type": "qio_qspi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_USB_CDC_ON_BOOT=0",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_USB_MODE=0"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      ["0x239A", "0x811B"],
      ["0x239A", "0x011B"],
      ["0x239A", "0x811C"]
    ],
    "mcu": "esp32s3",
    "variant": "test_board"
  },
  "connectivity": ["wifi", "bluetooth"],
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": ["esp-builtin"],
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": ["arduino", "espidf"],
  "name": "Test Board",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "use_1200bps_touch": true,
    "wait_for_upload_port": false,
    "require_upload_port": true,
    "speed": 460800
  }
}

joshcorn, 2023-04-25:
Hi @valeros, Thanks for getting back! Here is my `platformio.ini` file: 
```
[env:test1]
platform = espressif32@6.1.0
board = test1
framework = arduino

board_build.partitions = default_8MB.csv 
board_upload.flash_size=8MB

platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git

build_flags = 
	${env.build_flags} 
	-D=${PIOENV}
	-DCORE_DEBUG_LEVEL=5

build_src_filter = +<*> -<.git/> -<.svn/> -<LocalServer.cpp> -<LocalServer.h>

build_type = debug

upload_port = /dev/tty.usbserial-14220
monitor_port = /dev/tty.usbserial-14220
monitor_filters = esp32_exception_decoder
monitor_speed = 115200
lib_deps = 
	h2zero/NimBLE-Arduino@^1.4.1
```

Jason2866, 2023-04-25:
By setting this
```
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": ["esp-builtin"],
    "openocd_target": "esp32s3.cfg"
```
you will not get any output on serial UART.

Jason2866, 2023-04-30:
This 
```
   "use_1200bps_touch": true,
    "wait_for_upload_port": false,
```
is only for MCUs with full featured USB HID implementation. The C3 and S3 do not have.
Afaik only the S2 does have. Anyways it should not do any harm.
Can try your setup tomorrow and will come back with my findings...

Jason2866, 2023-05-01:
Get messages via UART with this one
```
{
  "build": {
    "arduino": {
      "ldscript": "esp32s3_out.ld",
      "memory_type": "qio_qspi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_USB_CDC_ON_BOOT=0",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_USB_MODE=0"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": ["wifi", "bluetooth"],
  "frameworks": ["arduino", "espidf"],
  "name": "Test Board",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 460800
},
"url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/",
"vendor": "Espressif"
}
```

joshcorn, 2023-05-01:
Hi @Jason2866 thanks for the follow-up. Your last post led me to the answer. The reason seems to be a difference in the variant file. When I remove 
```
#define TX1 TX  
#define RX1 RX
```
from my `pins_arduino.h`, and I use Serial instead of Serial1, I am able to get the backtrace to show up. Thank you!

closed 2023-05-13 by joshcorn