platformio/platform-espressif32 #1088: ESP32-C3 Upload via USB CDC without Manually Entering Bootloader

hyansuper, 2023-04-30:
Hello,
I have a custom board with ESP32-C3-MINI-N4 module, I use USB CDC for uploading and serial without usb-uart convert chip.
Everytime I upload a sketch, I'll have to manually press reset button and boot button to upload, and then press reset button again for the new sketch to start running.

Do I understand it correctly? #591 suggests these procedure can be automatic without Manually Entering Bootloader. but I can't do it with ESP32-C3

My board has the usb+/usb- lines directly connected to a usb port and ESP32-C3.

The platformio.ini file:
```
[env:esp32-c3-mini]
platform_packages =
    toolchain-riscv32-esp @ 8.4.0+2021r2-patch5

platform = espressif32
board = esp32-c3-devkitm-1

board_build.mcu = esp32c3
board_build.flash_mode = qio
board_build.f_cpu = 160000000L

framework = arduino
upload_protocol = esptool

build_flags = 
    -DCORE_DEBUG_LEVEL=0
    -DARDUINO_USB_CDC_ON_BOOT=1
    -DARDUINO_USB_MODE=1
```

Jason2866, 2023-04-30:
The solution for the S2 is not possible for the C3. It is completely different.
You can use USB/JTAG mode for C3 with the native inbuilt USB modem. Using this mode makes reset of device possible.

hyansuper, 2023-04-30:
I found #874 and solved it：
```
[env:esp32-c3-mini]
platform_packages =
    toolchain-riscv32-esp @ 8.4.0+2021r2-patch5

;platform = https://github.com/platformio/platform-espressif32.git
platform = espressif32@4.4.0
board = esp32-c3-devkitm-1

board_build.mcu = esp32c3
board_upload.before_reset = usb_reset

framework = arduino
upload_protocol = esptool

build_flags = 
    -DARDUINO_USB_CDC_ON_BOOT=1
    -DARDUINO_USB_MODE=1
```

closed 2023-04-30 by hyansuper