platformio/platform-espressif32 #1095: Can you add esp32-s3-eye 8mb flash and 8mb PSRAM

hzhh110, 2023-05-13:
I used vscode + platformio + esp32-s3-eye to integrate esp-who through arduino, but it has not been successful, it seems that it is stuck in PSRAM, what should I do?

Jason2866, 2023-05-13:
It is already there. Use `esp32s3camlcd` as board.

hzhh110, 2023-05-15:
How to set this?If I use this setting, arduinoKeep breaking down

[env:esp32s3camlcd]
platform = espressif32 @^5.2
board = esp32s3camlcd
framework = arduino
​
​
​
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376cfc
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fce3808,len:0x39c
load:0x403c9700,len:0x9bc
load:0x403cc700,len:0x2900
entry 0x403c98c0
E (82) cpu_start: Octal Flash option selected, but EFUSE not configured!

void setup() {
  Serial.begin(115200);
  delay(5000);
  Serial.println("Error initing PSRAM");
}
void loop() {
}

hzhh110, 2023-05-15:
> It is already there. Use `esp32s3camlcd` as board.

How to set this?If I use this setting, arduinoKeep breaking down

[env:esp32s3camlcd]
platform = espressif32 @^5.2
board = esp32s3camlcd
framework = arduino
​
​
​
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376cfc
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fce3808,len:0x39c
load:0x403c9700,len:0x9bc
load:0x403cc700,len:0x2900
entry 0x403c98c0
E (82) cpu_start: Octal Flash option selected, but EFUSE not configured!

void setup() {
Serial.begin(115200);
delay(5000);
Serial.println("Error initing PSRAM");
}
void loop() {
}

Jason2866, 2023-05-15:
Opened PR #1096 to fix the wrong boards.json

closed 2023-05-16 by valeros

hzhh110, 2023-05-17:
> Opened PR #1096 to fix the wrong boards.json

Still no, not with arduino or IDF, IDFCannot add esp32-camera component, internal SDK seems to be 3.5
As long as the arduino is compiled with this setting, it will crash.
platform = espressif32
platform = espressif32 @^5.2

for arduino
`ELF file SHA256: 4a54e85c28146a60

E (106) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376cb8
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9b4
load:0x403cc700,len:0x28fc
entry 0x403c98bc
E (80) cpu_start: Octal Flash option selected, but EFUSE not configured!

abort() was called at PC 0x40376895 on core 0

Backtrace: 0x403770aa:0x3fceb260 0x40379aed:0x3fceb280 0x4037f3a5:0x3fceb2a0 0x40376895:0x3fceb320 0x403cd61b:0x3fceb350 0x403cd8da:0x3fceb380 0x403c9911:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED`

hzhh110, 2023-05-18:
for :
esp32s3camlcd.json

`{
  "build": {
    "arduino":{
      "ldscript": "esp32_out.ld",
      "memory_type": "opi_opi"
    },
    "boot": "opi",
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_ESP32S3_CAM_LCD",
      "-DARDUINO_USB_MODE=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dout",
    "hwids": [
      [
        "0X303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3camlcd"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "ESP32S3 CAM LCD",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "https://www.espressif.com/en/news/Maple_Eye_ESP32-S3",
  "vendor": "Espressif"
}`

hzhh110, 2023-05-18:
{
  "build": {
    "arduino":{
      "ldscript": "esp32_out.ld",
      "memory_type": "opi_opi"
    },
    "boot": "opi",
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_ESP32S3_CAM_LCD",
      "-DARDUINO_USB_MODE=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dout",
    "hwids": [
      [
        "0X303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3camlcd"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "ESP32S3 CAM LCD",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "https://www.espressif.com/en/news/Maple_Eye_ESP32-S3",
  "vendor": "Espressif"
}

hzhh110, 2023-05-18:
for :
esp32s3camlcd.json

{
  "build": {
    "arduino":{
      "ldscript": "esp32_out.ld",
      "memory_type": "opi_opi"
    },
    "boot": "opi",
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_ESP32S3_CAM_LCD",
      "-DARDUINO_USB_MODE=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dout",
    "hwids": [
      [
        "0X303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3camlcd"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "ESP32S3 CAM LCD",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "https://www.espressif.com/en/news/Maple_Eye_ESP32-S3",
  "vendor": "Espressif"
}

hzhh110, 2023-05-18:
> It is already there. Use `esp32s3camlcd` as board.

I use the latest modified board information, but I still can't compile it successfully. I use @ 6.2 version and @ 5.2 version. The way of arduino is crashed, and I can't succeed.Yes

hzhh110, 2023-05-18:
> It is already there. Use `esp32s3camlcd` as board.

I use the latest modified board information, but I still can't compile it successfully. I use @ 6.2 version and @ 5.2 version. The way of arduino is crashed, and I can't succeed

AlexC-0806, 2025-01-31:
I'm trying to run projects on this board, but still i get the same errors about psram and efuses. Hasn't it been solved yet? I've created an [issue](https://github.com/platformio/platform-espressif32/issues/1535) but no one has answered yet

commit 5fdb90c7 (2023-05-16): Fixed flash memory size for ESP32S3 CAM LCD board (#1096)

S3 Eye has 8MB OPI Flash

Resolves #1095