platformio/platform-espressif32 #1097: Issue with Special Characters and ESP32

nh1628, 2023-05-15:
When trying to compile an ESP32-S3 through my onedrive folder containing a "-" and a "," it has issues with generating firmware map files and compiling. When creating a new project and compiling in a folder without spaces and special characters everything generates and compiles correctly. Copying the generated files to the onedrive folder will also not work.

Jason2866, 2023-05-15:
Using special chars in path is always a bad idea.

valeros, 2023-05-16:
Hi @nh1628 , I'm not sure about the `-` character in paths, but `,` is definitely not desirable, please don't use such special characters in the path to you project. As for the Onedrive folder,  please share here the error message.

closed 2023-08-10 by stale[bot]