platformio/platform-espressif32 #1098: Add support for the board **ESP32-DevKitM-1**

loicmorvan, 2023-05-16:
What kind of issue is this?

- [x] **Feature Request**.
      Add support for the board **ESP32-DevKitM-1**.

------------------------------------------------------------------

### Configuration

**Operating system**: macOS

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.7

### Description of problem

PlatformIO does not seem to have support for the **ESP32-DevKitM-1**.
The main issue with this board is that build must be executed with the option `CONFIG_FREERTOS_UNICORE` set in *menuconfig*.

#### Steps to Reproduce

1. Create project with platform `espressif32` and board `esp32dev`.
2. Build and upload the default program on the board `ESP32-DevKitM-1`.
3. Open the monitor to check the messages from the board.

### Actual Results

Board is rebooting in an infinite loop with this message:
```
E (124) cpu_start: Running on single core variant of a chip, but app is built with multi-core support.
E (124) cpu_start: Check that CONFIG_FREERTOS_UNICORE is enabled in menuconfig
```

### Expected Results

A specific board is available in PlatformIO for **ESP32-DevKitM-1**.
The program is executed as expected on that board.

Jason2866, 2023-05-16:
Whats wrong setting the needed sdkconfig option `CONFIG_FREERTOS_UNICORE` ?

loicmorvan, 2023-05-16:
Would it be that easy? 😄 
Where do you do that? On the `platform.ini` file?

Jason2866, 2023-05-16:
Nearly, make a file called `sdkconfig.defaults` (placed in the root of the project) and place the text `CONFIG_FREERTOS_UNICORE` in the file.

loicmorvan, 2023-05-16:
I'm not sure to understand all the implications of all of that, but it seems to work as expected, so thank you very much 😃

valeros, 2023-05-19:
Please use `sdkconfig.defaults` or menuconfig.

closed 2023-05-19 by valeros