platformio/platform-espressif32 #1099: [BUG REPORT] "flash_mode": "qio" in the board config does not work.

wuhao19920619, 2023-05-16:
**Operating system**: 

Windows 10 HOME

VScode Version:
Version: 1.78.2 (user setup)
Commit: b3e4e68a0bc097f0ae7907b217c1119af9e03435
Date: 2023-05-10T14:39:26.248Z
Electron: 22.5.2
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Windows_NT x64 10.0.19045
Sandboxed: Yes

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.7

### Description of problem

Here is my custmized board config for ESP32-S3-WROOM-1-N16R8
**The content of `esp32-s3-n16r8.json`:**
```json
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_16MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_ESP32S3_DEV",
      "-DARDUINO_USB_MODE=1",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DARDUINO_USB_CDC_ON_BOOT=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": [
      "esp-builtin"
    ],
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Espressif ESP32-S3-N16R8 (16 MB QD, 8M OL PSRAM)",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_cn.pdf",
  "vendor": "Espressif"
}
```

However when use the verbose upload, it gives:
```
"C:\Users\user\.platformio\penv\Scripts\python.exe" "C:\Users\user\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32s3 --port "COM19" --baud 460800 --before default_reset --after hard_reset write_flash -z 
--flash_mode dio --flash_freq 80m --flash_size 16MB 0x0000 D:\OneDrive\VScode\arduino\ESP32-S3-BOX-LITE\.pio\build\release\bootloader.bin 0x8000 D:\OneDrive\VScode\arduino\ESP32-S3-BOX-LITE\.pio\build\release\partitions.bin 0xe000 C:\Users\user\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin 0x10000 .pio\build\release\firmware.bin
```

Clearly the flash_mode is still dio rather than qio.

#### Steps to Reproduce

1. Just use verbose upload. Then check the command.

### Actual Results
```
--flash_mode dio
```

### Expected Results
```
--flash_mode qio
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
default_envs = release

[env]
platform = espressif32
board = esp32-s3-n16r8
framework = arduino
monitor_speed = 115200
lib_deps = 
    adafruit/Adafruit NeoPixel@^1.11.0

[env:release]

```

Jason2866, 2023-05-16:
This is no bug. You can not flash in mode QIO. Fastest initial flash boot mode is DIO. The binary is built to run in flash mode QIO. The 2nd stage bootloader will switch from mode DIO to mode QIO.
You can query the flash mode in your program. There are APIs in IDF and Arduino for.

wuhao19920619, 2023-05-16:
Ok thank you for the information.

BTW, I tried to manually modify it to `--flash_mode qio` and upload manually with esptool. It look pretty normal. Nothing different from default automatic upload.

valeros, 2023-05-19:
Closing as `dio` is the correct option when uploading.