platformio/platform-espressif32 #1100: How to make a project work as a component

nicklasb, 2023-05-17:
Hi!

Is there a way to make an ESP-IDF project also work as a component?
As the project has "scriptable" parts in it CMakeLists.txt, it can't be used as a component through a by another project. 

Can it be detected if a CMakeLists.txt is being used for a project or component?

The reason I am asking is that RadioLib (that is used in many examples, for example LilyGo:s TTGO series) AFAIK currently cannot be used by a ESP-IDF-component. 

Example here, where I have tried to put it into components after a lot of variation:  https://github.com/nicklasb/RadioLib_PIO_ESPIDF

Any ideas?

Jason2866, 2023-05-17:
Here is a example how LittleFS is used a IDF component
https://github.com/Jason2866/Arduino_IDF_LittleFS/tree/63bdaac7fdbc285a216b9d4cced67135909cda28
and here it is used via the espressif component manager
https://github.com/Jason2866/Arduino_IDF_LittleFS

nicklasb, 2023-05-17:
But are they used by another component?

Jason2866, 2023-05-17:
Now, i got what you want to do. Have you got it working with plain IDF?
Imho Platformio Setup is second step.

nicklasb, 2023-05-17:
No, I haven't. On the other hand, that doesn't seem to be the issue here, as I don't even know where to begin. 
(To me it is very important that things can be used from components, otherwise it would be very difficult to scale the systems.)

nicklasb, 2023-05-21:
@Jason2866 
I suppose the comment above might have been interpreted a bit stand-off-isch, that wasn't my intention, sorry. 

The problem I have is not getting it to work in in ESP-IDF. Actually @jgromes just added an example for ESP-IDF that works for him at https://github.com/jgromes/RadioLib/tree/master/examples/NonArduino/ESP-IDF.
My problem is that I need to be able to use RadioLib as a component, from a component.

And for that, I don't know, but *assume* that there cannot be project() syntax in the CMakeTexts.txt. 

As it in general would be nice to have a project both be a project when one is developing, and a component when distributing, I was hoping that this could be possible. 

In another thread, I proposed that perhaps something with ifs in the CMakeLists.txt could be used to combine the two cases? Could that be something?

Jason2866, 2023-05-21:
As, i wrote needed some time to understand what you want. Not giving an answer is just because i have no idea how to achieve. Maybe it is not possible at all, idk.

closed 2025-04-25 by stale[bot]