platformio/platform-espressif32 #1101: Problem with ESP32 S3

gauix4, 2023-05-17:
Modul esp32  S3 16Mo flash extern:
https://fr.aliexpress.com/item/32901557709.html?spm=a2g0o.order_list.order_list_main.25.4cd25e5bcq0por&gatewayAdapt=glo2fra

I test the chips in this support:
https://fr.aliexpress.com/item/1005002950744932.html?spm=a2g0o.order_list.order_list_main.39.532a5e5b8APO96&gatewayAdapt=glo2fra

I have a lot of difficulty getting my ESP32 S3 to work properly
i use Platform IO is the only way to make them work and hack with espressif32@5.3.0 version and use esp32-s3-devkitc-1 board but it not stable at all plus flash and often corrupted the write location and at 0x0 instead of 0x1000 for the USB driver versions CP21... or CH34...
do you by any chance have a better solution? Thanks in advance for your answer and reading :) 

[image]

code:
```cpp

#include <Arduino.h>

#define noAnalogRef    // define to remove analogRef test that some cores do not support 
//#define noDtostrfTest  // define to remove the dtostrf test that some cores do not support
//#define noBvTest       // define to remove bv test that some cores do not support
//#define USBserialWait  // define to force the main sketch to wait for USB to Serial connection - needed on some native serial boards
//#define noLtoaTest     // define to remove Itoa test that some cores do not support

// Uncomment only your specific board, comment the others
//#define BOARD_ATMEGA328
//#define BOARD_LEONARDO
//#define BOARD_STM32F103RB
//#define BOARD_DUE
//#define BOARD_ZERO
//#define BOARD_ESP8266
//#define BOARD_NODEMCU
#define BOARD_ESP32
//#define BOARD_ESP32
//#define BOARD_NRF52

// Arduino ATMEGA328
#ifdef BOARD_ATMEGA328
#define PIN_1 3
#define PIN_2 4
#endif

// Arduino Leonardo
#ifdef BOARD_LEONARDO
#define PIN_1 2
#define PIN_2 3
#define noBvTest 
#define USBserialWait
#endif

// Nucleo STM32F103RB - not tested yet
#ifdef BOARD_STM32F103RB
#define PIN_1 12
#define PIN_2 13
#define noAnalog 
#define noBvTest 
#define noLtoaTest
#endif

// Arduino Due
#ifdef BOARD_DUE
#define PIN_1 2
#define PIN_2 3
#define noDtostrfTest
#define noBvTest
#define USBserialWait
#endif

// Arduino Zero
#ifdef BOARD_ZERO
#define PIN_1 2
[... 1323 lines cut ...]
      delayMicroseconds(100);
      delayMicroseconds(100);
      delayMicroseconds(100);
      delayMicroseconds(100);
      delayMicroseconds(100);
      delayMicroseconds(100);
      delayMicroseconds(100);
      delayMicroseconds(100);
      delayMicroseconds(100);
      delayMicroseconds(100);
      delayMicroseconds(100);
      delayMicroseconds(100);
    }
  }
  n=millis();
  d = ((float)n - (float)m) / ((float)i * (float)j);
  d *= 1000.0;
  d -= overhead;
  d /= 20.0;
  Serial.print (d,3);
  Serial.println (F(" us"));

  Serial.println(F("-----------"));
}
```

Jason2866, 2023-05-18:
This is not a forum for discussing general issues.
There is no issue writing and flashing an ESP32-S3 with Platformio.

valeros, 2023-05-19:
Hi @gauix4 ! I'm afraid the issue tracker in this repo is not the right place to look for help with specific projects.

closed 2023-05-19 by valeros

gauix4, 2023-06-22:
> Salut@gauix4! Je crains que l'outil de suivi des problèmes de ce référentiel ne soit pas le bon endroit pour rechercher de l'aide pour des projets spécifiques.

it does exactly the same thing on Arduino.....
[image]

there is just a stability problem many projects crash the ESP S3 reboot
I have several new chips they all do exactly the same thing regardless of the IDE used