platformio/platform-espressif32 #1109: Networking uses 80k more memory when updating to PlatformIO ver 6.3.1

gmdriscoll, 2023-05-28:
Hello,

Has anything been added as a new default into menuconfig or ver 6.3.1 that could cause over 80k more RAM to be used during networking?  I upgraded today from ver 6.1.0.

I continuously monitor memory usage in 2 different applications and have been running about 109k free.  After updating free RAM is about 23k.  This happens with an ESPNOW instantiation, a WIFI instantiation, and an Ethernet instantiation. 

Currently running tasks use about 47-50k of RAM.

Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.3.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50002.230512 (5.0.2)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)       
 - toolchain-xtensa-esp32 @ 11.2.0+2022r1
Reading CMake configuration...

Thanks,
-Greg

Jason2866, 2023-05-28:
Since 6.3.0 the IDF 5.0.2 release is used. Before it was IDF 5.0.1.
Platformio uses the menuconfig defaults from IDF. So you have to look there what has changed.

gmdriscoll, 2023-05-28:
Installed 6.10 again, same issue.  Restored minimal code changes to serial communication, same issue.  Went through every setting in menuconfig and cannot find anything.  Hopefully somebody else experiences this.  Will also post in IDF forum.  Something got added by default and it's not readily apparent.

gmdriscoll, 2023-05-28:
Was successful in backing down to 6.1.0 to restore expected RAM usage.  sdkconfig.esp32dev was highly modified in update to 6.3.1 to remove legacy settings.  Will update this thread if successful in figuring out change that adds in the 80k of memory usage.

gmdriscoll, 2023-05-29:
Closing this issue due to typo in code.

closed 2023-05-29 by gmdriscoll