platformio/platform-espressif32 #1110: BLE Server works in Arduino IDE but not in PlatformIO

TanaroSch, 2023-05-29:
When using the BLE_server example from the Arduino IDE I can see the services and UUIDs in nRF Connect on my phone. However when I build the project with the exact same code (but with added #include <Arduino.h>) nRF keeps trying to discover the services but never gets any data.

Code:
```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

void setup() {
  Serial.begin(9600);
  Serial.println("Starting BLE work!");

  BLEDevice::init("Long name works now");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```
my platformio.ini
```
[env:nodemcu-32s]
platform = espressif32@3.5.0
board = nodemcu-32s
framework = arduino
lib_deps = 
	adafruit/Adafruit NeoPixel@^1.11.0
board_build.partitions = huge_app.csv
```

TanaroSch, 2023-05-29:
When I leave this blank the compilation does not work at all. Is there a version you would recommend?

valeros, 2023-05-29:
> When I leave this blank the compilation does not work at all. Is there a version you would recommend?

What version of ESP32 core you're using in Arduino IDE?

TanaroSch, 2023-05-29:
In Arduino IDE I am using version 2.0.9. However when I try to use this version in platformio, I get the following error:
```
Platform Manager: Installing espressif32 @ 2.0.9
Error: Could not find the package with 'espressif32 @ 2.0.9' requirements for your system 'windows_amd64'
```

valeros, 2023-05-30:
@TanaroSch You're mixing up two different concepts. The platform version `6.3.1` is the latest with Arduino core v2.0.9:
```ini
[env:nodemcu-32s]
platform = espressif32 @ ~6.3.1
board = nodemcu-32s
framework = arduino
lib_deps = 
	adafruit/Adafruit NeoPixel@^1.11.0
board_build.partitions = huge_app.csv
```

TanaroSch, 2023-05-30:
Thank you, that actually resolved the issue with BLE.

However, adding my old code again to bring my two desired components together, Husarnet does not work with the new version. When adding the required 
```
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/husarnet/arduino-esp32/releases/download/1.0.4-1/arduino-husarnet-esp32.zip
```
I get a ton of errors for BLE, e.g:
```
path/.platformio/packages/framework-arduinoespressif32@src-ac062c2d4978b25cad32b0aa2d894938/libraries/BLE/src/BLEDescriptor.cpp: In constructor 'BLEDescriptor::BLEDescriptor(BLEUUID, uint16_t)':
path/.platformio/packages/framework-arduinoespressif32@src-ac062c2d4978b25cad32b0aa2d894938/libraries/BLE/src/BLEDescriptor.cpp:40:36: error: 'malloc' was not declared in this scope
  m_value.attr_value   = (uint8_t*) malloc(max_len);  // Allocate storage for the value.
                                    ^~~~~~
path/.platformio/packages/framework-arduinoespressif32@src-ac062c2d4978b25cad32b0aa2d894938/libraries/BLE/src/BLEDescriptor.cpp:40:36: note: 'malloc' is defined in header '<cstdlib>'; did you forget to '#include <cstdlib>'?
path/.platformio/packages/framework-arduinoespressif32@src-ac062c2d4978b25cad32b0aa2d894938/libraries/BLE/src/BLEDescriptor.cpp:19:1:
+#include <cstdlib>

path/.platformio/packages/framework-arduinoespressif32@src-ac062c2d4978b25cad32b0aa2d894938/libraries/BLE/src/BLEDescriptor.cpp:40:36:
  m_value.attr_value   = (uint8_t*) malloc(max_len);  // Allocate storage for the value.
```

But when not adding it, I get the error `src/main.cpp:11:10: fatal error: Husarnet.h: No such file or directory`

Do you know a solution to this as well, or shall I open a new Issue for this?

Jason2866, 2023-05-31:
@TanaroSch The only place you can open an issue is the place where husarnet is maintained.
This is a fork of Platformio. No support here.

TanaroSch, 2023-05-31:
All right, but thanks :)

closed 2023-05-31 by TanaroSch