platformio/platform-espressif32 #1111: Interrupts conflict

notsleep196, 2023-05-31:
Hello!

I faced the issue with interrupts. Devise hangs on restart in some infinite loop in attachInterrupt() method.
<details><summary>Sketch to reproduce</summary>

```
#include <Arduino.h>
#include <esp_sleep.h>

static void IRAM_ATTR testInterruptHandler1() {
    if (1 == 0) {
        return;
    }
    return;
}

void setup() {
    Serial.begin(115200);
    Serial.println("Start initialization");
    pinMode(GPIO_NUM_4, OUTPUT);
    digitalWrite(GPIO_NUM_4, HIGH);
    Serial.println("Atach interrupt as always");
    attachInterrupt(GPIO_NUM_33, testInterruptHandler1, RISING);
    Serial.println("Configure wakeup interrupt");
    pinMode(GPIO_NUM_4, INPUT_PULLDOWN);
    gpio_wakeup_enable(GPIO_NUM_4, GPIO_INTR_HIGH_LEVEL); // RTC GPIOs can be used as source only
    esp_sleep_enable_gpio_wakeup();
    Serial.println("Initialization finished. Starting loop...");

}

void loop() {
    delay(10000);
    Serial.println("Running loop...");
    delay(2000);
    ESP.restart();

}
```

</details>

<details><summary>My platformio.ini</summary>

```
[env:wemos_d1_mini32]
platform = espressif32 @ 6.3.1
board = wemos_d1_mini32
framework = arduino
monitor_filters =
    esp32_exception_decoder
```

</details>

The main question: is this correct behaviour or this is a bug? I expected that all configuration related to my sketch is cleared after restart, but looks like it doesn't act like this. So my test sketch works only once after uploading firmware or power reboot and hangs in infinite loop after resetting via ESP.restart(). The same behaviour after reset because of unhandled exception (e.g. divide by zero)

Jason2866, 2023-05-31:
Please test with ArduinoIDE. Most probably the issue is in Arduino code and not Platformio
If so open an issue in espressif Arduino Github.

notsleep196, 2023-05-31:
Thank you for reply! Sorry for my mistake with repo.

closed 2023-06-12 by valeros