platformio/platform-espressif32 #1113: Debugging does not upload nor start.

mickeyl, 2023-05-31:
I'm on an Apple Mac Studio (Apple Silicon) with the latest vscode, platformio, etc.

I have a project w/ an ESP32C3. Compiling, uploading, running works fine.

Debugging does not work at all. It starts compiling the project, but then stops.

```
Successfully created esp32c3 image.
================================================================================================================================================== [SUCCESS] Took 19.38 seconds ==================================================================================================================================================
```

Here's my `platformio.ini`:

```
[platformio]
build_dir = /Volumes/ramdisk/platformio

[env:esp32-c3-wican]
platform = espressif32
board = esp32-c3-devkitm-1
framework = espidf
monitor_speed = 115200
monitor_filters = direct
build_flags = 
  -DCONFIG_BOARD_WICAN
  -DCONFIG_TWAI_TX=0
  -DCONFIG_TWAI_RX=3
  -DCONFIG_LED_HEARTBEAT=7
  -DCONFIG_LED_MOBILE=8
  -DCONFIG_LED_HARDWARE=9
extra_scripts = pre:computeVersion.py
upload_protocol = esp-builtin
debug_tool = esp-builtin
upload_port = /dev/cu.usbmodem2123101
monitor_port = /dev/cu.usbmodem2123101

board_build.flash_mode = qio
board_build.f_flash = 40000000L
board_build.partitions = partitions.csv
check_skip_packages = yes
```

Any idea what could be wrong?

valeros, 2023-06-01:
Hi @mickeyl, 

> In the tab 'Debugging-Console', I see the following:

Is there anything else? The log seems truncated.

mickeyl, 2023-06-01:
@valeros Unfortunately not. It just hangs there.

valeros, 2023-06-01:
Could you try running a debug session in the CLI mode? Just open a new terminal window in the IDE and run `pio debug --interface=gdb -- -x .pioinit`

mickeyl, 2023-06-02:
@valeros Pretty much the same:

```
 /V/S/D/l/CANyonerOS   *~$  pio debug --interface=gdb -- -x .pioinit                                                                                                                                                                                                                                   Fri Jun  2 12:59:59 2023
Reading symbols from /Volumes/ramdisk/platformio/esp32-c3-wican/firmware.elf...
I'm sorry, Dave, I can't do that.  Symbol format `elf32-littleriscv' unknown.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Info : esp_usb_jtag: serial (34:B4:72:01:74:7C)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on pipe
Info : accepting 'gdb' connection from pipe
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Found 8 triggers
Info : Flash mapping 0: 0x10020 -> 0x3c0a0020, 205 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 608 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
Info : Flash mapping 0: 0x10020 -> 0x3c0a0020, 205 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 608 KB
Info : Using flash bank 'esp32c3.irom' size 612 KB
Info : Flash mapping 0: 0x10020 -> 0x3c0a0020, 205 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 608 KB
Info : Using flash bank 'esp32c3.drom' size 208 KB
 /V/S/D/l/CANyonerOS   *~$                                       
```

valeros, 2023-06-02:
TBH, such behavior is a mystery for me. Have you tried debugging another project? Like simple `hello-world` or `arduino-blink`? Maybe you have a similar board to make sure it's not a hardware issue?

bdewalch, 2023-06-14:
I am having a similar problem.  I have a custom board with an ESP32-C3.  I am using vscode and Platformio on a windows machine.  I believe what is happening is that the firmware gets loaded and the IDE enters debug mode but the chip does not get reset automatically.   Therefore, the loaded program does not initialize and then run.  The code seems to be stuck in FreeRTOS and not the code that has been loaded.  The program will only run after the chip is reset.  If you reset the chip, you loose connection to the debugger.

This article refers to the possible problem: https://esp32s3.com/getting-started.html

"NOTE! There is a ROM Mask bug in the current ESP32-S3 silicon that prevents the flashing process from resetting the board after flashing has finished, which means for now, you need to manually press the [RESET] button after flashing to manually reset the board so your code start running. Updated silicon from Espressif will fix this issue hopefully!"

I have spent the last 3 weeks trying to simply get connected to the custom board that we are using so that I can debug it.  The code for the firmware is simple.  Trying to get the stinking chip to work with seemingly different permutations of settings, configuration files and hidden calls for every minor variation of board and chip is beyond frustrating.  The documentation is extensive but I have yet to find an explanation that explains, in a strait forward way, what is needed in general to get a chip running.  There are extensive forum posts such as this that many wonderful people provide help.  I would like to find a concise, explanation of what is required to get basic things (such as debug) running.  The ESP32 is a great platform, unfortunately there are too many hardware variations with each requiring an obscure setting or tweak to run or debug.  Certainly different IDE's create difficulties not controlled by Espressif.  I would love to find a good reference including a concise summary of exactly what is needed to get the chip to communicate and operate in debug mode.  There are many references to configuration settings without reference to where the actual files are or are supposed to be, each explained as if you already know what they are referring to.  The complexity continues with settings that go in the platformio.ini file and proper driver requirements, launch.json settings, etc..  The complexity is understandable the frustration comes from trying to piece this together based on clued gleamed from help forums.  

I am sorry for the rant, but hopefully, I have missed something and hopefully the frustrations listed will be helpful in this post to locate a solution.  I have read extensive posts over the last few weeks with people having similar problems.  It would be great to find a definitive, high level reference for this information that could be supplemented by Espressiff's detailed web reference.

Thanks to those of you that are giving your time to help people such as myself.

Jason2866, 2023-06-15:
@bdewalch  The issue here is for the C3. Some time ago i managed to get the inbuilt JTAG debugger going with the C3.
"Something" has changed "somewhere" since debugging fails now.
Tried the suggestions from @valeros and ended in the same error as @mickeyl 
```
hans@MacBook-Air-von-Johann Tasmota % pio debug --interface=gdb -- -x .pioinit
Reading symbols from /Users/hans/Git/Tasmota/.pio/build/tasmota32c3cdc-ocd-luatos/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Open On-Chip Debugger v0.12.0-esp32-20230313-208-g3723ea7d (2023-06-15-10:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz
Info : tcl server disabled
Info : telnet server disabled
Info : esp_usb_jtag: serial (84:F7:03:3B:08:14)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] datacount=2 progbufsize=16
Error: Timed out after 5s waiting for busy to go low (abstractcs=0x10001102). Increase the timeout with riscv set_command_timeout_sec.
Error: Timed out after 5s waiting for busy to go low (abstractcs=0x10001102). Increase the timeout with riscv set_command_timeout_sec.
Error: [esp32c3] Fatal: Failed to read s0.
Warn : target esp32c3 examination failed
Info : starting gdb server for esp32c3 on pipe
Info : accepting 'gdb' connection from pipe
Error: Target not examined yet
Error executing event gdb-attach on target esp32c3:
/Users/hans/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "/Users/hans/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9
Warn : No symbols for FreeRTOS!
Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
(gdb) .pioinit:11: Error in sourced command file:
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
```

valeros, 2023-06-15:
@Jason2866 I see you're using OpenOCD v0.12, I'd recommend staying on v0.11 as we're experiencing a strange issue with v0.12 version https://github.com/espressif/openocd-esp32/issues/285

Jason2866, 2023-06-15:
@valeros Good point. Going down OCD releases, and try all

Jason2866, 2023-06-15:
Using version v.0.11 solves the problem :-)

[image]

bdewalch, 2023-06-15:
@Jason2866  and @valeros  thanks for the quick help!  Unfortunately I am already using OpenOCD v0.11 so that does not seem to be my problem.  

I am pretty sure that the chip does not reboot after loading the firmware (as noted in my post above).  I am running a simple blink program with serial output (see below).  The blinking only starts when I pull the reset pin to low and and I am unable to get any serial output from the chip.  
 
I am including the output from the debug console, platformio.ini and launch.json file contents.  There are two issues from the debug console that may be an indicator of least some of my problems:

1.   "Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED".    I have researched this one with no success.  I have used Zadig to change the drivers but have not found any combination that solved this issue.  I am using a USB cable connected directly to the D+,D- pins on the chip.

2.  "Warn : Transport "jtag" was already selected."  I am also have researched it without results and unsure of why this is happening or how significant it is.

**Here is the output when I run "PIO Debug":**

undefinedC:\Users\bdewalch\.platformio\packages\toolchain-riscv32-esp@8.4.0+2021r2-patch5\bin\riscv32-esp-elf-gdb.exe: warning: Couldn't determine a path for the index cache directory.
Reading symbols from c:\Solutions\Blink - Test Project\.pio\build\esp32-c3-devkitm-1\firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : esp_usb_jtag: serial (84:F7:03:32:DB:18)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on pipe
Info : accepting 'gdb' connection from pipe
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Found 8 triggers
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 52 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 152 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 52 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 152 KB
Info : Using flash bank 'esp32c3.irom' size 156 KB
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 52 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 152 KB
Info : Using flash bank 'esp32c3.drom' size 56 KB
0x400462dc in ?? ()
Error: FreeRTOS maximum used priority is unreasonably big, not proceeding: 227
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
** Programming Started **
Reset cause (3) - (Software core reset)
** Programming Started **
Info : PROF: Erased 16384 bytes in 430.002 ms
PROF: Erased 16384 bytes in 430.002 ms
Info : PROF: Data transferred in 278.982 ms @ 57.3514 KB/s
PROF: Data transferred in 278.982 ms @ 57.3514 KB/s
Info : PROF: Wrote 16384 bytes in 733.672 ms (data transfer time included)
** Programming Finished in 1582 ms **
** Verify Started **
PROF: Wrote 16384 bytes in 733.672 ms (data transfer time included)
** Programming Finished in 1582 ms **
[... 125 lines cut ...]
            "loadMode": "manual"
        }
    ]

**Here is the code that I am using to test with:**

#include <Arduino.h>

void setup()
{
    pinMode(8, OUTPUT);
    Serial.begin(115200);
    delay(5000);
    Serial.println("setup complete");
}

void loop()
{
    digitalWrite(8, HIGH);
    Serial.println("on");
    delay(1000);
    digitalWrite(8, LOW);
    Serial.println("off");
    delay(1000);
}

Jason2866, 2023-06-15:
@bdewalch To get stuff going with Windows is a pain. Sorry i am out here, stopped using Windows in 2021. Tried with MacOs and Linux (Ubuntu). Both worked.

bdewalch, 2023-06-16:
@Jason2866   Danke!  I agree Window can be a pain.  I strongly suspect that this is related to the ROM mask problem that I linked to above.  It may be that it is not possible to get the chip to reboot and stay connected to the debugger.  I am going to investigate using the serial (TX RX) ports on the chip and resort to good old fashion print statements for debugging.

bdewalch, 2023-06-20:
Update:

I am using the (TX RX) ports on the chip to successfully monitor output from print statements in the code.  After 3+ weeks of effort to use the USB JTAG debugging, I finally decided that the article mentioned here: https://esp32s3.com/getting-started.html  is probably correct and that the ESP32-C3 most likely has a rom bug that prevents it from automatically rebooting.  If you try to manually reboot to attempt to run the code that you are debugging, platformio looses the connection with the chip.

I hope this helps anyone with the same issue.  Thanks to those who helped me.

closed 2025-04-25 by stale[bot]