platformio/platform-espressif32 #1122: Freenove ESP32-S3-WROOM Board support

taunoe, 2023-06-15:
I can't find the board definition that fits the Freenove ESP32-S3-WROOM board.
Board documentation is here: [github.com/Freenove/Freenove_ESP32_S3_WROOM_Board](https://github.com/Freenove/Freenove_ESP32_S3_WROOM_Board)

valeros, 2023-06-16:
Hi @taunoe, accoring to the tutorial in that repository, you can use a generic `esp32-s3-devkitc-1` as an equivalent.

closed 2023-06-16 by valeros

Rhvarrier, 2023-07-03:
Hi @valeros, indeed `esp32-s3-devkitc-1` seems as equivalent except that the Freenove ESP32 S3 board has 8MB of PSRAM while `esp32-s3-devkitc-1` doesn't have PSRAM

taunoe, 2023-07-04:
I added these build flags to the platformio.ini file:
`build_flags =`
`    -DBOARD_HAS_PSRAM`
`    -mfix-esp32-psram-cache-issue`
`board_build.arduino.memory_type = dio_opi`

And my test script says that Total PSRAM: 8386295
Otherwise, it did not see anything.