platformio/platform-espressif32 #1123: Add board definition for YB-ESP32-S3-ETH dev board

yellobyte, 2023-06-16:
Please add board definition for **YB-ESP32-S3-ETH** development board (ESP32-S3, Ethernet, WiFi, FLASH, PSRAM, 2xUSB, LEDs).
  https://github.com/yellobyte/ESP32-DevBoards-Getting-Started/tree/main/boards/YB-ESP32-S3-ETH(YelloByte)
Thanks 
yellobyte

Board definition files for the 2 released variants attached:
[yb-esp32-s3-eth.zip](https://github.com/platformio/platform-espressif32/files/11776790/yb-esp32-s3-eth.zip)

Jason2866, 2023-06-17:
Actual ESP32 Arduino does NOT support the PHY W5500. It conflicts with SPI.
There is a open PR https://github.com/espressif/arduino-esp32/pull/7163
No support planned until ESP32 Arduino Core 3.0.0 is released

yellobyte, 2023-06-17:
Thanks. I'll reapply when this issue has been attended to by a future core version.

yellobyte, 2023-10-19:
According to https://github.com/espressif/arduino-esp32/pull/7163#issuecomment-1771171477 SPI Ethernet is now supported in 3.0.0.
Is there any chance now for my board https://github.com/yellobyte/ESP32-DevBoards-Getting-Started/tree/main/boards/YB-ESP32-S3-ETH(YelloByte) getting added to the official board list ?
Thanks
Board definition files are attached.
[yb-esp32-s3-eth.zip](https://github.com/platformio/platform-espressif32/files/13044974/yb-esp32-s3-eth.zip)

Jason2866, 2023-10-19:
Platformio crew adds boards only when the are already added in Arduino espressif github.
Furthermore there is no released core 3.0.0. Only an alpha version. There are no platformio release done for alpha or beta versions.

closed 2023-10-19 by yellobyte