platformio/platform-espressif32 #1126: [ESP32S3,MacOSM1] Built-in debugging packages not found for 'darwin_arm64'

vabatta, 2023-06-18:
Reading through [community@jtag-debugging-on-esp32-devkit/32609](https://community.platformio.org/t/jtag-debugging-on-esp32-devkit/32609) and https://github.com/espressif/esp-idf/issues/10381#issuecomment-1353944059 it seems that for M1 Macs I need to update the following packages:
- platformio/tool-openocd-esp32@^2.1200.20230419
- espressif/toolchain-xtensa-esp32s3@^12.2.0+20230208

But while changing the platform_packages with those signatures, I get the following:
```log
Resolving espidf dependencies...
Tool Manager: Installing espressif/^12.2.0+20230208
Error: Could not find the package with 'espressif/^12.2.0+20230208' requirements for your system 'darwin_arm64'
```

PlatformIO.ini
```ini
[env:espidf]
platform = espressif32
board = t-sim7080g-s3 ; this is custom board def with core:esp32, mcu:esp32s3 and variant:esp32s3
framework = espidf
upload_protocol = esp-builtin
upload_speed = 921600
upload_port = /dev/cu.usbmodem2101
monitor_speed = 115200
monitor_port = /dev/cu.usbmodem2101
build_flags = 
	-DBOARD_HAS_PSRAM
	-DCONFIG_SPIRAM_CACHE_WORKAROUND
board_build.partitions = partitions/custom.csv
platform_packages = 
	platformio/tool-openocd-esp32@^2.1200.20230419
	espressif/toolchain-xtensa-esp32s3@^12.2.0+20230208
build_type = debug
debug_tool = esp-builtin
debug_speed = 5000
debug_load_mode = manual
```

What am I missing?

vabatta, 2023-06-18:
It seems PlatformIO packages do not accept `+<version>` in packages, so using `espressif/toolchain-xtensa-esp32s3@^12.2.0` worked, leading to the issue: 

```
Processing espidf (platform: espressif32; board: t-sim7080g-s3; framework: espidf)
-------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/t-sim7080g-s3.html
PLATFORM: Espressif 32 (6.3.1) > LilyGO T-SIM7080G-S3
HARDWARE: ESP32S3 240MHz, 512KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50002.230601 (5.0.2) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.9.0 
 - tool-openocd-esp32 @ 2.1200.20230419 (12.0) 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
-- Found Git: /opt/homebrew/bin/git (found version "2.41.0") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/vabatta/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /Users/vabatta/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /Users/vabatta/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Users/vabatta/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-g++
-- Check for working CXX compiler: /Users/vabatta/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (5.0.2)
-- Project sdkconfig file /Users/vabatta/Projects/CubeLock-PIO/sdkconfig.espidf
-- Configuring incomplete, errors occurred!
See also "/Users/vabatta/Projects/CubeLock-PIO/.pio/build/espidf/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /Users/vabatta/.platformio/packages/framework-espidf/tools/cmake/tool_version_check.cmake:36 (message):
  

  Tool doesn't match supported version from list ['esp-2022r1-11.2.0']:
  /Users/vabatta/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc

  Please try to run 'idf.py fullclean' to solve it.

Call Stack (most recent call first):
  /Users/vabatta/.platformio/packages/framework-espidf/components/esp_common/project_include.cmake:10 (check_expected_tool_version)
  /Users/vabatta/.platformio/packages/framework-espidf/tools/cmake/build.cmake:380 (include)
  /Users/vabatta/.platformio/packages/framework-espidf/tools/cmake/build.cmake:599 (__build_process_project_includes)
  /Users/vabatta/.platformio/packages/framework-espidf/tools/cmake/project.cmake:447 (idf_build_process)
  CMakeLists.txt:4 (project)

=============================== [FAILED] Took 2.45 seconds ===============================

Environment    Status    Duration
-------------  --------  ------------
espidf         FAILED    00:00:02.446
========================== 1 failed, 0 succeeded in 00:00:02.446 ==========================
```

Jason2866, 2023-06-18:
It does work with
```
platform = espressif32 @ ^6.3.1
```
See https://github.com/platformio/platform-espressif32/issues/1113#issuecomment-1592806919

vabatta, 2023-06-18:
@Jason2866 I'm running already on espressif32 @ ^6.3.1. If I try to run Debug (without uploading) this is the result on debug console:
```
Reading symbols from /Users/vabatta/Projects/CubeLock-PIO/.pio/build/espidf/firmware.elf...
I'm sorry, Dave, I can't do that.  Symbol format `elf32-xtensa-le' unknown.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Info : esp_usb_jtag: serial (34:85:18:6C:72:E4)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000001
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : starting gdb server for esp32s3.cpu0 on pipe
Info : accepting 'gdb' connection from pipe
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Set GDB target to 'esp32s3.cpu0'
Info : Flash mapping 0: 0x10020 -> 0x3c060020, 104 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 338 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c060020, 104 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 338 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 340 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c060020, 104 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 338 KB
Info : Using flash bank 'esp32s3.cpu0.drom' size 108 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
```

So by looking up, I have an error `I'm sorry, Dave, I can't do that.  Symbol format `elf32-xtensa-le' unknown.` which was the initial research on how to get this working.

vabatta, 2023-06-18:
Running `pio debug --interface=gdb -- -x .pioinit` I get:
```
Reading symbols from /Users/vabatta/Projects/CubeLock-PIO/.pio/build/espidf/firmware.elf...
I'm sorry, Dave, I can't do that.  Symbol format `elf32-xtensa-le' unknown.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Info : esp_usb_jtag: serial (34:85:18:6C:72:E4)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000001
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : starting gdb server for esp32s3.cpu0 on pipe
Info : accepting 'gdb' connection from pipe
Exception in callback DebugSubprocessProtocol.pipe_data_received(2, b'Open On-Chi...n from pipe\n')
handle: <Handle DebugSubprocessProtocol.pipe_data_received(2, b'Open On-Chi...n from pipe\n')>
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/python@3.11/3.11.4/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/Users/vabatta/.platformio/penv/lib/python3.11/site-packages/platformio/debug/process/base.py", line 43, in pipe_data_received
    pipe_to_cb[fd](data)
  File "/Users/vabatta/.platformio/penv/lib/python3.11/site-packages/platformio/debug/process/gdb.py", line 172, in stderr_data_received
    super().stderr_data_received(data)
  File "/Users/vabatta/.platformio/penv/lib/python3.11/site-packages/platformio/debug/process/base.py", line 128, in stderr_data_received
    print(
BlockingIOError: [Errno 35] write could not complete without blocking
--- Logging error ---
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Set GDB target to 'esp32s3.cpu0'
Info : Flash mapping 0: 0x10020 -> 0x3c060020, 104 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 338 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c060020, 104 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 338 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 340 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c060020, 104 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 338 KB
Info : Using flash bank 'esp32s3.cpu0.drom' size 108 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
pio debug --interface=gdb -- -x .pioinit  61.47s user 31.72s system 416% cpu 22.398 total

```

Jason2866, 2023-06-18:
The S3 inbuilt JTAG is not reliable working. Don't know what's the reason why the same setup  works (for me) and for others it does not.

vabatta, 2023-06-18:
What do you mean by "a second or third time"? I've been trying to start it in all different "Debug" launch modes, and by changing the dependencies. No matter what combinations of `platform` or `platform_packages` and `debug launch modes` I use, the debug output is always the one above so far.

Jason2866, 2023-06-18:
How is the USB port from the S3 connected to the Mac?
Maybe this is relevant? I am using a HP G5 Dockingstation which provides USB Ports.

vabatta, 2023-06-18:
What I've discovered is that using `Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)` it gives the error in the base.py file, while using `v0.12.0-esp32-20230419` gives no error besides the missing symbols.
This still stops debugging right when reaching the end of the log, meaning I never get to attach to the device.

```
Reading symbols from /Users/vabatta/Projects/CubeLock-PIO/.pio/build/espidf/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = custom
I'm sorry, Dave, I can't do that.  Symbol format `elf32-xtensa-le' unknown.
PlatformIO: Initializing remote target...
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz
Info : tcl server disabled
Info : telnet server disabled
Info : esp_usb_jtag: serial (34:85:18:6C:72:E4)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on pipe
Info : accepting 'gdb' connection from pipe
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000001
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c060020, 104 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 338 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c060020, 104 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 338 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 340 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c060020, 104 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 338 KB
Info : Using flash bank 'esp32s3.cpu0.drom' size 108 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
```

ini
```
platform_packages = 
	platformio/tool-openocd-esp32@^2.1200.20230419
debug_tool = custom
debug_server =
  $PLATFORMIO_CORE_DIR/packages/tool-openocd-esp32/bin/openocd
  -f
  $PLATFORMIO_CORE_DIR/packages/tool-openocd-esp32/share/openocd/scripts/board/esp32s3-builtin.cfg
debug_init_break = tbreak app_main
```

@Jason2866 Yes, I've been trying all options every time I do a change.

vabatta, 2023-06-18:
Actually, if I reset the board and run the debug without uploading, even on `v0.12` I get errors:

```
Reading symbols from /Users/vabatta/Projects/CubeLock-PIO/.pio/build/espidf/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = custom
PlatformIO: Initializing remote target...
I'm sorry, Dave, I can't do that.  Symbol format `elf32-xtensa-le' unknown.
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz
Info : tcl server disabled
Info : telnet server disabled
Info : esp_usb_jtag: serial (34:85:18:6C:72:E4)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on pipe
Info : accepting 'gdb' connection from pipe
Info : [esp32s3.cpu0] Target halted, PC=0x4037EC9A, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40383AB7, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Memory protection is enabled. Reset target to disable it...
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Target halted, PC=0x4037EB02, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2434, debug_reason=00000001
Error: Failed to get flash maps (5)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2434, debug_reason=00000001
Exception in callback DebugSubprocessProtocol.pipe_data_received(2, b"Info : [esp...on=00000001\n")
handle: <Handle DebugSubprocessProtocol.pipe_data_received(2, b"Info : [esp...on=00000001\n")>
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/python@3.11/3.11.4/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/Users/vabatta/.platformio/penv/lib/python3.11/site-packages/platformio/debug/process/base.py", line 43, in pipe_data_received
    pipe_to_cb[fd](data)
  File "/Users/vabatta/.platformio/penv/lib/python3.11/site-packages/platformio/debug/process/gdb.py", line 172, in stderr_data_received
    super().stderr_data_received(data)
  File "/Users/vabatta/.platformio/penv/lib/python3.11/site-packages/platformio/debug/process/base.py", line 128, in stderr_data_received
    print(
BlockingIOError: [Errno 35] write could not complete without blocking
--- Logging error ---
Info : [esp32s3.cpu0] Target halted, PC=0x403B2434, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 0 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 0 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2434, debug_reason=00000001
Error: Failed to get flash maps (5)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2434, debug_reason=00000001
Info : Using flash bank 'esp32s3.cpu0.irom' size 0 KB
Exception in callback DebugSubprocessProtocol.pipe_data_received(2, b"Info : [esp...' size 0 KB\n")
handle: <Handle DebugSubprocessProtocol.pipe_data_received(2, b"Info : [esp...' size 0 KB\n")>
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/python@3.11/3.11.4/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/Users/vabatta/.platformio/penv/lib/python3.11/site-packages/platformio/debug/process/base.py", line 43, in pipe_data_received
    pipe_to_cb[fd](data)
  File "/Users/vabatta/.platformio/penv/lib/python3.11/site-packages/platformio/debug/process/gdb.py", line 172, in stderr_data_received
    super().stderr_data_received(data)
  File "/Users/vabatta/.platformio/penv/lib/python3.11/site-packages/platformio/debug/process/base.py", line 128, in stderr_data_received
    print(
BlockingIOError: [Errno 35] write could not complete without blocking
--- Logging error ---
Info : [esp32s3.cpu0] Target halted, PC=0x403B2434, debug_reason=00000001
Error: Failed to get flash maps (5)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2434, debug_reason=00000001
Info : Using flash bank 'esp32s3.cpu0.drom' size 0 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
```

I have no clue on what's wrong here.

Jason2866, 2023-06-18:
OpenOCD v0.12 does not work. That's why Platformio uses v0.11

vabatta, 2023-06-18:
I just tested on a native espidf project with the same code and it works. It uses `Open On-Chip Debugger v0.11.0-esp32-20221026 `, any chance that it could be the issue seen that PIO uses `v0.11.0-esp32-20220706`?

Jason2866, 2023-06-19:
Maybe, but for me the `v0.11.0-esp32-20220706` does work.

vabatta, 2023-06-19:
Yep, doesn't work still so must be something else. Is there a way to see the full log of GDB commands issued from PIO?

valeros, 2023-06-19:
Thanks @vabatta, this line looks particularly suspicious:

> Memory protection is enabled. Reset target to disable it...

Is any sort of memory protection indeed enabled? 

> I'm sorry, Dave, I can't do that.  Symbol format `elf32-xtensa-le' unknown.

Hmm, let's try a workaround with a standalone GDB package as suggested in `espressif/esp-idf#10381` . Please download `https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v11.2_20220823/xtensa-esp-elf-gdb-11.2_20220823-aarch64-apple-darwin21.1.tar.gz` and extract it somewhere on your machine. Add an extra scripts that overrides the default GDB executable:

```
platform = espressif32@^6.3.1
board = t-sim7080g-s3
framework = espidf
...
extra_scripts = 
    replace_gdb.py
```
`replace_gdb.py`:
```
Import ("env")

env.Replace(GDB="/YOUR/PATH/TO/EXTRACTED/GDB/bin/xtensa-esp32s3-elf-gdb")
```

Clean, run and debug again.

vabatta, 2023-06-19:
> Is any sort of memory protection indeed enabled?

According to [espidf configs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/security.html) there is and seems to be enabled by default (I haven't set this manually).
```
CONFIG_ESP_SYSTEM_MEMPROT_FEATURE=y
CONFIG_ESP_SYSTEM_MEMPROT_FEATURE_LOCK=y
```

Despite having the memory protection ON, with the given GDB package I was able to **successfully** connect to the device and debug.

<details>
<summary>Debug log</summary>

```log
Reading symbols from /Users/vabatta/Projects/CubeLock-PIO/.pio/build/espidf/firmware.elf...
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Warning: 'flushregs', an alias for the command 'maintenance flush register-cache', is deprecated.
Use 'maintenance flush register-cache'.
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Info : esp_usb_jtag: serial (34:85:18:6C:72:E4)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on pipe
Info : accepting 'gdb' connection from pipe
Info : [esp32s3.cpu0] Target halted, PC=0x4037979A, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x4037979A, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Target halted, PC=0x4037979A, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x4037979A, debug_reason=00000000
Memory protection is enabled. Reset target to disable it...
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
[... 52 lines cut ...]
Info : [esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40041A76, debug_reason=00000000
[esp32s3.cpu1] Target halted, PC=0x40041A76, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
[esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Target halted, PC=0x420014D4, debug_reason=00000001
[esp32s3.cpu0] Target halted, PC=0x420014D4, debug_reason=00000001
Info : Set GDB target to 'esp32s3.cpu0'
Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x4037979A, debug_reason=00000000
[esp32s3.cpu1] Target halted, PC=0x4037979A, debug_reason=00000000
Info : Detected FreeRTOS version: (10.4.3)
[New Thread 1070178992]
[New Thread 1070182768]
[New Thread 1070180880]
[New Thread 1070169772]
[New Thread 1070174220]
[New Thread 1070168140]
[Switching to Thread 1070178992]

Thread 2 "main" hit Temporary breakpoint 1, app_main () at src/main.c:21
21	{
```
</details>

mhaberler, 2023-06-28:
watching closely as I'm in the same pits 

a working combination of toolchain, openocd and gdb for ESP32 S3 working with platformio would be great

let me know if I can help

paltforms: Mac M1 + amd64 linux

vabatta, 2023-07-20:
@mhaberler it actually works if you follow the instructions given by @valeros in https://github.com/platformio/platform-espressif32/issues/1126#issuecomment-1597690513

@valeros I guess that package needs to be update in this repo to make it work out of the box

mhaberler, 2023-07-21:
this worked on the command line for me, but I would prefer to stay with VSCode

patching up the pio gdb path via replace_gdb.py causes the  VSCode platformio plugin to generate a launch.json file which has the wrong toolchain paths - the gdb path is taken and set as toolchainBinDir:

https://github.com/platformio/platformio-core/blob/5c9b373b656f088b4fd506f49512e1880168da4f/platformio/project/integration/tpls/vscode/.vscode/launch.json.tpl#L20

next I tried `gdb_path = /path/t/replacement/gdb` 
this makes pio create a soft link to the replacement gdb

however this stalls as well as the replacement gdb wrapper script calls the wrong binary

what DID work was patching up the wrapper script or hard-linking the replacement gdb into the .platformio toolchain subdirectory

vabatta, 2023-07-22:
I did exactly what was written in the comment and in visual studio code, so I was able to hook up the debugger from within it and use it. 
I’m not sure why I won’t work for you, but if you replace ‘gdb_path’ it should automatically pick it up even if the file is regenerated.

mhaberler, 2023-07-22:
things work fine on the M1 with the stock -patch5 toolchain and openocd - so issue closed so far as the subject of this issue goes, no need to replace the gdb path there

my issue was with Linux, where I got stuck with the stock gdb requiring  a Python2.7 shared lib which does not exist on bookworkm anymore, so I tried the replace_gdb.py and 'gdb_path = /replacement/gdb' methods

now in both cases VSCode will regenerate the launch.json file with the whole toolChainDir now pointing to the directory where the replacement gdb lives in, and from there on things go wrong because after launch ONLY the replacement gdb is found

so I gave up on the replacement idea, built the Python2.7 shared lib as outlined and things are fine on linux as well - same setup as on M1

closed 2025-02-25 by stale[bot]