platformio/platform-espressif32 #1128: ESP32-S3 undefined reference to `_bt_bss_start'

Kampi, 2023-06-20:
I´m compiling firmware for an ESP32-S3 by using the following configuration:

```
[platformio]
src_dir = src
default_envs = debug

# Common build environment that is used for each of the tasks
[env]
board = esp32-s3-devkitc-1

platform = espressif32
framework = espidf

# Flash memory and partition selection
# NOTE: You have to edit the parameter `CONFIG_PARTITION_TABLE_CUSTOM_FILENAME` in `sdkconfig` too!

# 16 MB module version
#board_upload.flash_size = 16MB
#board_upload.maximum_size = 16777216
#board_build.partitions = partitions.csv

# 8 MB module version
board_upload.flash_size = 8MB
board_upload.maximum_size = 8388608
board_build.partitions = partitions.csv

board_build.filesystem = littlefs
board_build.embed_txtfiles =
    files/Cert_Server.pem

# Settings for serial upload
upload_speed = 921600
upload_port = COM12

# Settings for serial console
monitor_speed = 115200
monitor_port = COM12
monitor_filters = esp32_exception_decoder

# Additional packages
platform_packages =

# Aditional build scripts for all environments to run before (`pre`) and after (`post`) the main build script
extra_scripts =
    pre:scripts/Clean.py

# Common build flags for all environments
build_flags =

lib_deps =

[env:debug]
build_type = debug

build_flags =
    ${env.build_flags}
    -D DEBUG

lib_deps =
    ${env.lib_deps}
    ${env.lib_deps}

[env:release]
build_type = release

build_flags =
    ${env.build_flags}
    -D RELEASE

lib_deps =
    ${env.lib_deps}
    ${env.lib_deps}

extra_scripts =
    ${env.extra_scripts}
    post:scripts/Release.py
```

And the following tools:
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.3.1) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50002.230601 (5.0.2)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 11.2.0+2022r1
 - toolchain-xtensa-esp32s3 @ 11.2.0+2022r1
 ```

I receive the following linking error randomly at the end of the build process:

```
...platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\debug\esp-idf\bt\libbt.a(bt.o):(.literal.esp_bt_mem_release+0x0): undefined reference to `_bt_bss_end'
...platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\debug\esp-idf\bt\libbt.a(bt.o):(.literal.esp_bt_mem_release+0x8): undefined reference to `_bt_bss_start'
```

The code compiles successfully for an ESP32 and sometimes also successfully for an ESP32-S3.

Jason2866, 2023-06-20:
Filesystem uses LittleFS. Are you using the FS in your code?
Have you added LittleFS as component? It is NOT not included in esp-idf by default.

valeros, 2023-06-20:
Probably not a PlatformIO issue, see https://github.com/espressif/esp-idf/issues/10427

Kampi, 2023-06-20:
> Filesystem uses LittleFS. Are you using the FS in your code? Have you added LittleFS as component? It is NOT not included in esp-idf by default.

Hi,

I have removed the `board_build.filesystem = littlefs` directive,  but the issue stays. I check the solution from @valeros

Kampi, 2023-06-20:
Hi,

the solution from @valeros work. I add

*/main/ESP32-S3_Fix.lf*
```
# This fixes an ESP-IDF 4.4.4/5.0.1 linker issue on incremental builds
#   "undefined reference to '_bt_bss_start'" for Bluetooth component library function 'esp_bt_mem_release'
# References:
#   https://github.com/espressif/esp-idf/issues/10427
#   https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/linker-script-generation.html

[sections:bt_bss_fix]
entries:
    .bss+

[scheme:bt_extram_bss_fix]
entries:
    bt_bss_fix -> extern_ram

[mapping:bt_fix]
archive: libbt.a
entries:
    if ESP_ALLOW_BSS_SEG_EXTERNAL_MEMORY = y:
        * (extram_bss)
```

and 

*/main/CMakeLists.txt*
```
if(IDF_TARGET STREQUAL "esp32s3")
    idf_component_register(SRCS ${app_sources}
                        REQUIRES json
                        LDFRAGMENTS "ESP32-S3_Fix.lf")
else()
    idf_component_register(SRCS ${app_sources}
                        REQUIRES json
                        )
endif()
```

to my project and the issue is gone.

Kampi, 2023-06-21:
I have to correct my answer. The issue is still present.

Kampi, 2023-06-26:
Hi,

it´s not that easy by using windows. That´s the reason for using PIO. I try to get it runnable in docker.

Jason2866, 2023-06-30:
Tried `CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=n`?

tttonyyy, 2023-07-31:
@Kampi have you tried my original solution?  It's not quite the same as the one you posted here but has worked for me in 4.4.4 and 4.4.5 (still present).

https://github.com/espressif/esp-idf/issues/10427#issuecomment-1470068645

closed 2024-02-05 by Kampi