platformio/platform-espressif32 #1129: Is there anyway to use IDF v5.1-rc1 pre-release with platformio?

torntrousers, 2023-06-20:
Hi using the platformio idf framework support (not arduino) is there any way to use the IDF v5.1-rc1 pre-release https://github.com/espressif/esp-idf/releases/tag/v5.1-rc1?

TIA!

Jason2866, 2023-06-20:
Tried, built a framework from latest (branch release/v5.1) solved all minor problems but idk how to solve the linker error at the end of the compile run,
https://github.com/Jason2866/platform-espressif32/actions/runs/5244591812/jobs/9470823705
So short answer: **No**

torntrousers, 2023-06-22:
Ok, thanks for the investigation @Jason2866

torntrousers, 2023-07-01:
IDF has done the official 5.1 release now - https://github.com/espressif/esp-idf/releases/tag/v5.1 - does that make this any more possible yet?

Jason2866, 2023-07-27:
@torntrousers The branch mentioned above does now work with IDF 5.1 :-)
The apha version of Arduino 3.0 is working too.

valeros, 2023-12-06:
Support for IDF v5.1 was added in [v6.4.0](https://github.com/platformio/platform-espressif32/releases/tag/v6.4.0).

closed 2023-12-06 by valeros