platformio/platform-espressif32 #1131: Please can you create a framework-espidf for v4.4.5 that was recently released.

bensuffolk, 2023-06-22:
I'm not sure where the framework releases live, but the latest one I can find in the repository for the v4.4 branch of esp-idf is framework-espidf@3.40404.0 which is V4.4.4

V4.4.5 was released a couple of weeks ago and I'd like to know if we can get a new release to match this.

valeros, 2023-06-23:
Hi @bensuffolk, PlatformIO supports the latest stable v5.0.2 https://github.com/espressif/esp-idf/releases/tag/v5.0.2

bensuffolk, 2023-06-23:
Sorry in what way is this completed?

I am unable to use esp-idf V5.0 as it has some bugs in it that stop my project working (I have been trying for the last couple of weeks to make it work).

I need to stick with the 4.4 branch for the moment until I raise the bugs and see if they can be fixed. I would like the recent bug fixes from V4.4.5 though

Regards

Ben

valeros, 2023-06-23:
A hot fix release with support for IDF v4.4.5 has been published https://github.com/platformio/platform-espressif32/releases/tag/v5.4.0
```ini
[env:esp32dev]
platform = espressif32 @ ^5.4.0
framework = espidf
board = esp32dev
```

bensuffolk, 2023-06-23:
Except unfortunately that is downloading IDF V4.4.3 not V4.4.5

valeros, 2023-06-23:
Please share your logs here

bensuffolk, 2023-06-23:
I started by deleting all packages and platforms from the .platformio directory so it had to download everything fresh:

```
Platform Manager: Installing espressif32 @ ^5.4.0
Unpacking  [####################################]  100%
Platform Manager: espressif32@5.4.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch5 has been installed!
Tool Manager: Installing platformio/toolchain-esp32ulp @ ~1.23500.0
Unpacking  [####################################]  100%
Tool Manager: toolchain-esp32ulp@1.23500.220830 has been installed!
Tool Manager: Installing platformio/framework-espidf @ ~3.40405.0
Unpacking  [####################################]  100%          
Tool Manager: framework-espidf@3.40405.0 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40400.0
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40400.0 has been installed!
Tool Manager: Installing platformio/tool-cmake @ ~3.16.0
Unpacking  [####################################]  100%          
Tool Manager: tool-cmake@3.16.4 has been installed!
Tool Manager: Installing platformio/tool-ninja @ ^1.7.0
Unpacking  [####################################]  100%
Tool Manager: tool-ninja@1.9.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40502.0
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40502.0 has been installed!
```

If I look at versions.txt from framework-espidf is shows  4.4.3

Build logs show its using the 3.40405.0 framework
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 (5.4.0) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 520KB RAM, 16MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.40405.0 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
```

Running the app shows it built using 4.4.3

```
I (0) cpu_start: App cpu up.
I (1120) cpu_start: Compile time:     Jun 23 2023 16:54:20
I (1127) cpu_start: ELF file SHA256:  8db7a59eb66aa273...
I (1133) cpu_start: ESP-IDF:          4.4.3
I (1138) cpu_start: Min chip rev:     v3.0
I (1142) cpu_start: Max chip rev:     v3.99 
I (1147) cpu_start: Chip rev:         v3.0
```

valeros, 2023-06-23:
Seems like there is an outdated `version.txt` file in the root of the new package. Please delete the broken `framework-espidf` package somewhere here `c:\Users\YOUR_USER\.platformio\packages` or `~/.platformio/packages` if you're on Unix and try again.

bensuffolk, 2023-06-23:
Its reporting as 4.4.5 now.

Thanks

closed 2023-06-23 by bensuffolk