platformio/platform-espressif32 #1132: Esp32-C3 EEprom fail

oddwatcher, 2023-06-23:
I'm using the official preference library to store some caliburation data of mpu 6050 , but the preference lib is returning errors, don't know how to fix, tired to change the platform from default espressif32 to this git but still not working the code and serial monitor is as follows :
```
#include <Wire.h>
#include <mpu_6050_dmp.h>
#include <VL53L0X.h>
#include <BleGamepad.h>
#include <Preferences.h>
// Wiring
#define Debug
#define SDA 4     //        yellow  wire
#define SCL 5     //        orange  wire
#define mpuINT 11 //    blue    wire
#define LED1 12
#define LED2 13
mpudata mpu1;
MPU6050 mpu;
VL53L0X sensor;
BleGamepad bleGamepad;
Preferences mpupref;

volatile bool mpuInterrupt = false;
// ================================================================
// ===                  Interrupt Service Routine               ===
// ================================================================

void dmpDataReady()
{
    mpuInterrupt = true;
    digitalWrite(LED1, HIGH);
}
// ================================================================
// ===                      INITIAL SETUP                       ===
// ================================================================

void setup()
{
    // join I2C bus (I2Cdev library doesn't do this automatically)

    Wire.begin(SDA, SCL);
    Wire.setClock(400000); // 400kHz I2C clock. Comment this line if having compilation difficulties
    // initialize serial communication
    Serial.begin(115200);

    // Sensor MPU
    mpustartup(&mpu1, &mpu, dmpDataReady, mpuINT);

    // Sensor ToF
    sensor.setTimeout(500);
    if (!sensor.init())
    {
#ifdef Debug
        Serial.println("Failed to detect and initialize sensor!");
#endif
    }
    // Long Range mode
    //  lower the return signal rate limit (default is 0.25 MCPS)
    sensor.setSignalRateLimit(0.1);
    // increase laser pulse periods (defaults are 14 and 10 PCLKs)
    sensor.setVcselPulsePeriod(VL53L0X::VcselPeriodPreRange, 18);

    // setup BLE for gamepad HID
    bleGamepad.begin();

    // start datastorage
    mpupref.begin("mpuprefdata", false);
    bool mpuinit = mpupref.isKey("init");
    if (mpuinit == false)
    {
        mpupref.putBool("init", true);
        mpu.CalibrateAccel();
        mpu.CalibrateAccel();
        mpu.GetActiveOffsets();
        mpupref.putFloat("o0", mpu.offsets[0]);
        mpupref.putFloat("o1", mpu.offsets[1]);
        mpupref.putFloat("o2", mpu.offsets[2]);
        mpupref.putFloat("o3", mpu.offsets[3]);
        mpupref.putFloat("o4", mpu.offsets[4]);
        mpupref.putFloat("o5", mpu.offsets[5]);
    }
    else
    {
        mpu.setXGyroOffset(mpupref.getFloat("o3"));
        mpu.setYGyroOffset(mpupref.getFloat("o4"));
        mpu.setZGyroOffset(mpupref.getFloat("o5"));
        mpu.setXAccelOffset(mpupref.getFloat("o0"));
        mpu.setYAccelOffset(mpupref.getFloat("o1"));
        mpu.setZAccelOffset(mpupref.getFloat("o2"));
    }
    mpupref.end();
#ifdef Debug
    mpu.PrintActiveOffsets();
#endif

    // start LEDs
    pinMode(LED1, OUTPUT);
    pinMode(LED2, OUTPUT);
    digitalWrite(LED1, LOW);
    digitalWrite(LED2, LOW);
    delay(1000); // wait till the serial mon start
}

// ================================================================
// ===                    MAIN PROGRAM LOOP                     ===
// ================================================================
int distance = 0;
int16_t joylocations[3];
void loop()
{
    // MPU polling
    if (mpuInterrupt == true)
    {
        mpudataupdate(&mpu1, &mpu); // will use mpu instance to update data in mpudata struct
        mpuInterrupt = false;       // this method is to prevent reading data from fifo when there is no data ready, in this case no vaild data will return

#ifdef Debug
        Serial.print("mpu data:");
        Serial.print(mpu1.ypr[0]);
        Serial.print('\t');
        Serial.print(mpu1.ypr[1]);
        Serial.print('\t');
        Serial.println(mpu1.ypr[2]);
#endif
        digitalWrite(LED1, LOW);
    }
    // laser distance polling
    distance = sensor.readRangeSingleMillimeters();
    if (sensor.timeoutOccurred())
    {
#ifdef Debug
        Serial.print(" TIMEOUT -- Resetting Sensor");
#endif
        if (!sensor.init())
        {
#ifdef Debug
            Serial.println("Failed to detect and initialize sensor!");
#endif
        }
    }
    // ble data updating
    joylocations[0] = (int16_t)(mpu1.ypr[0] / 180) * 32767;
    joylocations[1] = (int16_t)(mpu1.ypr[1] / 180) * 32767;
    joylocations[2] = (int16_t)(mpu1.ypr[2] / 180) * 32767;
#ifdef Debug
    Serial.print("joy loc:");
    Serial.print(joylocations[0]);
    Serial.print('\t');
    Serial.print(joylocations[1]);
    Serial.print('\t');
    Serial.println(joylocations[2]);
    Serial.print("distance :");
    Serial.println(distance);
#endif
    bleGamepad.setAxes(joylocations[0], joylocations[1], joylocations[2]);
    delay(20);
}
```

the error is 
```
-1162.00000,    581.00000,      1788.00000,     220.00000,      76.00000,       -85.00000

[   713][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: o3 NOT_FOUND
[   714][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: o4 NOT_FOUND
[   718][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: o5 NOT_FOUND
[   726][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: o0 NOT_FOUND
[   734][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: o1 NOT_FOUND
[   741][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: o2 NOT_FOUND
-1.00000,       -1.00000,       -1.00000,       -1.00000,       -1.00000,       -1.00000

joy loc:0       0       0
distance :8190
joy loc:0       0       0
distance :8190
joy loc:0       0       0
distance :8190

oddwatcher, 2023-06-23:
the pio.ini is as follows i'm using a LUATOS CORE-ESP32-C3 board, not the offical kit.
'''
[env:esp32-c3-devkitm-1]
platform =  espressif32 @ ^6.0.1
board = esp32-c3-devkitm-1
framework = arduino
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
monitor_speed = 115200
upload_speed = 3000000
board_build.flash_mode = dio
lib_deps = 
	pololu/VL53L0X@^1.3.1
	h2zero/NimBLE-Arduino@^1.4.0
'''

valeros, 2023-06-23:
Hi @oddwatcher ! Does it work in the Arduino IDE?

closed 2025-04-25 by stale[bot]