platformio/platform-espressif32 #1136: heltec_wifi_kit_32_V3 board ID not recognized on fresh PlatformIO core install on ubuntu-latest GitHub Actions

rbergen, 2023-06-26:
In the context of a project I'm working on ([NightDriverStrip](https://github.com/PlummersSoftwareLLC/NightDriverStrip)), we're running into an issue that the 'heltec_wifi_kit_32_V3' board ID is not recognized on a clean install of PlatformIO core.

The [GitHub Action workflow](https://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/.github/workflows/CI.yml) performs a clean install of PlatformIO using the recommended approach (i.e. the curl downloaded Python script). All projects that are defined in [the project's platformio.ini](https://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/platformio.ini) build, except for the one based on heltec_wifi_kit_32_V3. The workflow fails while resolving dependencies (`pio pkg install --no-save`) with the following error in the log:

```text
Resolving heltecv3demo dependencies...
Error: Unknown board ID 'heltec_wifi_kit_32_V3'
Error: Process completed with exit code 1.
```

valeros, 2023-06-26:
Hi @rbergen, please share here the entire log or at least the part related to the platform installation process.

rbergen, 2023-06-26:
Certainly. Log of installing PlatformIO:

```log
python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
  echo "/home/runner/.platformio/penv/bin" >> $GITHUB_PATH
  shell: /usr/bin/bash -e {0}
  env:
    ENABLE_PAGES_PUSH: false
    pythonLocation: /opt/hostedtoolcache/Python/[3](https://github.com/rbergen/NightDriverStrip/actions/runs/5375276242/jobs/9751558524#step:4:3).11.[4](https://github.com/rbergen/NightDriverStrip/actions/runs/5375276242/jobs/9751558524#step:4:4)/x64
    PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.11.4/x64/lib/pkgconfig
    Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.4/x64
    Python2_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.4/x64
    Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.4/x64
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.11.4/x64/lib
Installer version: 1.1.3
Platform: Linux-[5](https://github.com/rbergen/NightDriverStrip/actions/runs/5375276242/jobs/9751558524#step:4:5).15.0-1040-azure-x8[6](https://github.com/rbergen/NightDriverStrip/actions/runs/5375276242/jobs/9751558524#step:4:6)_64-with-glibc2.35
Python version: 3.11.4 (main, Jun  [7](https://github.com/rbergen/NightDriverStrip/actions/runs/5375276242/jobs/9751558524#step:4:7) 2023, 11:01:02) [GCC 11.3.0]
Python path: /opt/hostedtoolcache/Python/3.11.4/x64/bin/python
Creating a virtual environment at /home/runner/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Downloading platformio-6.1.7-py3-none-any.whl (41[8](https://github.com/rbergen/NightDriverStrip/actions/runs/5375276242/jobs/9751558524#step:4:8) kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 41[9](https://github.com/rbergen/NightDriverStrip/actions/runs/5375276242/jobs/9751558524#step:4:9).0/419.0 kB 5.3 MB/s eta 0:00:00
Collecting bottle==0.12.* (from platformio)
  Downloading bottle-0.12.25-py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.2/90.2 kB 31.6 MB/s eta 0:00:00
Collecting click<9,>=8.0.4 (from platformio)
  Downloading click-8.1.3-py3-none-any.whl (96 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 96.6/96.6 kB 35.5 MB/s eta 0:00:00
Collecting colorama (from platformio)
  Downloading colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting marshmallow==3.* (from platformio)
  Downloading marshmallow-3.19.0-py3-none-any.whl (49 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 49.1/49.1 kB 17.5 MB/s eta 0:00:00
Collecting pyelftools<1,>=0.27 (from platformio)
  Downloading pyelftools-0.29-py2.py3-none-any.whl (174 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 174.3/174.3 kB 48.6 MB/s eta 0:00:00
Collecting pyserial==3.5.* (from platformio)
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 36.4 MB/s eta 0:00:00
Collecting requests==2.* (from platformio)
  Downloading requests-2.31.0-py3-none-any.whl (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.6/62.6 kB 27.6 MB/s eta 0:00:00
Collecting urllib3<2 (from platformio)
  Downloading urllib3-1.26.16-py2.py3-none-any.whl (143 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 143.1/143.1 kB 51.5 MB/s eta 0:00:00
Collecting semantic-version==2.[10](https://github.com/rbergen/NightDriverStrip/actions/runs/5375276242/jobs/9751558524#step:4:10).* (from platformio)
  Downloading semantic_version-2.10.0-py2.py3-none-any.whl (15 kB)
Collecting tabulate==0.9.* (from platformio)
  Downloading tabulate-0.9.0-py3-none-any.whl (35 kB)
Collecting aiofiles==23.1.* (from platformio)
  Downloading aiofiles-23.1.0-py3-none-any.whl (14 kB)
Collecting ajsonrpc==1.* (from platformio)
  Downloading ajsonrpc-1.2.0-py3-none-any.whl (22 kB)
Collecting starlette==0.26.* (from platformio)
  Downloading starlette-0.26.1-py3-none-any.whl (66 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 66.9/66.9 kB 28.3 MB/s eta 0:00:00
Collecting uvicorn==0.22.* (from platformio)
  Downloading uvicorn-0.22.0-py3-none-any.whl (58 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.3/58.3 kB 24.1 MB/s eta 0:00:00
Collecting wsproto==1.2.* (from platformio)
  Downloading wsproto-1.2.0-py3-none-any.whl (24 kB)
Collecting packaging>=17.0 (from marshmallow==3.*->platformio)
  Downloading packaging-23.1-py3-none-any.whl (48 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 48.9/48.9 kB 18.4 MB/s eta 0:00:00
Collecting charset-normalizer<4,>=2 (from requests==2.*->platformio)
  Downloading charset_normalizer-3.1.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (199 kB)
[... 284 lines cut ...]
Library Manager: Installing git+https://github.com/PlummersSoftwareLLC/ESPAsyncWebServer.git
git version 2.41.0
Cloning into '/home/runner/.platformio/.cache/tmp/pkg-installing-kvea5sig'...
Library Manager: ESP Async WebServer@1.2.3+sha.a6cd518 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ ^6.8.14
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: ArduinoJson@6.21.2 has been installed!
Library Manager: Installing thomasfredericks/Bounce2 @ ^2.7.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: Bounce2@2.71.0 has been installed!
Library Manager: Installing git+https://github.com/PlummersSoftwareLLC/RemoteDebug
git version 2.41.0
[... cut ...]
Library Manager: Installing Bodmer/TJpg_Decoder @ ^1.0.8
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: TJpg_Decoder@1.0.8 has been installed!
Library Manager: Installing plageoj/UrlEncode @ ^1.0.1
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90%
Library Manager: UrlEncode@1.0.1 has been installed!
Resolving heltecv3demo dependencies...
Error: Unknown board ID 'heltec_wifi_kit_[32](https://github.com/rbergen/NightDriverStrip/actions/runs/5375276242/jobs/9751558524#step:5:33)_V3'
Error: Process completed with exit code 1.
```

valeros, 2023-06-26:
> Platform Manager: espressif32@5.4.0 has been installed!

The `heltec_wifi_kit_32_V3` board was added in [v6.1.0](https://github.com/platformio/platform-espressif32/releases/tag/v6.1.0) of this platform and you're installing v5.4.0.

rbergen, 2023-06-26:
That's obviously something to change... At the same time, then the question becomes how this could ever have worked. The reason I'm saying that is that it literally did until yesterday.

Jason2866, 2023-06-26:
You could speed up your GH Actions process by running the build process jobs in parallel (with matrix).
Maybe you find some "inspiration" [here](https://github.com/arendst/Tasmota/blob/development/.github/workflows/build_all_the_things.yml)

rbergen, 2023-06-26:
@Jason2866 That's a good suggestion, I appreciate you making it. The thing is that I'd want to put the matrix together dynamically (i.e., pull the environments from platformio.ini). Which is possible of course by using a separate job to do so, but requires some parsing logic I haven't gotten round to yet. It's on my list. :)

rbergen, 2023-06-26:
The CI build of the heltec V3 build just completed successfully after I pinned the espressif32 version to @^6.3.0, so I'm closing this.

Many thanks to @valeros and @Jason2866 for your quick responses!

closed 2023-06-26 by rbergen

valeros, 2023-06-26:
@rbergen FYI, that was actually a bug on our side, the v5.4.0 was mistakenly marked as the latest and hence installed by default. Anyway, it's fixed now, so many thanks for reporting.