platformio/platform-espressif32 #1138: Please add support for wt32-sc01 plus

aseanwatson, 2023-06-28:
What kind of issue is this?

- [X] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.7

### Description of problem
Please add support for wt32-sc01 plus. There are a few other projects (see below) which use platformio to build solutions for this board, but it's hard to tell what's really necessary.

### Additional info

#### See https://github.com/fritsjan/WT32-SC01-PLUS-PLATFORMIO which has:

[README.md](https://github.com/fritsjan/WT32-SC01-PLUS-PLATFORMIO/blob/master/README.md):

> After struggling to get it working with this parallel display, I found out that some of the display pins on the WT32-SC01-PLUS had a number higher than 31.
> Unfortunately the TFT_eSPI library uses a trick which makes it impossible to use pins numbered above 31...
> 
> Fortunately LovyanGFX library does not have this restriction.

...

> First of all the WT32-SC01-PLUS is using a custom ESP32-S3 module made by WIRELESS-TAG.
> It has 8Mb FLASH and 2Mb PSRAM. In platform.io the closest esp32-s3 for this is the um_tinys3 board.
> So I used this here, see platform.ini for details
>
> Note that after uploading the firmware, the device is NOT automatically restarted, you have to do this manually (BUG in ESP32 library)

and [platformio.ini](https://github.com/fritsjan/WT32-SC01-PLUS-PLATFORMIO/blob/master/platformio.ini)

``` ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:WT32-SC01-PLUS]
platform = espressif32
board = um_tinys3
framework = arduino
upload_speed = 921600
monitor_speed = 115200
;board_build.partitions = no_ota.csv
build_flags = 
	-DARDUINO_USB_CDC_ON_BOOT
	-DLV_CONF_INCLUDE_SIMPLE
	-DLV_COMP_CONF_INCLUDE_SIMPLE
	-DLV_LVGL_H_INCLUDE_SIMPLE
	-DBOARD_HAS_PSRAM
	;-DDEBUG_TOUCH
	-mfix-esp32-psram-cache-issue
	-I src/ui/
	-I src/
lib_deps = 
	SPI
	;Wire
	lovyan03/LovyanGFX@^0.4.18
	lvgl/lvgl@^8.3.2
```

#### Also see https://github.com/HASwitchPlate/openHASP:
Which has this in [platformio.ini](https://github.com/HASwitchPlate/openHASP/blob/master/platformio.ini):

``` ini
; MIT License - Copyright (c) 2019-2023 Francis Van Roie
; For full license information read the LICENSE file in the project folder
;
; PlatformIO Project Configuration File
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

;**************************************************************
;          Common project settings
;**************************************************************
[platformio]
extra_configs =
    user_setups/lcd_config.ini
    ; --  Base configurations per platform
    user_setups/esp32/_esp32.ini
    user_setups/esp32s2/_esp32s2.ini
    user_setups/esp32s3/_esp32s3.ini
    ; --  Put custom [env] files in this dir to be included in the build menu
    user_setups/custom/*.ini
    ; --  Add customizations to this file only (copy and rename the template):
    platformio_override.ini

; --- snip ---
```

And this in [user_setups/esp32s3/_esp32s3.ini](https://github.com/HASwitchPlate/openHASP/blob/master/user_setups/esp32s3/_esp32s3.ini)
``` ini
; -- Platform specific build flags
[esp32s3]
extends = esp32
board_build.mcu = esp32s3

; --- snip ---

; -- The Arduino ESP32 v2.0.2 with 3 available flash sizes:
[arduino_esp32s3_v2]
extends = esp32s3
framework = arduino
;platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.01.01/platform-espressif32.zip
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.06.04/platform-espressif32.zip
;platform = espressif32@5.3.0
```

And this in [user_setups/esp32s3/wt32-sc01-plus.ini](https://github.com/HASwitchPlate/openHASP/blob/master/user_setups/esp32s3/wt32-sc01-plus.ini)
``` ini
[wt32-sc01-plus]
extends = arduino_esp32s3_v2
board = esp32s3_qio_qspi
board_build.arduino.memory_type = qio_qspi 

; --- snip ---

[env:wt32-sc01-plus_8MB]
extends = wt32-sc01-plus, flash_8mb

[env:wt32-sc01-plus_16MB]
extends = wt32-sc01-plus, flash_16mb
```

And this in [boards/esp32s3_qio_qspi.json](https://github.com/HASwitchPlate/openHASP/blob/master/boards/esp32s3_qio_qspi.json):
``` ini
{
    "build": {
        "arduino": {
            "ldscript": "esp32s3_out.ld",
            "memory_type": "qio_qspi"
        },
        "boot": "qio",
        "psram_type": "qspi",
        "core": "esp32",
        "extra_flags": [
            "-DARDUINO_RUNNING_CORE=1",
            "-DARDUINO_EVENT_RUNNING_CORE=1"
        ],
        "f_cpu": "240000000L",
        "f_flash": "80000000L",
        "flash_mode": "qio",
        "mcu": "esp32s3",
        "variant": "esp32s3",
        "partitions": "esp32_partition_app1856k_spiffs320k.csv"
    },
    "connectivity": [
        "wifi"
    ],
    "debug": {
        "openocd_target": "esp32s3.cfg"
    },
    "frameworks": [
        "espidf",
        "arduino"
    ],
    "name": "Espressif Generic ESP32-S3 4M Flash, Tasmota 1856k Code/OTA, 320k FS",
    "upload": {
        "flash_size": "4MB",
        "maximum_ram_size": 327680,
        "maximum_size": 4194304,
        "require_upload_port": true,
        "speed": 921600
    },
    "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/",
    "vendor": "Espressif"
}
```

aseanwatson, 2023-06-29:
### More information

Wireless-Tag (the manufacturer) has as page for the [WT-SC01 PLUS](http://en.wireless-tag.com/product-item-26.html) which has a link to the datasheet for the board. That, references a `WT32-S3-WROVER-N16R2`.

The manufacturer has a page for [WT32-S3-WROVER](http://en.wireless-tag.com/product-item-17.html), Its datasheet lists 12 variants, but not N16R2. The pattern is `WT32-S3-WROVER{,-I}-N{4,8,16}{,-R8}` where `-I` means an IPEX antenna vs without `-I` means PCB. The 4/8/16 is flash size in MB, and the `R8` has 8Mb PSRAM.

Espressif's [esp32-s3_datasheet-en.pdf](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf) has `ESP32-S3R2` having 2 MB (Quad SPI) rather than 8Mb for the R8.

valeros, 2023-06-30:
Hi @aseanwatson, it's highly desirable that this board is added to the official Arduino core for ESP32 https://github.com/espressif/arduino-esp32 with a proper variant file.

aseanwatson, 2023-06-30:
@valeros , thank you for responding.

I see that other users have been successful at building on the board, so I know it's feasible for Platformio to support the hardware. But if I had have skills or expertise to make it happen myself, I would be submitting pull requests rather than raising issues. ;-)

I am happy to do the legwork, but I do not know how to go about it.

Should I raise an issue in the https://github.com/espressif/arduino-esp32 repo?

Wireless tag also makes the `wt32-eth01` which is supported by Platform IO and has [variants/wt32-eth01/pins_arduino.h](https://github.com/espressif/arduino-esp32/blob/master/variants/wt32-eth01/pins_arduino.h).

valeros, 2023-07-03:
> I see that other users have been successful at building on the board, so I know it's feasible for Platformio to support the hardware.

It's indeed possible to use this board with some workarounds, but if you want hassle-free experience then this board has to be added to  https://github.com/espressif/arduino-esp32 with a proper `pins_arduino.h` file. As an example you can take a look at this [PR]( https://github.com/espressif/arduino-esp32/pull/8257/files).

schuppeste, 2023-12-21:
A little bit late.. You can check this Repo. Quick&Dirty changed only for WT32-sc01Plus.

https://github.com/schuppeste/TFT_eSPI-WT32-sc01-Plus/

c4801725870, 2024-02-23:
I was able to build my first project using platformio based on um_tinys3 board for wt32-sc01 plus.
Serial2 pins are 12 for TX and 13 for RX. I sent commands over the uart and operate the display(change arc and label text) using this method. 
I cannot believe pins_arduino.h is not yet added. I am not sure how to add it, otherwise I would. I see the example submission in the PR referenced above. I will look at submitting PR for variants/WT32-SC01-PLUS/pins_arduino.h. 

I don't know where to get the extra boards.txt information, I can just copy the um_tinys3 text out of boards.txt

c4801725870, 2024-02-24:
https://github.com/espressif/arduino-esp32/pull/9284

c4801725870, 2024-02-27:
They merged it 😁😁😁

aseanwatson, 2024-02-29:
@DrNeuroSurg here's a recap:

* I asked PlatformIO to support the board
* Valeros asked me (or someone) to get Espressif to support the board as a prerequisite
* I was stumped on this, but c4801725870 was able to figure it out
* c4801725870's PR https://github.com/espressif/arduino-esp32/pull/9284 was just merged by the Espressif folks

Presumably, after PlatformIO picks up whatever build of the espressif repo contains the change, it'll be possible to add support for the board to PlatformIO.

c4801725870, 2024-02-29:
> @c4801725870 where ? Can't see it ...

yeah right now it says "pending merge" tagged, they have an external build that's public, and internal code base. So right now they are merging the public code with their private code. At least that's how I understand it. There are other PRs to include besides this one.

Right now you could download the public code to the location C:\Users\{USER}\.platformio\packages\framework-arduinoespressif32\boards.txt and also include the new variants folder. The actual latest release is still https://github.com/espressif/arduino-esp32/releases/tag/2.0.14

I don't know how VS Code would react to having the files overwritten, I only tested this in Arduino IDE. For now I am using um_tinys3 for development and defining the pins manually.

mrstas, 2024-07-18:
@valeros  Sorry for resurrecting old issue. Can this board be added now? The board definition has been merged to espressif/arduino-esp32:master [https://github.com/espressif/arduino-esp32/pull/9284] a while ago and board is available in latest releases