platformio/platform-espressif32 #1141: Cannot use ESP-IDF framework on PlatformIO ("Could not find the package with...")

empewoow, 2023-07-04:
Unfortunately I cannot get ESP-IDF to work on a fresh PlatformIO project initialization. When I try `framework = espidf` in my ini and try `pio run` I get:
```
Resolving esp32dev dependencies...
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2
Error: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2' requirements for your system 'windows_amd64'
```

This is my platformio.ini:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
monitor_speed = 115200
```

When I try `framework = arduino` it works.

I have an ESP32 Dev Kit C4. Using Visual Studio Code on Windows 10 x64, with the PlatformIO extension. The command `pio --version` returns:
```
PlatformIO Core, version 6.1.7
```

Found this thread, but there's no solution:

https://community.platformio.org/t/cant-compile-project-for-s2/33397/3

Any help is appreciated.

Jason2866, 2023-07-04:
Change to:

```
[env:esp32dev]
platform = espressif32 @ ^6.3.2
board = esp32dev
framework = espidf
monitor_speed = 115200
```

empewoow, 2023-07-04:
This worked, thanks! Although for some reason I had to put `extern "C"` in front of my `void app_main()`.

By the way, how could I have known that adding `@ ^6.3.2` would work? What version does it use when you don't specifically mention it?

closed 2023-07-04 by empewoow

Jason2866, 2023-07-04:
Without pinning a version platformio uses the version which is installed.
This can be an old one....

empewoow, 2023-07-04:
Thanks again!

kaanbuka, 2025-04-02:
Hello, I am getting a similar error as a mac user:
UnknownPackageError: Could not find the package with 'espressif/arduino-esp32' requirements for your system 'darwin_arm64'
I changed the code as mentioned above but the same problem persists

[image]

Jason2866, 2025-04-02:
@kaanbuka  Your setup is wrong. To use the pioarduino fork just change the `platform` entry.
Remove `platform_packages` completely!

kaanbuka, 2025-04-02:
I deleted platform_packages but I'm getting still same error.

Jason2866, 2025-04-02:
Please post the platformio.ini and the source code. Best would be to upload the project in a repo.

kaanbuka, 2025-04-03:
I uploaded the project. 
https://github.com/kaanbuka/esp32

kaanbuka, 2025-04-04:
I changed the parts you changed in my project (actually everything you did looks the same). But I still get the same problem