platformio/platform-espressif32 #1144: Request: ESP32-S3 ULP support in PlatformIO.

peteDDD, 2023-07-07:
ULP example code does not build for ESP32-S3 under PlatformIO

see: https://community.platformio.org/t/asking-again-getting-ulp-tool-example-espidf-ulp-pulse-to-work/34639

Jason2866, 2023-07-07:
Which example have you tried? Both (FSM and risc-v) ULP code does compile fine in my tests on the S3.
https://github.com/Jason2866/platform-espressif32/actions/runs/5357819229/jobs/9719238771
https://github.com/Jason2866/platform-espressif32/actions/runs/5357819229/jobs/9719241951

Jason2866, 2023-07-07:
Oh just noticed, there are the old examples in the repo. In my fork I have updated to the recent ones from espressif.
So yes there is action necessary

peteDDD, 2023-07-07:
Thank you Jason.   This is on PlatformIO that you are compiling?  If so,
could you share the platformIio.ini file you are using?  Thank you again!

Jason2866, 2023-07-07:
@peteDDD It is here https://github.com/Jason2866/platform-espressif32/tree/Tasmota/2010/examples/espidf-ulp

peteDDD, 2023-07-07:
Thank you Jason.  I will try it and let you know.
Pete

peteDDD, 2023-07-08:
Hi Jason,
Knowing that you have gotten this to compile (and I assume run) has given
me resolve to keep working until I have the same success.

I have spent most of today trying to sort this out.  Although I have built
hundreds of projects with many platforms with VSC/Platformio over several
years, I have never used the espidf framework before and getting that
running has been a challenge for some reason.  A clean install fixed that
  I *completely* removed all traces of visual studio code and platformio
from my computer and reinstalled.  This went a long way to getting rid of
all of the odd error messages I was getting trying to build *anything*
with the espidf.  By "anything", I mean even just   void app_main(void) {}

Now that I am over that hump,  I am stuck with the following errors:

Generating ULP configuration
*** [.pio\build\esp32-s3\.pio\build\esp32-s3\ulp_main.bin.o] Source
`C:\Users\peted\Documents\PlatformIO\Projects\ULP2\.pio\build\esp32-s3\ulp_main.bin.S'
not found, needed by target
`.pio\build\esp32-s3\.pio\build\esp32-s3\ulp_main.bin.o'.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version
"2.41.0.windows.1")
fatal: not a git repository (or any of the parent directories): .git

I looked into the ulp_main.S issue and found a reference to needing to
enable the ULP in the component config...the posting I found read as
follows:

"
I get the same error when I just try to build the ulp_adc example. Looking
through the configuration (using `menuconfig`) I found that the ULP was not
enabled by default.

I was able to build the example after enabling ULP via:
- Component config
-- ESP32S2-specific
--- Enable Ultra Low Power (ULP) Coprocessor
"
(from https://www.esp32.com/viewtopic.php?t=21322)

but, I can't make heads or tails of where to find menuconfig or component
config in VSC/Platformio and I don't find anything in the
.pio/build/wes32-s3 folder structure.

Can you give me a clue on this?

Thanks again!
Pete

Jason2866, 2023-07-09:
Do you have copied the whole example folder with everything in there?
You have nothing to configure in menuconfig. All needed settings are in `sdkconfig.defaults.esp32s3`. Only set the S3 as env you want to compile. Hit compile button. That's all.

peteDDD, 2023-07-09:
Hi Jason.
I hear you.  And I have done that.  In fact, I just did a full clean
download of the whole tasmota tree from the link you provided.  Then I
opened the espidf-ulp folder in Platformio.
commented out the non S3 env's.  But I get even more unfound include
files... including ulp_main.h, stdio, esp_sleep, and driver/gpio.h    (so
things are even worse than what I was doing before doing it this way).

On the other hand, with my other work, I am only missing ulp_main.h and
esp32s3/ulp.h which I think means something failed in the espressif32
install but I find nothing wrong in the log file.  (attached)

I searched my whole disk and can't find either file.
***@***.*** is already installed

Any ideas?
Thanks again,
Pete

Tool Manager: Installing espressif/toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: ***@***.***+2021r2-patch5 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (2023.06.04) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.10 
 - framework-espidf @ 3.40405.0 (4.4.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 1.40602.0 (4.6.2) 
 - tool-idf @ 1.0.1 
 - tool-mconf @ 1.4060000.20190628 (406.0.0) 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Generating ULP configuration
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/runneradmin/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc.exe
Generating LD script .pio\build\esp32-s3\memory.ld
-- The C compiler identification is GNU 8.4.0
Compiling .pio\build\esp32-s3\app_trace\app_trace.o
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
Compiling .pio\build\esp32-s3\app_trace\app_trace_util.o
-- Check for working C compiler: C:/Users/runneradmin/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building ULP app ulp_main for esp32s3
-- Configuring done
-- Generating done
-- Build files have been written to: D:/a/platform-espressif32/platform-espressif32/examples/espidf-ulp-riscv/.pio/build/esp32-s3/esp-idf/main/ulp_main
Generating ULP project files .pio\build\esp32-s3\esp-idf\main\ulp_main\ulp_main.h .pio\build\esp32-s3\esp-idf\main\ulp_main\ulp_main.ld .pio\build\esp32-s3\esp-idf\main\ulp_main\ulp_main.bin .pio\build\esp32-s3\esp-idf\main\ulp_main\esp32.ulp.ld
[1/13] Generating esp32s3.ulp.riscv.ld
[2/13] Generating C:/Users/runneradmin/.platformio/packages/framework-espidf/components/ulp/ulp_riscv/ulp_riscv_adc.c

[3/13] Generating D:/a/platform-espressif32/platform-espressif32/examples/espidf-ulp-riscv/ulp/main.c
[... 889 lines cut ...]
Compiling .pio\build\esp32-s3\tcp_transport\transport_ws.o
Compiling .pio\build\esp32-s3\tcpip_adapter\tcpip_adapter_compat.o
Archiving .pio\build\esp32-s3\esp-idf\tcp_transport\libtcp_transport.a
Indexing .pio\build\esp32-s3\esp-idf\tcp_transport\libtcp_transport.a
Compiling .pio\build\esp32-s3\ulp\ulp.o
Archiving .pio\build\esp32-s3\esp-idf\tcpip_adapter\libtcpip_adapter.a
Indexing .pio\build\esp32-s3\esp-idf\tcpip_adapter\libtcpip_adapter.a
Compiling .pio\build\esp32-s3\ulp\ulp_macro.o
Compiling .pio\build\esp32-s3\ulp\ulp_riscv.o
Compiling .pio\build\esp32-s3\ulp\ulp_riscv_adc.o
Compiling .pio\build\esp32-s3\unity\unity\src\unity.o
Archiving .pio\build\esp32-s3\esp-idf\ulp\libulp.a
Indexing .pio\build\esp32-s3\esp-idf\ulp\libulp.a
Compiling .pio\build\esp32-s3\unity\unity_runner.o
Compiling .pio\build\esp32-s3\unity\unity_port_esp32.o
Compiling .pio\build\esp32-s3\usb\hcd_dwc.o
Archiving .pio\build\esp32-s3\esp-idf\unity\libunity.a
Indexing .pio\build\esp32-s3\esp-idf\unity\libunity.a
Compiling .pio\build\esp32-s3\usb\hub.o
Compiling .pio\build\esp32-s3\usb\usb_helpers.o
Compiling .pio\build\esp32-s3\usb\usb_host.o
Compiling .pio\build\esp32-s3\usb\usb_private.o
Compiling .pio\build\esp32-s3\usb\usbh.o
Compiling .pio\build\esp32-s3\usb\usb_phy.o
Compiling .pio\build\esp32-

Jason2866, 2023-07-09:
Mhh, do you have a config which uses different drives?
One part is done on `[3/13] Generating D:/a/platform-espressif32/platform-espressif32/examples/espidf-ulp-riscv/ulp/main.c`
All other is on drive `c:/Users/runneradmin/`
This one makes me believe something weird with your path(s) is going on
`CMakeFiles/ulp_main.dir/C_/Users/runneradmin/`

Do you have whitespaces or special chars in a path?
Do you use a network or cloud drive?

Whitespaces and special chars WILL make issues.
Cloud drives can make troubles.

peteDDD, 2023-07-09:
Good observation.   Very odd.  I had a removable drive plugged in while I
did the install (where I had backed up my config).
I will clean it out again and try again without any external drives.
Very odd.  I'll let you know if that fixes it but it sure makes sense that
it would.
Thanks!
Pete

peteDDD, 2023-07-09:
Another clean install... but still can't find esp32s4/ulp.h  and ulp_main.h

Where are those files located on your installation?

Jason2866, 2023-07-09:
The are in the IDF framework which is located in `.platformio/packages/framework-espidf/components/ulp/ulp_riscv/`
The used framework source:
https://github.com/tasmota/esp-idf/tree/release/v4.4/components/ulp/include/esp32s3

peteDDD, 2023-07-09:
I can find ulp.c in ... components\ulp\ulp_fsm
 But can't find ulp_main.h anywhere...

very frustrating

Jason2866, 2023-07-10:
Just tried. Both of this config do work:
```
[platformio]
src_dir = main

[env:esp32dev]
platform = espressif32 @ ^5.4.0
framework = espidf
board = esp32dev

[env:esp32-s2]
platform = espressif32 @ ^5.4.0
framework = espidf
board = esp32-s2-saola-1

[env:esp32-s3]
platform = espressif32 @ ^5.4.0
framework = espidf
board = esp32-s3-devkitc-1
```
and
```
[platformio]
src_dir = main

[env:esp32dev]
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.06.04/platform-espressif32.zip
framework = espidf
board = esp32dev

[env:esp32-s2]
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.06.04/platform-espressif32.zip
framework = espidf
board = esp32-s2-saola-1

[env:esp32-s3]
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.06.04/platform-espressif32.zip
framework = espidf
board = esp32-s3-devkitc-1
```

valeros, 2023-07-10:
Hi @peteDDD, if I'm not mistaken, the `ulp_main.h` file is autogenerated and should be located somewhere in your build directory, e.g. `.pio\build\YOUR_ENV\esp-idf\main\ulp_main`. Anyway, after all previous steps you tried, what error message do you see?

peteDDD, 2023-07-10:
Valeros.  My understand is the same... ulp_main.h should be autogenerated and located in the build directory.  But it is nowhere to be found on my computer.

The error is simply "cannot open source file  for that the #include statements for "esp32s2/ulp.h" and for "ulp_main.h" 
(pretty standard sort of error that I know how to fix... when the files actually exist and can be found)

peteDDD, 2023-07-10:
Hi Jason,
Thanks again for your perseverance in helping me.

I tried the approach of this platformio.ini
`[platformio]
src_dir = main

[env:esp32-s3]
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.06.04/platform-espressif32.zip
framework = espidf
board = esp32-s3-devkitc-1`

and got the same results... 

`Processing esp32-s3 (platform: https://github.com/tasmota/platform-espressif32/releases/download/2023.06.04/platform-espressif32.zip; framework: espidf; 
board: esp32-s3-devkitc-1)
---------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (2023.6.4) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.10
 - framework-espidf @ 3.40405.0 (4.4.5)
 - tool-cmake @ 3.21.3
 - tool-esptoolpy @ 1.40602.0 (4.6.2)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Generating ULP configuration
*** [.pio\build\esp32-s3\.pio\build\esp32-s3\ulp_main.bin.o] Source `C:\Users\peted\Documents\PlatformIO\Projects\ULP2\.pio\build\esp32-s3\ulp_main.bin.S' not found, needed by target `.pio\build\esp32-s3\.pio\build\esp32-s3\ulp_main.bin.o'.
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/peted/.platformio/packages/toolchain-esp32ulp/bin/esp32ulp-elf-as.exe
-- The C compiler identification is GNU 8.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/peted/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building ULP app ulp_main for esp32s3
-- ULP assembler version: 2.35_20220830
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/owner/Documents/PlatformIO/Projects/ULP2/.pio/build/esp32-s3/esp-idf/main/ulp_main
============================================================== [FAILED] Took 7.19 seconds ==============================================================` 

I also tried:

`[platformio]
src_dir = main

[env:esp32-s3]
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.06.04/platform-espressif32.zip
framework = espidf
board = esp32-s3-devkitc-1
lib_ldf_mode = deep
lib_extra_dirs = 
c:\Users\peted\Documents\PlatformIO\Projects\ULP2\.pio\build\esp32-s3\esp-idf
;\main\ulp_main`

and got the same results.

valeros, 2023-07-10:
@peteDDD Please share here the contents of your `CMakeLists.txt` from the `main` folder.

peteDDD, 2023-07-10:
[CODE]
idf_component_register(SRCS "ulp_example_main.c"
                    INCLUDE_DIRS ""
                    REQUIRES soc nvs_flash ulp)
#
# ULP support additions to component CMakeLists.txt.
#
# 1. The ULP app name must be unique (if multiple components use ULP).
set(ulp_app_name ulp_${COMPONENT_NAME})
#
# 2. Specify all assembly source files.
#    Files should be placed into a separate directory (in this case, ulp/),
#    which should not be added to COMPONENT_SRCS.
set(ulp_s_sources "ulp/pulse_cnt.S" "ulp/wake_up.S")
#
# 3. List all the component source files which include automatically
#    generated ULP export file, ${ulp_app_name}.h:
set(ulp_exp_dep_srcs "../ulp_example_main.c")
#
# 4. Call function to build ULP binary and embed in project using the
argument
#    values above.
ulp_embed_binary(${ulp_app_name} "${ulp_s_sources}" "${ulp_exp_dep_srcs}")

[/CODE]

valeros, 2023-07-10:
> set(ulp_s_sources "ulp/pulse_cnt.S" "ulp/wake_up.S")

Where did you get that example? This line should contain proper relative paths as shown in our [example](https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-ulp-pulse/src/CMakeLists.txt#L13) in this repository:
```
set(ulp_s_sources "../ulp/pulse_cnt.S" "../ulp/wake_up.S")
```

peteDDD, 2023-07-10:
Valerii,
scroll up this message chain and you will see the url for the example I am
using from Jason2866.

Jason2866, 2023-07-10:
@valeros Both variants do work.

Jason2866, 2023-07-10:
The updated code from espressif
```
/* ULP Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <stdio.h>
#include "esp_sleep.h"
#include "nvs.h"
#include "nvs_flash.h"
#include "soc/rtc_cntl_reg.h"
#include "soc/sens_reg.h"
#include "soc/rtc_periph.h"
#include "driver/gpio.h"
#include "driver/rtc_io.h"

#if CONFIG_IDF_TARGET_ESP32
#include "esp32/ulp.h"
#elif CONFIG_IDF_TARGET_ESP32S2
#include "esp32s2/ulp.h"
#elif CONFIG_IDF_TARGET_ESP32S3
#include "esp32s3/ulp.h"
#endif

#include "ulp_main.h"

extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t ulp_main_bin_end[]   asm("_binary_ulp_main_bin_end");

static void init_ulp_program(void);
static void update_pulse_count(void);

void app_main(void)
{
    esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();
    printf("cause %d\n", cause);
    if (cause != ESP_SLEEP_WAKEUP_ULP) {
        printf("Not ULP wakeup, initializing ULP\n");
        init_ulp_program();
    } else {
        printf("ULP wakeup, saving pulse count\n");
        update_pulse_count();
    }

    printf("Entering deep sleep\n\n");
    ESP_ERROR_CHECK( esp_sleep_enable_ulp_wakeup() );
    esp_deep_sleep_start();
}

static void init_ulp_program(void)
{
    esp_err_t err = ulp_load_binary(0, ulp_main_bin_start,
            (ulp_main_bin_end - ulp_main_bin_start) / sizeof(uint32_t));
    ESP_ERROR_CHECK(err);

    /* GPIO used for pulse counting. */
    gpio_num_t gpio_num = GPIO_NUM_0;
    int rtcio_num = rtc_io_number_get(gpio_num);
    assert(rtc_gpio_is_valid_gpio(gpio_num) && "GPIO used for pulse counting must be an RTC IO");

    /* Initialize some variables used by ULP program.
     * Each 'ulp_xyz' variable corresponds to 'xyz' variable in the ULP program.
     * These variables are declared in an auto generated header file,
     * 'ulp_main.h', name of this file is defined in component.mk as ULP_APP_NAME.
     * These variables are located in RTC_SLOW_MEM and can be accessed both by the
     * ULP and the main CPUs.
     *
     * Note that the ULP reads only the lower 16 bits of these variables.
     */
    ulp_debounce_counter = 3;
    ulp_debounce_max_count = 3;
    ulp_next_edge = 0;
    ulp_io_number = rtcio_num; /* map from GPIO# to RTC_IO# */
    ulp_edge_count_to_wake_up = 10;

    /* Initialize selected GPIO as RTC IO, enable input, disable pullup and pulldown */
    rtc_gpio_init(gpio_num);
    rtc_gpio_set_direction(gpio_num, RTC_GPIO_MODE_INPUT_ONLY);
    rtc_gpio_pulldown_dis(gpio_num);
    rtc_gpio_pullup_dis(gpio_num);
    rtc_gpio_hold_en(gpio_num);

#if CONFIG_IDF_TARGET_ESP32
    /* Disconnect GPIO12 and GPIO15 to remove current drain through
     * pullup/pulldown resistors on modules which have these (e.g. ESP32-WROVER)
     * GPIO12 may be pulled high to select flash voltage.
     */
    rtc_gpio_isolate(GPIO_NUM_12);
    rtc_gpio_isolate(GPIO_NUM_15);
#endif // CONFIG_IDF_TARGET_ESP32

    esp_deep_sleep_disable_rom_logging(); // suppress boot messages

    /* Set ULP wake up period to T = 20ms.
     * Minimum pulse width has to be T * (ulp_debounce_counter + 1) = 80ms.
     */
    ulp_set_wakeup_period(0, 20000);

    /* Start the program */
    err = ulp_run(&ulp_entry - RTC_SLOW_MEM);
    ESP_ERROR_CHECK(err);
}

static void update_pulse_count(void)
{
    const char* namespace = "plusecnt";
    const char* count_key = "count";

    ESP_ERROR_CHECK( nvs_flash_init() );
    nvs_handle_t handle;
    ESP_ERROR_CHECK( nvs_open(namespace, NVS_READWRITE, &handle));
    uint32_t pulse_count = 0;
    esp_err_t err = nvs_get_u32(handle, count_key, &pulse_count);
    assert(err == ESP_OK || err == ESP_ERR_NVS_NOT_FOUND);
    printf("Read pulse count from NVS: %5d\n", pulse_count);

    /* ULP program counts signal edges, convert that to the number of pulses */
    uint32_t pulse_count_from_ulp = (ulp_edge_count & UINT16_MAX) / 2;
    /* In case of an odd number of edges, keep one until next time */
    ulp_edge_count = ulp_edge_count % 2;
    printf("Pulse count from ULP: %5d\n", pulse_count_from_ulp);

    /* Save the new pulse count to NVS */
    pulse_count += pulse_count_from_ulp;
    ESP_ERROR_CHECK(nvs_set_u32(handle, count_key, pulse_count));
    ESP_ERROR_CHECK(nvs_commit(handle));
    nvs_close(handle);
    printf("Wrote updated pulse count to NVS: %5d\n", pulse_count);
}
```

peteDDD, 2023-07-10:
> Since the default (platformio) folder structure is used this part is not
> needed at all
> Just try and comment the line. Compile will be successful

Jason,
Still no joy.
I have a hunch that I need to dig into.  This laptop is new to me and has
two users "owner" and "peted".  I think there might be some path issues.

valeros, 2023-07-11:
@peteDDD You've already tried a lot, but I'd like ask you to follow these steps:

1. Remove all previous platform packages `espressif*` in this folder `C:/Users/YOUR_USER/.platformio/platforms`
2. Download this example [espidf-ulp-pulse-test.zip](https://github.com/platformio/platform-espressif32/files/12014166/espidf-ulp-pulse-test.zip) and compile it
3. Share here the error message if any

peteDDD, 2023-07-11:
Holly cow!  Success!!!   But why did it suddenly work???

[loglog.txt](https://github.com/platformio/platform-espressif32/files/12015673/loglog.txt)

peteDDD, 2023-07-11:
BUT... on next run to upload... failed the same way.

So, I ran the clean and run again (without verbose logging) and was back to the same failure.

SO... repeated clean and verbose build... and it builds.   

Now I guess I need to do a verbose upload.

peteDDD, 2023-07-11:
Verbose upload compiled and uploaded.
When funning, the monitor shows the processor going to ULP deep sleep but nothing happens after that.
No number of IO0 button presses trigger a wakeup and report.
Tried on two different ESP32S3 boards.
Checked in menuconfig that the ULP was enabled and set to FSM with 1024 bytes of RTC slow memory reserved.

great progress.... but still not quite there.  Thank you!
I eager for more ideas.

One note.  The building seems to be pretty non-repeatable.  I have to go back and do a clean and a verbose build each time.  Any other approach, such as just doing a verbose upload, does fails.  Perhaps Platformio is going out and downloading more "meta" files that we have to clear out.

Jason2866, 2023-07-11:
So this one works for me with Plain IDF 5.1 and `espressif32 @ ^6.3.2` and `platform = espressif32 @ ^5.4.0`

[espidf-ulp-pulse-test.zip](https://github.com/platformio/platform-espressif32/files/12017287/espidf-ulp-pulse-test.zip)

peteDDD, 2023-07-12:
I have to focus on some other things for a few days but will return to this
shortly.
Please don't mistake my lack of work on this now as any less appreciation
of your help.
Thanks for your help and understanding.
Pete

valeros, 2023-07-13:
@peteDDD If possible, I'd like to look into a dependency tree generated on your machine. Please open the terminal in the IDE and follow these steps:

```bash
# Clean the project before build
pio run -t clean

# Set some special SCons flags
set SCONSFLAGS=--tree=prune

# Run the project and redirect the output to a file
pio run > esp32s3ulp-deptree-prune.txt

# Clean variable
set SCONSFLAGS=
```
Please share here the `esp32s3ulp-deptree-prune.txt` file.

peteDDD, 2023-08-26:
Well life got a little out of hand for a while there, but I am finally back to this.

Summary of where I am now... 
1: Today I downloaded [espidf-ulp-pulse-test.zip](https://github.com/platformio/platform-espressif32/files/12017287/espidf-ulp-pulse-test.zip) which Jason2866 posted here on 11 July.  The only change I made was to add monitor_speed = 115200 to the platformio.ini
2: This compiles and loads.
3: I have tried to run this on two different ESP32S3 boards.  One is the genuine Espressif ESP32-S3-DevKitC-1U and one is an aliexpress knock-off with a wider footprint and more flash.
4: On the Expressif card, I get the following on the monitor... and then nothing else, no matter how many times I press the pin0 boot botton.
`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x11e0
load:0x403c9700,len:0xa54 
load:0x403cc700,len:0x2cc4
entry 0x403c9894
Not ULP wakeup, initializing ULP
Entering deep sleep
`
On the knockoff ESP32-S3, I get the following:
`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0))
waiting for download
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x11e0
load:0x403c9700,len:0xa54
load:0x403cc700,len:0x2cc4
entry 0x403c9894
␛[0;33mW (90) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.␛[0m
Not ULP wakeup, initializing ULP
Entering deep sleep
`

5: @valeros you asked for the esp32s3ulp-deptree-prune.txt, It is attached.

While running the pio run to generate the deptree, it completed, but I got the following messages (my entries included here also for completeness):
`PS C:\Users\peted\Downloads\espidf-ulp-pulse-test\espidf-ulp-pulse-test> pio run -t clean
Processing esp32-s3 (platform: espressif32 @ ^6.3.2; framework: espidf; board: esp32-s3-devkitc-1)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled vVerbose mode can be enabled via `-v, --verbose` option
Removing .pio\build\esp32-s3
Done cleaning
===================================================================================================================================================================================================== [SUCCESS] Took 0.99 seconds ===================================================================================================================================================================================================== 
PS C:\Users\peted\Downloads\espidf-ulp-pulse-test\espidf-ulp-pulse-test> set SCONSFLAGS=--tree=prune
PS C:\Users\peted\Downloads\espidf-ulp-pulse-test\espidf-ulp-pulse-test> pio run > esp32s3ulp-deptree-prune.txt
fatal: not a git repository (or any of the parent directories): .git
`
6: I also reconfigured my platformio.ini and uploaded the project to an ESP32 card (TTGO T-Display) and got similar behavior in that I got nothing beyond the "Entering deep sleep" message.

Thanks again for all the help and patience.  Any ideas on why none of these processors is exiting the deep sleep?

[esp32s3ulp-deptree-prune.txt](https://github.com/platformio/platform-espressif32/files/12446698/esp32s3ulp-deptree-prune.txt)

@peteDDD @valeros