platformio/platform-espressif32 #1146: Build option build_src_flags in platformio.ini doesn't work

Ar7eniyan, 2023-07-13:
Setting `build_src_flags` to anything doesn't pass the flags to the compiler on esp32 boards. Creating the issue here because the same configuration does compile on `platform = atmelavr` and `board = uno`.

**platformio.ini**
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf

build_src_flags = -D THIS_MACRO_WONT_WORK=42
```

**main.c**
```c
#include "stdio.h"

void app_main() {
    printf("%d", THIS_MACRO_WONT_WORK);
}
```

**Verbose build log**
```
 *  Executing task: C:\Users\Ar7eniyan\.platformio\penv\Scripts\platformio.exe run --verbose --environment esp32dev 

Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf; build_src_flags: -D THIS_MACRO_WONT_WORK=42)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50001.0 (5.0.1) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40500.0 (4.5.0) 
 - tool-idf @ 1.0.1 
 - tool-mconf @ 1.4060000.20190628 (406.0.0) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 11.2.0+2022r1
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Using tempfile F:\Projects\Platformio\issue-showcase-esp-idf\.pio\build\esp32dev\tmp5j5605zx.tmp for command line:
xtensa-esp32-elf-gcc -o .pio/build/esp32dev/src/main.o -c -Og -Wall -Werror=all -Wextra -Wno-enum-conversion -Wno-error=deprecated-declarations -Wno-error=unused-but-set-variable -Wno-error=unused-function -Wno-error=unused-variable -Wno-frame-address -Wno-old-style-declaration -Wno-sign-compare -Wno-unused-parameter -fdata-sections -ffunction-sections -fmacro-prefix-map=C:/Users/Ar7eniyan/.platformio/packages/framework-espidf=/IDF -fmacro-prefix-map=F:/Projects/Platformio/issue-showcase-esp-idf=. -fno-jump-tables -fno-tree-switch-conversion -fstrict-volatile-bitfields -gdwarf-4 -ggdb -mlongcalls -std=gnu17 -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=F:/Projects/Platformio/issue-showcase-esp-idf=. -fmacro-prefix-map=C:/Users/Ar7eniyan/.platformio/packages/framework-espidf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -Wno-old-style-declaration -D_POSIX_READER_WRITER_LOCKS -DESP_PLATFORM -DIDF_VER=\"5.0.1\" -D_GNU_SOURCE -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -DPLATFORMIO=60110 -DARDUINO_ESP32_DEV -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/xtensa/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/xtensa/esp32/include -I.pio/build/esp32dev/config -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/newlib/platform_include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/freertos/esp_additions/include/freertos -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/freertos/esp_additions/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_hw_support/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_hw_support/include/soc -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_hw_support/include/soc/esp32 -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_hw_support/port/esp32 -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_hw_support/port/esp32/private_include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/heap/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/log/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/soc/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/soc/esp32 -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/soc/esp32/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/hal/esp32/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/hal/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/hal/platform_port/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_rom/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_rom/include/esp32 -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_rom/esp32 -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_common/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_system/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_system/port/soc -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_system/port/include/private -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/lwip/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/lwip/include/apps -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/lwip/include/apps/sntp -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/lwip/lwip/src/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/lwip/port/esp32/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/lwip/port/esp32/include/arch -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_ringbuf/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/efuse/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/efuse/esp32/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_timer/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/driver/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/driver/deprecated -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/driver/esp32/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_pm/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/mbedtls/port/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/mbedtls/mbedtls/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/mbedtls/mbedtls/library -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/mbedtls/esp_crt_bundle/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_app_format/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/bootloader_support/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/bootloader_support/bootloader_flash/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_partition/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/app_update/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/spi_flash/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/pthread/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/app_trace/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_event/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/nvs_flash/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/esp_phy/include -IC:/Users/Ar7eniyan/.platformio/packages/framework-espidf/components/e [... cut ...]
src/main.c: In function 'app_main':
src/main.c:4:18: error: 'THIS_MACRO_WONT_WORK' undeclared (first use in this function)
    4 |     printf("%d", THIS_MACRO_WONT_WORK);
      |                  ^~~~~~~~~~~~~~~~~~~~
src/main.c:4:18: note: each undeclared identifier is reported only once for each function it appears in

... goes on ...

====================================================================== [FAILED] Took 3.70 seconds ======================================================================
```

Jason2866, 2023-07-13:
Maybe `build_src_flags = -DTHIS_MACRO_WONT_WORK=42`?

Ar7eniyan, 2023-07-13:
> Maybe `build_src_flags = -DTHIS_MACRO_WONT_WORK=42`?

This doesn't build either. It's not just macro definitions that don't work, but any compiler options as well.

Jason2866, 2023-07-13:
Macros can be tricky. This one works

```
[env:esp32dev]
platform = espressif32 @ 6.1.0
board = esp32dev
framework = espidf
build_flags = -DTHIS_MACRO_WONT_WORK='42'
```

Ar7eniyan, 2023-07-13:
> `build_flags = -DTHIS_MACRO_WONT_WORK='42'`
Not `build_flags`, but `build_src_flags`. The point is that the first one works and the second one doesn't.

Jason2866, 2023-07-13:
Your orig. example is not correct at all. Without escaping the value it shouldnt work.
But yes there is a bug with `build_src_flags = -DTHIS_MACRO_WONT_WORK='42'`

Ar7eniyan, 2023-07-14:
Found a workaround by using [target_compile_options](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#controlling-component-compilation) in a main component's cmakelists (\<project\>/src/CMakeLists.txt):
`target_compile_options(${COMPONENT_LIB} PRIVATE <options>)`

Can someone mark this issue as a bug?

valeros, 2023-07-21:
Thanks for reporting, should be fixed in the dev branch.

closed 2023-07-21 by valeros

udem1234, 2024-03-10:
I'm not sure if my problem is related but I used build_src_flags in platformio.ini specific board sections to include appropriate header file based on the board used.  Here is a short example :

```
...

[env:lilygo-t-display-s3]
...
build_src_flags =
${env.build_src_flags}
-include lib/boards/Lilygo-T-Display-S3.hpp

[env:makerfabs-s3-35]
...
build_src_flags =
${env.build_src_flags}
-include lib/boards/MakerFabs_Parallel_S3.hpp

...
```

But it seems that the appropriate #include is not considered in src/main.cpp as expected when building.  A specific function is not found with this method (but is found when I add the #include directly in the main.cpp).

Or is it me who don't understand how to implement the behavior I need ?

commit 26aa55aa (2023-07-21): Process "build_src_flags" option in IDF build script

Resolves #1146