platformio/platform-espressif32 #1148: new board: Heltec Wireless Tracker

codeadventure20, 2023-07-17:
I kindly request the inclusion of the Heltec Wireless Tracker board in the list of supported boards for PlatformIO. The Heltec Wireless Tracker board is an ESP32-based development board with built-in GPS, LoRaWAN, and Bluetooth capabilities, making it suitable for GPS tracking and IoT applications.

Currently, the Heltec Wireless Tracker board is not available as an option in PlatformIO. Adding official support for this board would greatly benefit developers working with the Heltec Wireless Tracker, enabling them to utilize the full potential of the board and leverage the extensive PlatformIO ecosystem.

Board Details:

Board Name: Heltec Wireless Tracker
Microcontroller: ESP32
Built-in Features: GPS, LoRaWAN, Bluetooth
Manufacturer: Heltec Automation

it is implemented in the wifi_kit_series release 0.0.8 
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series

valeros, 2023-07-17:
Hi @codeadventure20, it's highly desirable that this board is added to the official Arduino core for ESP32 https://github.com/espressif/arduino-esp32 with a proper variant file.

codeadventure20, 2023-07-27:
I'm not sure how to add it there.
Here is the file: heltec_wireless_tracker/pins_arduino.h

#ifndef Pins_Arduino_h
#define Pins_Arduino_h

#include <stdint.h>
#include "soc/soc_caps.h"

#define USB_VID 0x303a
#define USB_PID 0x1001

#define EXTERNAL_NUM_INTERRUPTS 46
#define NUM_DIGITAL_PINS 48
#define NUM_ANALOG_INPUTS 20

// Some boards have too low voltage on this pin (board design bug)
// Use different pin with 3V and connect with 48
// and change this setup for the chosen pin (for example 38)
static const uint8_t LED_BUILTIN = SOC_GPIO_PIN_COUNT+48;
#define BUILTIN_LED LED_BUILTIN // backward compatibility
#define LED_BUILTIN LED_BUILTIN
#define RGB_BUILTIN LED_BUILTIN
#define RGB_BRIGHTNESS 64

#define analogInputToDigitalPin(p) (((p)<20)?(analogChannelToDigitalPin(p)):-1)
#define digitalPinToInterrupt(p) (((p)<48)?(p):-1)
#define digitalPinHasPWM(p) (p < 46)

static const uint8_t TX = 43;
static const uint8_t RX = 44;

static const uint8_t SDA = 41;
static const uint8_t SCL = 42;

static const uint8_t SS = 8;
static const uint8_t MOSI = 10;
static const uint8_t MISO = 11;
static const uint8_t SCK = 9;

static const uint8_t A0 = 1;
static const uint8_t A1 = 2;
static const uint8_t A2 = 3;
static const uint8_t A3 = 4;
static const uint8_t A4 = 5;
static const uint8_t A5 = 6;
static const uint8_t A6 = 7;
static const uint8_t A7 = 8;
static const uint8_t A8 = 9;
static const uint8_t A9 = 10;
static const uint8_t A10 = 11;
static const uint8_t A11 = 12;
static const uint8_t A12 = 13;
static const uint8_t A13 = 14;
static const uint8_t A14 = 15;
static const uint8_t A15 = 16;
static const uint8_t A16 = 17;
static const uint8_t A17 = 18;
static const uint8_t A18 = 19;
static const uint8_t A19 = 20;

static const uint8_t T1 = 1;
static const uint8_t T2 = 2;
static const uint8_t T3 = 3;
static const uint8_t T4 = 4;
static const uint8_t T5 = 5;
static const uint8_t T6 = 6;
static const uint8_t T7 = 7;
static const uint8_t T8 = 8;
static const uint8_t T9 = 9;
static const uint8_t T10 = 10;
static const uint8_t T11 = 11;
static const uint8_t T12 = 12;
static const uint8_t T13 = 13;
static const uint8_t T14 = 14;

static const uint8_t Vext = 36;
static const uint8_t LED = 18;
static const uint8_t RST_OLED = 21;
static const uint8_t SCL_OLED = 18;
static const uint8_t SDA_OLED = 17;
#endif /* Pins_Arduino_h */

codeadventure20, 2023-07-27:
don't you need the json file as well?

{
  "build": {
    "arduino": {
      "ldscript": "esp32s3_out.ld",
      "partitions": "default.csv"
    },
    "core": "esp32",
    "extra_flags": [
      "-DWireless_Track",
      "-DARDUINO_USB_MODE=1",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1"      
     ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "heltec_wireless_tracker"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "lora",
    "gps"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Heltec Wireless Tracker",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 1036800
  },
  "url": "https://heltec.org/project/wireless-tracker/",
  "vendor": "Heltec"
}

codeadventure20, 2023-07-27:
also platformio.ini file:

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:heltec_wireless_tracker]
platform = espressif32
board = heltec_wireless_tracker
framework = arduino
monitor_speed = 115200
;debug_tool = esp-prog
;debug_init_break = tbreak setup

; -lheltec_0 ; for no CDC on boot or 
; -lheltec_1 ; for CDC on BOOT
; -DBOARD_HAS_PSRAM ; add it if it has psram
; -DLoRaWAN_DEBUG_LEVEL =0 ; 0 = None / 1 = Freq / 2 = Freq && DIO / 3 = Freq && DIO && PW
; -DREGION_xxnnn = EU868 ; EU868 / EU433 / CN470 / US915 / AU915 / CN779 / AS923 / KR920 / IN865 / US915_HYBRID
; -DACTIVE_REGION=LORAMAC_REGION_xxnnn  ; EU868 / EU433 / CN470 / US915 / AU915 / CN779 / AS923 / KR920 / IN865 / US915_HYBRID
; -DLORAWAN_PREAMBLE_LENGTH=8 ; 8 / 16
; -DLORAWAN_DEVEUI_AUTO=0 ; 0 = Custom / 1 = Generate By ChipID

;build_flags = 
;   -DLoRaWAN_DEBUG_LEVEL=0
;   -DREGION_EU868 
;   -DACTIVE_REGION=LORAMAC_REGION_EU868
;   -DLoRaWAN_DEBUG_LEVEL=0
;   -DLORAWAN_PREAMBLE_LENGTH=8
;   -DLORAWAN_DEVEUI_AUTO=0
;   -D${board}
;   -lheltec_0

codeadventure20, 2023-07-27:
note: I made it work on my local platformio using the above files and release 0.0.8 of https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series

jhiggason, 2023-10-05:
@codeadventure20 Are you able to give a rundown of what you did to make this work?

songpola, 2024-02-02:
The [`framework-arduinoespressif32`](https://github.com/espressif/arduino-esp32), which is the dependency of this platform (`platform-espressif32`), now supported the board. How can we make this platform support the board? @valeros

grostim, 2025-03-31:
Are there any plan for this ?

fredvol, 2025-04-14:
+1 that would be nice .
Thanks for your work

closed 2026-10-16 by valeros