platformio/platform-espressif32 #1149: esp_websocket_client component is missing.

KonradStanski, 2023-07-19:
Looking in my .platformio/packages/framework-espidf/components folder, I find many of the folder here:
https://github.com/espressif/esp-idf/tree/v4.2/components/
missing. The one that I am interested in particular is 
https://github.com/espressif/esp-idf/tree/v4.2/components/esp_websocket_client
Is this something that is being considered to be added?
Thansk!

Jason2866, 2023-07-19:
IDF 4.2 is outdated and out of support. IDF 4.4 and IDF 5.0 are currently used.
IDF 4.4 is needed since Arduino is based on and IDF 5.0 is the latest supported from Platformio. Platformio uses what's provided from espressif in IDF. The component you mentioned is not in actual IDF variants.
So the answer is, no it will not included in provided framework. You have to add as additional component as described in the Docu.

KonradStanski, 2023-07-19:
I've found that the esp_websocket_client component was moved into another repo, but can be added like any other component. For anyone else who comes across this:
https://github.com/espressif/esp-protocols/tree/master/components/esp_websocket_client
How to add components:
https://docs.platformio.org/en/latest/frameworks/espidf.html#esp-idf-components
(just copy the component folder into your components folder)