platformio/platform-espressif32 #1150: Unable to create project on Mac M1 for ESP32 using ESPIDF Framework #5961

evvaroli, 2023-07-19:
Hi!

There is a problem during project creation on **mac m1** with **espidf**. 

- It seems there is a problem with this python module: ModuleNotFoundError: No module named 'chardet'

- Error occurs about 5-10 sec after starting project creation. It seems the error appears during tool-chain installation.

- Have tested it also on a Windows and Linux system. Problem appears only on the Mac-System

**Releases:** 
    - VSCODE: June 2023 (version 1.80)
    - PlatformIO: 3.3.0 (2023-07-10)

PIO Core Call Error: "The following files/directories have been created in /Users/brx/Downloads/esp-32-test\ninclude - Put project header files here\nlib - Put project specific (private) libraries here\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nResolving az-delivery-devkit-v4 dependencies...\nAlready up-to-date.\nUpdating metadata for the vscode IDE...\n\n\nError: Processing az-delivery-devkit-v4 (platform: espressif32; board: az-delivery-devkit-v4; framework: espidf)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/az-delivery-devkit-v4.html\nPLATFORM: Espressif 32 (6.1.0) > AZ-Delivery ESP-32 Dev Kit C V4\nHARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash\nDEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\nPACKAGES: \n - framework-espidf @ 3.50001.0 (5.0.1) \n - tool-cmake @ 3.16.4 \n - tool-esptoolpy @ 1.40500.0 (4.5.0) \n - tool-ninja @ 1.9.0 \n - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) \n - toolchain-xtensa-esp32 @ 11.2.0+2022r1\nReading CMake configuration...\n-- Found Git: /usr/bin/git (found version \"2.37.1 (Apple Git-137.1)\") \n-- The C compiler identification is GNU 11.2.0\n-- The CXX compiler identification is GNU 11.2.0\n-- The ASM compiler identification is GNU\n-- Found assembler: /Users/brx/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc\n-- Check for working C compiler: /Users/brx/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc\n-- Check for working C compiler: /Users/brx/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- works\n-- Detecting C compiler ABI info\n-- Detecting C compiler ABI info - done\n-- Detecting C compile features\n-- Detecting C compile features - done\n-- Check for working CXX compiler: /Users/brx/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++\n-- Check for working CXX compiler: /Users/brx/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ -- works\n-- Detecting CXX compiler ABI info\n-- Detecting CXX compiler ABI info - done\n-- Detecting CXX compile features\n-- Detecting CXX compile features - done\n-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.\n-- Building ESP-IDF components for target esp32\n-- Configuring incomplete, errors occurred!\nSee also \"/Users/brx/Downloads/esp-32-test/.pio/build/az-delivery-devkit-v4/CMakeFiles/CMakeOutput.log\".\n\nfatal: not a git repository (or any of the parent directories): .git\nCMake Error at /Users/brx/.platformio/packages/framework-espidf/tools/cmake/build.cmake:519 (message):\n Traceback (most recent call last):\n\n File \"/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/requests/compat.py\", line 11, in <module>\n import chardet\n\n ModuleNotFoundError: No module named 'chardet'\n\n \n\n During handling of the above exception, another exception occurred:\n\n \n\n Traceback (most recent call last):\n\n File \"/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/runpy.py\", line 197, in _run_module_as_main\n return _run_code(code, main_globals, None,\n File \"/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/runpy.py\", line 87, in _run_code\n exec(code, run_globals)\n File \"/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/idf_component_manager/prepare_components/__main__.py\", line 4, in <module>\n from .prepare import main\n File \"/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/idf_component_manager/prepare_components/prepare.py\", line 18, in <module>\n from ..core import ComponentManager\n File \"/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/idf_component_manager/core.py\", line 17, in <module>\n import requests\n File \"/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/requests/__init__.py\", line 45, in <module>\n from .exceptions import RequestsDependencyWarning\n File \"/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/requests/exceptions.py\", line 9, in <module>\n from .compat import JSONDecodeError as CompatJSONDecodeError\n File \"/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/requests/compat.py\", line 13, in <module>\n import charset_normalizer as chardet\n File \"/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/charset_normalizer/__init__.py\", line 24, in <module>\n from .api import from_bytes, from_fp, from_path\n File \"/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/charset_normalizer/api.py\", line 5, in <module>\n from .cd import (\n File \"/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/charset_normalizer/cd.py\", line 9, in <module>\n from .md import is_suspiciously_successive_range\n\n ImportError:\n dlopen(/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/charset_normalizer/md.cpython-39-darwin.so,\n 0x0002): tried:\n '/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/charset_normalizer/md.cpython-39-darwin.so'\n (mach-o file, but is an incompatible architecture (have 'arm64', need\n 'x86_64')),\n '/System/Volumes/Preboot/Cryptexes/OS/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/charset_normalizer/md.cpython-39-darwin.so'\n (no such file),\n '/Users/brx/.platformio/penv/.espidf-5.0.1/lib/python3.9/site-packages/charset_normalizer/md.cpython-39-darwin.so'\n (mach-o file, but is an incompatible architecture (have 'arm64', need\n 'x86_64'))\n\nCall Stack (most recent call first):\n /Users/brx/.platformio/packages/framework-espidf/tools/cmake/project.cmake:440 (idf_build_process)\n CMakeLists.txt:3 (project)\n\n\n\n========================== [FAILED] Took 2.18 seconds =========================="

Jason2866, 2023-07-19:
Use latest espressif framework. This issue is solved. You are using v6.1.0.

evvaroli, 2023-07-20:
> Use latest espressif framework v6.3.2. This issue is solved. You are using v6.1.0.

Made update to new framework v6.3.2, problem is still here.

PIO Core Call Error: "The following files/directories have been created in /Users/brx/Workspace/projects/ESP/test_esp32-espidf_v1\ninclude - Put project header files here\nlib - Put project specific (private) libraries here\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nResolving az-delivery-devkit-v4 dependencies...\nAlready up-to-date.\nUpdating metadata for the vscode IDE...\n\n\nError: Processing az-delivery-devkit-v4 (platform: espressif32; board: az-delivery-devkit-v4; framework: espidf)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/az-delivery-devkit-v4.html\nPLATFORM: Espressif 32 (6.3.2) > AZ-Delivery ESP-32 Dev Kit C V4\nHARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash\nDEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\nPACKAGES: \n - framework-espidf @ 3.50002.230601 (5.0.2) \n - tool-cmake @ 3.16.4 \n - tool-esptoolpy @ 1.40501.0 (4.5.1) \n - tool-ninja @ 1.9.0 \n - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) \n - toolchain-xtensa-esp32 @ 11.2.0+2022r1\nReading CMake configuration...\n-- Found Git: /usr/bin/git (found version \"2.37.1 (Apple Git-137.1)\") \n-- The C compiler identification is GNU 11.2.0\n-- The CXX compiler identification is GNU 11.2.0\n-- The ASM compiler identification is GNU\n-- Found assembler: /Users/brx/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc\n-- Check for working C compiler: /Users/brx/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc\n-- Check for working C compiler: /Users/brx/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- works\n-- Detecting C compiler ABI info\n-- Detecting C compiler ABI info - done\n-- Detecting C compile features\n-- Detecting C compile features - done\n-- Check for working CXX compiler: /Users/brx/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++\n-- Check for working CXX compiler: /Users/brx/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ -- works\n-- Detecting CXX compiler ABI info\n-- Detecting CXX compiler ABI info - done\n-- Detecting CXX compile features\n-- Detecting CXX compile features - done\n-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.\n-- Building ESP-IDF components for target esp32\n-- Configuring incomplete, errors occurred!\nSee also \"/Users/brx/Workspace/projects/ESP/test_esp32-espidf_v1/.pio/build/az-delivery-devkit-v4/CMakeFiles/CMakeOutput.log\".\n\nfatal: not a git repository (or any of the parent directories): .git\nCMake Error at /Users/brx/.platformio/packages/framework-espidf/tools/cmake/build.cmake:519 (message):\n Traceback (most recent call last):\n\n File \"/Users/brx/.platformio/penv/.espidf-5.0.2/lib/python3.9/site-packages/requests/compat.py\", line 11, in <module>\n import chardet\n\n ModuleNotFoundError: No module named 'chardet'\n\n \n\n During handling of the above exception, another exception occurred:\n\n \n\n Traceback (most recent call last):\n\n File \"/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/runpy.py\", line 197, in _run_module_as_main\n return _run_code(code, main_globals, None,\n File \"/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/runpy.py\", line 87, in _run_code\n exec(code, run_globals)\n File \"/Users/brx/.platformio/penv/.espidf-5.0.2/lib/python3.9/site-packages/idf_component_manager/prepare_components/__main__.py\", line 4, in <module>\n from .prepare import main\n File \"/Users/brx/.platformio/penv/.espidf-5.0.2/lib/python3.9/site-packages/idf_component_manager/prepare_components/prepare.py\", line 18, in <module>\n from ..core import ComponentManager\n File \"/Users/brx/.platformio/penv/.espidf-5.0.2/lib/python3.9/site-packages/idf_component_manager/core.py\", line 17, in <module>\n import requests\n File \"/Users/brx/.platformio/penv/.espidf-5.0.2/lib/python3.9/site-packages/requests/__init__.py\", line 45, in <module>\n from .exceptions import RequestsDependencyWarning\n File \"/Users/brx/.platformio/penv/.espidf-5.0.2/lib/python3.9/site-packages/requests/exceptions.py\", line 9, in <module>\n from .compat import JSONDecodeError as CompatJSONDecodeError\n File \"/Users/brx/.platformio/penv/.espidf-5.0.2/lib/python3.9/site-packages/requests/compat.py\", line 13, in <module>\n import charset_normalizer as chardet\n File \"/Users/brx/.platformio/penv/.espidf-5.0.2/lib/python3.9/site-packages/charset_normalizer/__init__.py\", line 24, in <module>\n from .api import from_bytes, from_fp, from_path, is_binary\n File \"/Users/brx/.platformio/penv/.espidf-5.0.2/lib/python3.9/site-packages/charset_normalizer/api.py\", line 5, in <module>\n from .cd import (\n File \"/Users/brx/.platformio/penv/.espidf-5.0.2/lib/python3.9/site-packages/charset_normalizer/cd.py\", line 9, in <module>\n from .md import is_suspiciously_successive_range\n\n ImportError:\n dlopen(/Users/brx/.platformio/penv/.espidf-5.0.2/lib/python3.9/s [... cut ...] cpython-39-darwin.so'\n (no such file),\n '/Users/brx/.platformio/penv/.espidf-5.0.2/lib/python3.9/site-packages/charset_normalizer/md.cpython-39-darwin.so'\n (mach-o file, but is an incompatible architecture (have 'arm64', need\n 'x86_64'))\n\nCall Stack (most recent call first):\n /Users/brx/.platformio/packages/framework-espidf/tools/cmake/project.cmake:447 (idf_build_process)\n CMakeLists.txt:3 (project)\n\n\n\n========================== [FAILED] Took 1.64 seconds =========================="

mhaberler, 2023-07-20:
it seems the replacement gdb is lacking the Python framework it was built with 

they seem to be build in the CI jobs, but the debugger obviously wont run with different Python install paths

where is the download for those Python frameworks?

Jason2866, 2023-07-20:
@evvaroli  Have you deleted the whole (hidden) folder `.platformio`?
Old installed stuff can still make problems. With deleting the folder Platformio installs everything fresh.

evvaroli, 2023-07-21:
removing `.platformio` and reinstallation solved the problem. Thx to all.

closed 2023-08-01 by valeros