platformio/platform-espressif32 #1151: Board request: ESP8684-DevKitM-1

Kuenlun, 2023-07-19:
Please, could someone add or help me add this board based on the ESP-C2?
Or just give a quick fix so I (and many of us) can program this devkit.

https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp8684/esp8684-devkitm-1/user_guide.html

Thank you!

Jason2866, 2023-07-19:
To add the board is not the only needed change. See provided [PR](https://github.com/platformio/platform-espressif32/pull/1054) for supporting C6. The change for the C2 is similar.

Kuenlun, 2023-07-19:
Great, I will wait patiently.
I decided to create this Issue since there seems to be a lack of information regarding the ESP8684-DevKitM-1 board and the ESP-C2 microcontroller on the platform. Considering that the ESP-C2 is quite commonly used nowadays and was released earlier than the ESP-C6, it strikes me as odd that there is no board support available in Platformio.

Thank you so much!

Kuenlun, 2023-07-20:
Thank you very much, @Jason2866, for your contribution and especially for your promptness. I've tested it on an ESP8684-DevKitM-1 board and it's working properly.
Just for those who are going to use this board, please remember that it has a quartz crystal operating at 26 MHz, not 40 MHz as most of Espressif's boards do, so you need to configure the corresponding option in menuconfig (https://github.com/espressif/esp-idf/issues/10782#issuecomment-1484390947).
Also, to properly view the serial monitor, I had to set the baud rate to 74800.
I'm leaving the "platformio.ini" file I'm using here for reference:
```
[platformio]
default_envs = ESP8684_Debug

[env]
platform = https://github.com/Jason2866/platform-espressif32.git#C2_IDF50
framework = espidf
board = esp32-c2-devkitm-1
monitor_speed = 74800
monitor_raw = yes
build_flags =
  -Wall
  -Wextra
  -Werror

[env:ESP8684_Debug]
build_type = debug
build_flags =
  -D configUSE_TRACE_FACILITY=1
  -D configGENERATE_RUN_TIME_STATS=1
  -D configUSE_STATS_FORMATTING_FUNCTIONS=1

[env:ESP8684_Release]
build_type = release
```

It's odd that when "env" is named "mydebug" in the "platformio.ini" file, it triggers an error:
```
[env:mydebug]
platform = https://github.com/Jason2866/platform-espressif32.git#C2_IDF50
framework = espidf
board = esp32-c2-devkitm-1
```

```
Building .pio\build\mydebug\firmware.bin
esptool.py v4.6.2
Creating esp32c2 image...
Merged 2 ELF sections

A fatal error occurred: Contents of segment at SHA256 digest offset 0xb0 are not all zero. Refusing to overwrite.
*** [.pio\build\mydebug\firmware.bin] Error 2
```

Jason2866, 2023-07-20:
@Kuenlun  To avoid to forget to set the correct XTAL frequency, adding a sdkconfig.defaults file in the project folder is a safe way. This can be done MCU specific. See the example [here](https://github.com/Jason2866/platform-espressif32/blob/C2_IDF50/examples/espidf-ulp/sdkconfig.defaults.esp32s3)

Jason2866, 2023-10-02:
Solved with PR https://github.com/platformio/platform-espressif32/pull/1203

soufiane707, 2024-04-18:
> Thank you very much, @Jason2866, for your contribution and especially for your promptness. I've tested it on an ESP8684-DevKitM-1 board and it's working properly. Just for those who are going to use this board, please remember that it has a quartz crystal operating at 26 MHz, not 40 MHz as most of Espressif's boards do, so you need to configure the corresponding option in menuconfig ([espressif/esp-idf#10782 (comment)](https://github.com/espressif/esp-idf/issues/10782#issuecomment-1484390947)). Also, to properly view the serial monitor, I had to set the baud rate to 74800. I'm leaving the "platformio.ini" file I'm using here for reference:
> 
> ```
> [platformio]
> default_envs = ESP8684_Debug
> 
> [env]
> platform = https://github.com/Jason2866/platform-espressif32.git#C2_IDF50
> framework = espidf
> board = esp32-c2-devkitm-1
> monitor_speed = 74800
> monitor_raw = yes
> build_flags =
>   -Wall
>   -Wextra
>   -Werror
> 
> [env:ESP8684_Debug]
> build_type = debug
> build_flags =
>   -D configUSE_TRACE_FACILITY=1
>   -D configGENERATE_RUN_TIME_STATS=1
>   -D configUSE_STATS_FORMATTING_FUNCTIONS=1
> 
> [env:ESP8684_Release]
> build_type = release
> ```
> 
> It's odd that when "env" is named "mydebug" in the "platformio.ini" file, it triggers an error:
> 
> ```
> [env:mydebug]
> platform = https://github.com/Jason2866/platform-espressif32.git#C2_IDF50
> framework = espidf
> board = esp32-c2-devkitm-1
> ```
> 
> ```
> Building .pio\build\mydebug\firmware.bin
> esptool.py v4.6.2
> Creating esp32c2 image...
> Merged 2 ELF sections
> 
> A fatal error occurred: Contents of segment at SHA256 digest offset 0xb0 are not all zero. Refusing to overwrite.
> *** [.pio\build\mydebug\firmware.bin] Error 2
> ```

hi please , were you able to use the arduino framework

Jason2866, 2024-04-19:
@soufiane707 My fork (not fully compatible to original!) does support the C2 with Arduino
Can be used with
```
platform = https://github.com/Jason2866/platform-espressif32.git
```
Do NOT open an issue here when you encounter problems.

sermayoral, 2024-10-01:
Hi everyone!!

Why are pull requests about ESP32-C2 not being merged? What is the problem exactly? Because the only problem I see in all the pull requests is the lack of words. No one answers but no one does anything. We have been waiting for support for this board for over a year and it is still not available. There are people like @Jason2866 who have coded everything but it is totally ignored and we don't even know why. Please, can you accept this pull request or at least describe why it is not being merged? Thank you all very much.

ajowsey, 2025-01-04:
Just commenting this post was still very helpful for using ESP32-C2-DevKitM-1. I started the project with "Espressif ESP32-C3-DevKitM-1" and then just added the **platform.ini** I found in the comments

One error came up in the **platform.ini** with git because the branch no longer existed. And I had to change

`platform = https://github.com/Jason2866/platform-espressif32.git#C2_IDF50` 

to

`platform = https://github.com/Jason2866/platform-espressif32.git`

For the crystal. I changed the "**sdkconfig.esp32-c2-devkitm-1**" to stop using:

CONFIG_XTAL_FREQ_40=y
CONFIG_XTAL_FREQ=40

to 

CONFIG_XTAL_FREQ_26=y
CONFIG_XTAL_FREQ=26

Flashed and could see some demo serial prints. Yay!

hb020, 2025-05-30:
Have gotten nowhere so far.

Device: ESP8684-WROOM-04C with 2MB flash (so not in the devkit, but that shouldn't change much.)

Strapped EN, IO8 and IO9 to high. Buttons on EN and IO9 for resp. reset and flash mode.

```
[platformio]
default_envs = esp32-c2-devkitm-1

[env:esp32-c2-devkitm-1]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
framework = arduino
board = esp32-c2-devkitm-1
monitor_speed = 115200
build_flags =
```

When using `platform = https://github.com/Jason2866/platform-espressif32.git` I cannot compile:
```
sh: riscv32-esp-elf-g++: command not found
sh: riscv32-esp-elf-gcc: command not found
```

When using `platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip` I can compile and upload, but all programs I uploaded result in the same: 
* no output on serial, not even glitches during boot
* no control over the pins, they are all static (either GND or VCC), not even glitches during boot, except IO6, which has a glitch to GND about once every 3 seconds

Of course I wiped all `~/.platformio/platforms/espressif32` directories between platform changes.

Example of simple program I tested:
```
#include <Arduino.h>
void setup()
{
  Serial.begin(115200);
  Serial.println("Setup done");
}

void loop()
{
  delay(1000);
  Serial.println("Loop");
}
```

What are the settings required? 

FYI, the upload uses the following parameters to esptool:

```
--chip esp32c2 
--port "/dev/cu.usbserial-FTHJGC2Z" 
--baud 460800 
--before default_reset 
--after hard_reset 
write_flash 
-z 
--flash_mode dio 
--flash_freq 60m 
--flash_size detect 
0x0000 .pio/build/esp32-c2-devkitm-1/bootloader.bin 
0x8000 .pio/build/esp32-c2-devkitm-1/partitions.bin 
0xe000 ~/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin 
0x10000 .pio/build/esp32-c2-devkitm-1/firmware.bin
```

esptool produces the following:

 > General output:
esptool.py v4.8.9
Serial port /dev/cu.usbserial-FTHJGC2Z
Connecting....
Chip is ESP32-C2 (revision v1.2)
Features: WiFi, BLE
Crystal is 26MHz
MAC: c0:5d:89:50:f8:78
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.

> chip_id
Warning: ESP32-C2 has no Chip ID. Reading MAC instead.
MAC: c0:5d:89:50:f8:78

> flash_id 
Manufacturer: a1
Device: 4015
Detected flash size: 2MB

> read_flash_status
Status value: 0x0200

> get_security_info
Security Information:
=====================
Flags: 0x00000000 (0b0)
Chip ID: 12
API Version: 1
Secure Boot: Disabled
Flash Encryption: Enabled
SPI Boot Crypt Count (SPI_BOOT_CRYPT_CNT): 0x7

Jason2866, 2025-05-30:
It is not working since setting for the flash with 2MB is not done. The used boards manifest has a setting for 4MB.

hb020, 2025-05-31:
It looks like I have a bootstrapping problem, because even on a fresh ESP_IDF 6.0 install and installing the espressif "hello world" example it does the same.

Jason2866, 2025-05-31:
Issue is wrong addressed here. Platformio does **not** support this MCU.

hb020, 2025-05-31:
for posterity: the seller crippled the devices. So impossible to reprogram without knowing their encryption key. Bummer.