platformio/platform-espressif32 #1152: Add board ESP32-C6-EVB from Olimex

progettihwsw, 2023-07-20:
Hi, i need to use this board
https://www.olimex.com/Products/IoT/ESP32-C6/ESP32-C6-EVB/open-source-hardware
but i can't find a valid platformio.ini file
Thanks

Jason2866, 2023-07-20:
There is no C6 support until now. For C6 support there is IDF 5.1 needed.
Until now Platformio supports IDF 5.0.
So the open PR for supporting a C6 board is currently useless.
https://github.com/platformio/platform-espressif32/pull/1054

DAVe3283, 2023-11-12:
Now that #1054 is merged and there is initial ESP32-C6 support, is it possible to support this board?

GourouLubrik, 2024-06-25:
I've discovered the EVB-C6-EVB through this news that features what looks like a clone: https://www.cnx-software.com/2024/06/24/maker-go-esp32-c6-evb-quad-relay-board-supports-wi-fi-6-ble-5-zigbee-for-smart-home-applications/ 
I'd love to try zigbee with this board with an esphome, but it seems that platformio support is required. so I'd like to add my vote to support for this board: having wide input voltage, onboard octocouplers and relays and yet many GPIO routed out of the box would make my semi-diy appliance much easier.